vanvught/rpidmx512 #126: Debug UART: Implement read API's

vanvught, 2020-07-26:
This is a follow-up for #118 

It would make it possible to connect for example an ESP8266 ESP-01S WiFi module. 
Also it would be possible to have some kind of telnet session for configuration actions. 

Details to follow

cc: @hippyau

hippyau, 2020-07-26:
Is all that is needed to implement?

(https://github.com/uli/allwinner-bare-metal/blob/master/uart.c)

// UART has received data?
unsigned char uart_rx_ready()
{
  return (UART0_LSR & 1);
}

(https://github.com/mprymek/allwinner-baremetal/blob/master/H3/orange-pi/uart01/uart.c)
unsigned int uart_recv ( void )
{
      return(GET32(UART_RBR));
}

vanvught, 2020-07-26:
> Is all that is needed to implement?

That is just 1% of the coding ;-)

There must a proper C++ Class with a `readline` method. This class must also have a `run()` method.
Then the remote configuration framework must be re-written in order to make it possible to receive both UDP and UART commands.

hippyau, 2020-07-26:
hehe, yes, totally, and I'm keen to have a crack at the readline and beyone...  I have written a few CLI's in my time :-)

Just need to know where to get the data from...  not much needed in here though?

vanvught, 2020-07-26:
> not much needed in here though?
https://github.com/vanvught/rpidmx512/blob/master/lib-h3/src/h3_uart0_debug.c

Yes, there will be just the non-blocking FIFO reader.

hippyau, 2020-07-26:
I made these, 

```
uint8_t uart0_available()
{
	return (H3_UART0->LSR & 1);
}

uint8_t uart0_recv ( void )
{
	return((uint8_t)H3_UART0->RBR);
}
```

It's these bits I suck at... in recv

```
$MAKE_FLAGS [-DLTC_READER -DNDEBUG -DENABLE_TC1602]
src/h3_uart0_debug.c: In function 'uart0_recv':
src/h3_uart0_debug.c:93:26: error: 'H3_UART_TypeDef {aka struct T_H3_UART}' has no member named 'RBR'
  return((uint8_t)H3_UART0->RBR);
                          ^~
```

but I can see in "h3.h" the union in T_H3_UART that RBR exists?

```

typedef struct T_H3_UART {
	union {							///< 0x00
		__I uint32_t RBR;			///< Receive Buffer register
		__O uint32_t THR;			///< Transmit Holding register
		__IO uint32_t DLL;			///< Divisor Latch Low register
	} O00;
	union {							///< 0x04
		__IO uint32_t IER;			///< Interrupt Enable Register
		__IO uint32_t DLH;			///< Divisor Latch High register

	} O04;
	union {							///< 0x08
		__I uint32_t IIR;			///< Interrupt Identify Register
		__O uint32_t FCR;			///< FIFO Control Register
	} O08;
	__IO uint32_t LCR;				///< 0x0C Line Control Register
	__IO uint32_t MCR;				///< 0x10 Modem Control Register
	__I uint32_t LSR;				///< 0x14 Line Status Register
	__I uint32_t MSR;				///< 0x18 Modem Status Register
	__I uint32_t SCH;				///< 0x1C
	__I uint32_t RES1[23];			///< unused UART registers
	__I uint32_t USR;				///< 0x7C system status register
	__I uint32_t TFL;				///< 0x80
	__I uint32_t RFL;				///< 0x84
	__I uint32_t RES2[7]; 			///< unused UART registers
	__I uint32_t HALT; 				///< 0xA4 halt tx register
} H3_UART_TypeDef;
```

vanvught, 2020-07-26:
Requirements:

1. FIFO must be used.
2. It must be indicated when there is nothing to receive, including receiving time-out.
3. Non-blocking.

It should be something like this: https://github.com/vanvught/rpidmx512/blob/master/lib-device/src/sc16is740.cpp#L177
And there is some code which we can reuse from MIDI In.

hippyau, 2020-07-26:
Confirmed working :) 

```
uint8_t uart0_available(void) {
	return (H3_UART0->LSR & UART_LSR_DR);
}

uint8_t uart0_getc(void) {
	return((uint8_t)H3_UART0->O00.RBR);
}
```

vanvught, 2020-07-26:
> Confirmed working :)
> 
> uint8_t uart0_available(void) {
> 	return (H3_UART0->LSR & UART_LSR_DR);
> }
> 
> uint8_t uart0_getc(void) {
> 	return((uint8_t)H3_UART0->O00.RBR);
> }

But it does work with all the requirements :)

1. FIFO must be used.
2. It must be indicated when there is nothing to receive, including receiving time-out.
3. Non-blocking.

hippyau, 2020-07-26:
Sorry, yes it will eventually...

We can add this to https://github.com/vanvught/rpidmx512/blob/master/lib-h3/src/h3_uart0_debug.c 

These will enable non-blocking reception on uart 0, and that is working for me on the hardware.

TODO: FIFO.

vanvught, 2020-07-27:
Hi Hip,

The big question here, (1) are we going to work with run-time configuration or (2) are we working with SPI flash stores, similar to the UDP remote configuration frame work?

With (1) we would implement just CLI's for example the network configuration and some other generic configuration items. The change is immediately active and written to SPI flash.

Greetings, Arjan

hippyau, 2020-07-28:
Hey Arjan,

 I imagine a simple protocol 'transparent' way to talk to an ESP chip (or any mcu) could be by the oPi telling the ESP the relevant UDP ports it should open, then the ESP gathers them all up and they are sent as serial like `[5443]ltc!backward#5` or `[2905]!tftp#1` and compatible with the existing format.  In reverse, the oPi sends back to the ESP if a response like `[2905]display:Off` if expected, ESP sends to the last IP it received from on each port.   Can you see any problems with that, besides coding it?

I'd also been thinking a bit trickier, but I'd like to use very simple tree structure like OSC, to be able to: 

```
opi>  showfile/start
opi>  devices/led_type=WS2812B
opi>  artnet/direction=output
opi>  artnet/universe=0
opi>  ltc/time=12:34:56.29
opi>  ltc/start
```

But I realize that is a bunch of work to do.  

I just now I've pulled the experimental branch and will start working from your code and messing with in there. 
I do fail to compiler here though, and not too sure if a compiler version issue?  
```
$BUILD_DIRS [build_h3/src build_h3/src/h3 build_h3/posix build_h3/rtc build_h3/ff12c build_h3/ff12c/option build_h3/src/h3/sdcard build_h3/debug]
$TARGET [lib_h3/libhal.a ]
ff12c/ff.c:4:32: error: unknown option after '#pragma GCC diagnostic' kind [-Werror=pragmas]
 #pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
                                ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors

```

hippyau, 2020-07-28:
Oh wow, I just found the code you wrote already in shell.cpp... man you are fast!!  and a programming god! 

I am going to use this and hack together my own idea of a CLI, and it's going to be fat and ugly... I guarantee you.... but still should fit and run fast because it's not going to be used a huge amount :)

you are a legend!

vanvught, 2020-07-28:
> I'd also been thinking a bit trickier, but I'd like to use very simple tree structure like OSC, to be able to:
>
> opi>  showfile/start
> opi>  devices/led_type=WS2812B
> opi>  artnet/direction=output [...]

Yes, that is far too much work ;-)

This serial telnet look-a-like interface is just for doing some basic configuration work. It is not intended for using at run-time for example LTC. 

Basic configuration is for example the network and the remote configuration options. 

For example:

Run-time configuration:

`opi> set ip 192.168.2.251`

Or working with the SPI flash (reboot needed):

`opi> set network.txt ip_address=192.168.2.251`

vanvught, 2020-07-28:
> $BUILD_DIRS [build_h3/src build_h3/src/h3 build_h3/posix build_h3/rtc build_h3/ff12c build_h3/ff12c/option build_h3/src/h3/sdcard build_h3/debug]
> $TARGET [lib_h3/libhal.a ]
> ff12c/ff.c:4:32: error: [...]

Are you using a GCC compiler? Which version?

hippyau, 2020-07-28:
100%   keep it simple :)

```
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

vanvught, 2020-07-28:
> 
> gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
> Copyright (C) 2017 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

The arm cross compiler ?

I have 8.4.0 and 9.31

```
@nuc-i5:~$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GCC) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

@MacBook-Air ~ % arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

hippyau, 2020-07-28:
Duh!   i'm updating :)   thanks man.

```
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

vanvught, 2020-07-28:
One caveat here for example:

`opi>` get artnet.txt

As this 'txt file' contains a lot of characters, it might trigger the `wdog` and it reboots the OPi.

I've found this behavior in void SpiFlashStore::Dump() 
https://github.com/vanvught/rpidmx512/blob/master/lib-spiflashstore/src/spiflashstore.cpp#L342 

So retrieving a lot of characters must be implemented as a state-machine (somehow:).

hippyau, 2020-07-29:
> As this 'txt file' contains a lot of characters, it might trigger the wdog and it reboots the OPi.

I imagine could get away with h3_watchdog_restart() in uart0_puts() and just dump files line by line instead of while putc?

> ```
> Or working with the SPI flash (reboot needed):
> 
> opi> set network.txt ip_address=192.168.2.251
> ```

I don't yet know how to change just one line of a file, is there a function or will I need to read the file and make changes and save it all again?  I haven't dived into that part of the code yet :)

I think for usability, it would be nice for example to be able to 

```
opi> set ip 10.0.0.123  
opi> set hostname orangepi
opi> artnet 0 universe 1  

opi> store   // make permanent 
```

So I can call this to store current network settings?
`void StoreNetwork::Update(const struct TNetworkParams *pNetworkParams)`

vanvught, 2020-07-29:
> opi> set ip 10.0.0.123  
> opi> set hostname orangepi

These are already stored in the SPI flash. See here https://github.com/vanvught/rpidmx512/blob/master/lib-network/src/h3/networkh3emac.cpp#L287

And do not require a reboot.

Setting Art-Net should be done re-using the remote config methods. And is working on the SPI flash only, hence these configuration changes would need a reboot to be active

For example:

`set artnet.txt universe=1`

vanvught, 2020-07-29:
> I imagine could get away with h3_watchdog_restart() in uart0_puts() and just dump files line by line instead of while putc?

The main issue is that the OPi is 'locked' in outputting characters to the UART. Hence there is no network (no anything;). We can easily check the time spent in sending characters by checking the frequency of the blinking led.

vanvught, 2020-07-30:
[http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/uart0-shell](http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/uart0-shell)

hippyau, 2020-07-31:
Hey Arjan,

 I noted this, and I am curious...

```
	// TOOD We know the m_Argv[] length in ValidateArg. Let's store it in member variable?
	uint32_t nArgv0Length = strlen(m_Argv[0]);
```

I wonder should do  `m_nArgvLength[MAXARG]`, or per command implement (if needed)?

your code is elegant and refined,  mine is sloppy in comparison  :)

hippyau, 2020-07-31:
I want to expand on this, so that each firmware could support custom commands.

how do you want it?

```
#if defined (ARTNET_NODE)
		case CmdIndex::ARTNET:
			CmdArtnet();
			break;
#endif
#if defined (E131_BRIDGE)
		case CmdIndex::ARTNET:
			CmdE131();
			break;
#endif
```

and have a localshellcommands.cpp in each firmware?

or is a better way to do this across the firmwares?

vanvught, 2020-07-31:
I am not sure why we should have 

> custom commands

All possible configuraton items are available.

hippyau, 2020-07-31:
I am going for some real time control, which can be done via UDP messages, but not by .txt files.   My vision is to eventually

```
opi> tc stop
opi> tc start 12:34:56;01
opi> tc start
opi> tc start @ 12:34:56;01
opi> tc rewind 1s  // or (ms), or (f)rames
opi> tc stop
opi> ntp_sync
opi> tc start U12:34:56.01  // commence playback at specific time, synchronizing multiple devices with ntp time.

```

vanvught, 2020-07-31:
Real-time control should be done with the Ethernet interface only: 

- human readable UDP
- OSC

The .txt files are for working with the SPI flash -> reboot needed for activation.

The debug UART0 should be used for (initial) configuration only.

> Also it would be possible to have some kind of telnet session for configuration actions.

hippyau, 2020-07-31:
Okay, sorry... not real time control, I will explain my intentions...   I realize your system is so skinny and sexy - she is beautiful :)

but I guess I want to build an application on top of your system.....

Goal for me, I want for accurate continuous TC generation (well done :1st_place_medal:), controlled via a serial interface (in progress :+1: ) exactly telnet/shell style. 

I want for commands to control TC.  Further I will implement a `tc start U12:34:56.01` command that would start TC generation at an agreed NTP [wall clock time](https://en.wikipedia.org/wiki/Clock#Digital) as we have tens of ms or better resolution(?).  

I know [PTP](https://en.wikipedia.org/wiki/Precision_Time_Protocol) would be better, but NTP it is fit for purpose(?, for me).

For maybe just 1 or 2 max frames of latency from the UART transmission through to execution (just start/stop/set),  is not significant in a lot of potential applications.   

These simple start/stop serial commands useful with most every automation system on earth like [Crestron](https://en.wikipedia.org/wiki/Crestron_Electronics), [AMX](https://en.wikipedia.org/wiki/AMX) etc...

Maybe there is other firmwares that can benefit from 'real time' UART commands like `opi> osc send /the/osc/data`,  or I will just do my own separate stuff like this? 

Is there much room, is this a problem?  Do we have lots of RAM available  (how to report free memory?) ?  

Cheers,
  Hippy

vanvught, 2020-07-31:
Hi Hippy,

> Is there much room, is this a problem?

The issue here is that all debug uart0 functions are busy-waiting, hence it should not be used in run-time.

> Do we have lots of RAM available (how to report free memory?) ?

There is around 250 Mb free memory ;-)

I have the idea implementing the `dump malloc` Together with `dump linker` it gives an idea about the memory used (in debug mode;). 

When UART cmd's are really needed, then I would use another OPi for `UART -> UDP commands `conversion. 

I will read into PTP. Maybe we can add this option as well. I've never seen PTP before ;)

Priority, get the new release with #120 on GitHub ... this weekend. 

Greetings, Arjan

vanvught, 2020-07-31:
> There is around 250 Mb free memory ;-)

Oops, big mistake here. We need to check the compressed firmware. That may not exceed https://github.com/vanvught/rpidmx512/blob/master/scripts/build_this.sh#L18

vanvught, 2020-07-31:
> The debug UART0 should be used for (initial) configuration only.

We could do some investigation by having a scope on GPIO 19 and see the impact using the debug UART shell on the led frequency.

hippyau, 2020-07-31:
Hehe, oh yes, I remember firmware compression ;) 

I am first time contributor, I still learning how the universe works...  please you keep doing the important stuff,  I will do my less important stuff, and invite to pull.

git, and github, still  the process still confuses me.  I am learning :)

my real question is what is your preferred way to include an optional feature within a firmware.... like my suggested extensions to the LTC firmware.

#if defined(#SHELL_LTC)  or something else?

> I will read into PTP. Maybe we can add this option as well. I've never seen PTP before ;)

Awesome, it is the Arnolad Schwarzenegger of time protocols  :) :+1: 

Always learning, Cheers,  Hip

hippyau, 2020-07-31:
> > The debug UART0 should be used for (initial) configuration only.
> 
> We could do some investigation by having a scope on GPIO 19 and see the impact using the debug UART shell on the led frequency.

Something I can do eventually :)

hippyau, 2020-07-31:
sorry, another naive question....  can we use the other cores yet?  seems like a perfect application for an auxiliary service.

vanvught, 2020-07-31:
> can we use the other cores yet? 

The active core0 is running at 408 MHz, that should be enough for all applications.

Enabling more core's will include that we need to write code with semaphores  etc. And eventually, core's can be waiting on other cores.

vanvught, 2020-07-31:
> #if defined(#SHELL_LTC) or something else?

The same as used here -> https://github.com/vanvught/rpidmx512/blob/master/lib-remoteconfig/include/remoteconfig.h#L202

hippyau, 2020-07-31:
Thanks, I will proceed with ifdef(s)  and single core (cool temperature)

hippyau, 2020-08-02:
Hey,  I just realize most of the functionality I am looking to implement with serial commands shares code, can and does the same functionality as osc control,  can I just fudge into a cmd into osc.run() or do you want separate shell code?

`opi> /tc/start/set/10/00/00/00`

and then add new osc functionality as required, both win...

vanvught, 2020-08-02:
I am more in favor re-using the UDP commands.

For example splitting `ltc!start#hh:mm:ss.ff` into:

`set ltc start#hh:mm:ss.ff`

I have not checked yet how to make the UDP command interface public. It was not designed as such ;)

hippyau, 2020-08-02:
Yep, okay :)  Leave it with me,  i'll make a mess and see where we go from there :)

vanvught, 2020-08-04:
Hi Hippy,

Looks like that all commands are public:

```
	// Control
	void ActionStart(bool bDoReset = true);
	void ActionStop();
	void ActionResume();
	void ActionReset();
	void ActionSetStart(const char *pTimeCode);
	void ActionSetStop(const char *pTimeCode);
	void ActionSetRate(const char *pTimeCodeRate);
	void ActionGoto(const char *pTimeCode);
	void ActionSetDirection(const char *pTimeCodeDirection);
	void ActionSetPitch(float fTimeCodePitch);
	void ActionForward(int32_t nSeconds);
	void ActionBackward(int32_t nSeconds);
```

Basically we can adopt the same as in the OSC Server -> https://github.com/vanvught/rpidmx512/blob/master/opi_emac_ltc_smpte/lib/ltcoscserver.cpp#L190

So it could work out-of-the-box :) Just adding code into shellcmd.cpp Or do I miss something here?

Greetings, Arjan

hippyau, 2020-08-04:
Hey Arjan,

I did see I can get those functions, though the way I went about it makes the UDP commands and behavior are exactly the same as the shell commands, and processed by the same code.  I did this way to avoid duplication of code and commands, and to retain same functionality & behavior as UDP?

Happy to change, just seemed like a nice way to get there.

Cheers,
 hip

hippyau, 2020-08-04:
Confirmed internal works too:

```
Source : Internal
 LTC output is disabled
 NTP output is disabled
 Max7219 output is disabled
opi> [25] {set ltc start!12:12:12.12}
nCmdIndex=2
ValidateArg() src/h3/shell.cpp, line 174: m_Argc=2
0:[ltc]{3}
1:[start!12:12:12.12]{17}
LTC Generator Command: ltc!start!12:12:12.12
opi>
```

vanvught, 2020-08-04:
Hi Hip,

I am in favor of 

> retain same functionality & behavior as UDP

This

```
			m_Buffer[0] = 'l';
 			m_Buffer[1] = 't';
 			m_Buffer[2] = 'c';
 			m_Buffer[3] = '!';			
 			memcpy(&m_Buffer[4], sCommand, m_nBytesReceived);
 			uart0_printf("LTC Generator Command: %s\n", m_Buffer);
```

is shell specific and should take place in shellcmd.cpp

See here https://github.com/vanvught/rpidmx512/blob/experimental/lib-remoteconfig/src/remoteconfig.cpp#L470
and here https://github.com/vanvught/rpidmx512/blob/experimental/lib-remoteconfig/src/remoteconfig.cpp#L822
and https://github.com/vanvught/rpidmx512/blob/experimental/lib-remoteconfig/src/h3/shellcmd.cpp#L174

In shellcmd.cpp I construct the UPD command exactly and then pass it to the Remote Config framework.

So something like:

```
void LtcGenerator::HandleRequest(const char *pCommand, uint32_t nCommandLength) {

if (pCommand != nullptr) {		
 			memcpy(m_Buffer, pCommand, nCommandLength);		      
 
}
 	
```

The `m_nBytesReceived = Network::Get()->RecvFrom` cannot be in the `else`. This could discard the UDP message when working with the shell.

hippyau, 2020-08-04:
> This
> 
> 			m_Buffer[0] = 'l';
>  			m_Buffer[1] = 't';
>  			m_Buffer[2] = 'c';
>  			m_Buffer[3] = '!';			
>  			memcpy(&m_Buffer[4], sCommand, m_nBytesReceived);
>  			uart0_printf("LTC Generator Command: %s\n", m_Buffer);
> 
> is shell specific and should take place in shellcmd.cpp

yes okay I see that now :)

> The m_nBytesReceived = Network::Get()->RecvFrom cannot be in the else. This could discard the UDP message when working with the shell.

but it's always called anyway... when there is no argument as is run() for network in the main loop as usual... and called only an extra time with a argument from the shell input.  So both will happen in the same loop.

vanvught, 2020-08-04:
Meanwhile I have committed a new public interface. When you "move" the 

```
 	m_Buffer[0] = 'l';
  	m_Buffer[1] = 't';
  	m_Buffer[2] = 'c';
  	m_Buffer[3] = '!';
```

to shellcmd.cpp then it is working fine :)

vanvught, 2020-08-04:
This would be better:

```
void SystimeReader::HandleRequest(void *pBuffer, uint32_t nBufferLength) {
	if ((pBuffer != nullptr) && (nBufferLength <= sizeof(m_Buffer))) {
		memcpy(m_Buffer, pBuffer, nBufferLength);
	}
```

hippyau, 2020-08-04:
> Meanwhile I have committed a new public interface. When you "move" the
> 
> ```
>  	m_Buffer[0] = 'l';
>   	m_Buffer[1] = 't';
>   	m_Buffer[2] = 'c';
>   	m_Buffer[3] = '!';
> ```
> 
> to shellcmd.cpp then it is working fine :)

Okay, I just did move that now :)

vanvught, 2020-08-04:
`Is a null terminated string read by strlen(), so really need to pass length?`

There is no null terminated string in UDP messages.

hippyau, 2020-08-04:
I am only processing UDP as normal, if no null terminated string is supplied as an argument though? 
```
if (sCommand != nullptr) { // if an argument specified
		m_nBytesReceived = strlen(sCommand);
		memcpy(m_Buffer, sCommand, m_nBytesReceived);
	} else { // not a immediate command, check network
		m_nBytesReceived = Network::Get()->RecvFrom(m_nHandle, &m_Buffer, sizeof(m_Buffer), &nIPAddressFrom, &nForeignPort);
	}
```

vanvught, 2020-08-04:
Ah..I see the issue here https://github.com/vanvught/rpidmx512/blob/experimental/lib-ltc/src/h3/ltcgenerator.cpp#L511

It should be:

```
	if ((pBuffer != nullptr) && (nBufferLength <= sizeof(m_Buffer))) {
		memcpy(m_Buffer, pBuffer, nBufferLength);
		m_nBytesReceived = nBufferLength;
	}
```

I am sitting outside with no access to Zero for testing ;-)

hippyau, 2020-08-04:
Enjoy the outside :)  

And thanks putting up with me, the beginner, I am learning so much from your code ... :+1: :100: 

I'm still learning the git, github etc... but getting better it seems hopefully.

c++, I learn the mistakes :) :+1: 

I'll work on it, and always try to remember to `git pull https://github.com/vanvught/rpidmx512.git experimental`  before messing around, because I keep writing stuff and then realizing you have changed things between.

PS. F0xK COVID, so depressing.   Hope you are well friend :)

vanvught, 2020-08-07:
I have written two simple Java applications:

1. An UART <-> UDP bridge 
2. A front-end which connects to the above bridge. The front-end is shown below.  

This makes it easier debugging new features in the firmware;  no need to be around the OPi board anymore for working with the debug UART. Just all remotely. 

[image]