vanvught/rpidmx512 #130: Incorrect HandleTxtFile functions

rbarreiros, 2020-07-29:
I was having a hardtime working out the store saves in the remote config application, to find out there is a problem in the way the HandleTxtFile functions handle the struct buffer received from the network.

This issue exists in both master and experimental branch.

Check comments on the next snippet

```
void RemoteConfig::HandleTxtFileRconfig() {
        DEBUG_ENTRY

        RemoteConfigParams remoteConfigParams(StoreRemoteConfig::Get());

        DEBUG_PRINTF("m_nBytesReceived=%d, sizeof TRemoteConfigParams=%d", (m_nBytesReceived - SET_STORE_LENGTH - 11)+1, sizeof(struct TRemoteConfigParams));
#ifndef NDEBUG
        debug_dump(&m_pStoreBuffer[SET_STORE_LENGTH+11], m_nBytesReceived - SET_STORE_LENGTH - 10);
#endif

        // m_nBytesReceived is the size of the whole packet, which includes !store#rconfig.txt therefore, it'll be always greater than
        // sizeof(struct TRemoteConfigParams thus never calling remoteconfigparams.builder
        // also, the buffer sent to the remoteconfigparams can't have the !store#rconfig.txt, as it assumes it's the struct data only
        // advising for a better way of handling this ?
        // I don't know how to efficiently find the params filename size (well I know it's 11 for the rconfig.txt but, shouldn't be hardcoded!)
        // what's pUdpBuffer for ? should it be also &m_pUdpBuffer[SET_STORE_LENGTH + 11] ?

        // This pattern repeats itself through all HandleTxtFile*

        if ((m_tRemoteConfigHandleMode == REMOTE_CONFIG_HANDLE_MODE_BIN)
            && ( (m_nBytesReceived - SET_STORE_LENGTH - 11)+1 == sizeof(struct TRemoteConfigParams)))  {

                uint32_t nSize;
                remoteConfigParams.Builder(reinterpret_cast<const struct TRemoteConfigParams*>(&m_pStoreBuffer[SET_STORE_LENGTH + 11]),
                                           m_pUdpBuffer,
                                           udp::BUFFER_SIZE - (SET_STORE_LENGTH + 11),
                                           nSize);
                m_nBytesReceived = nSize;
        }

        remoteConfigParams.Load(m_pUdpBuffer, m_nBytesReceived);
#ifndef NDEBUG
        remoteConfigParams.Dump();
#endif

        DEBUG_EXIT
}
```

vanvught, 2020-07-29:
**Test application main.cpp** 

```
#include <stdio.h>
#include <string.h>
#include <stdint.h>

enum {
	REMOTE_CONFIG_DISPLAY_NAME_LENGTH = 24,
	REMOTE_CONFIG_ID_LENGTH = (32 + REMOTE_CONFIG_DISPLAY_NAME_LENGTH + 2) // +2, comma and \n
};

struct TRemoteConfigParams {
    uint32_t nSetList;
	bool bDisable;
	bool bDisableWrite;
	bool bEnableReboot;
	bool bEnableUptime;
	char aDisplayName[REMOTE_CONFIG_DISPLAY_NAME_LENGTH];
	bool bDisableRdmNetLlrpOnly;
} __attribute__((packed));

struct RemoteConfigParamsMask {
	static const uint32_t DISABLE = (1U << 0);
	static const uint32_t DISABLE_WRITE = (1U << 1);
	static const uint32_t ENABLE_REBOOT = (1U << 2);
	static const uint32_t ENABLE_UPTIME = (1U << 3);
	static const uint32_t DISPLAY_NAME = (1U << 4);
	static const uint32_t DISABLE_RDMNET_LLRP_ONLY = (1U << 5);
};

int main() {
	char buffer[128];
	strcpy(buffer, "!store#rconfig.txt");

	struct TRemoteConfigParams store;
	memset(&store, 0, sizeof(struct TRemoteConfigParams));

	//
	store.bEnableReboot = true;
	store.nSetList |= RemoteConfigParamsMask::ENABLE_REBOOT;
	//
	strncpy(store.aDisplayName, "Display", REMOTE_CONFIG_DISPLAY_NAME_LENGTH);
	store.nSetList |= RemoteConfigParamsMask::DISPLAY_NAME;

	const int cmdlen = strlen(buffer);

	memcpy(&buffer[cmdlen], &store, sizeof(struct TRemoteConfigParams));

	FILE *fp = fopen("store.bin", "wb");

	if (fp != NULL) {
		int i = fwrite(buffer, 1, cmdlen + sizeof(struct TRemoteConfigParams), fp);
		printf("%d %lu\n", i, cmdlen + sizeof(struct TRemoteConfigParams));
		fclose(fp);
	} else {
		perror("fopen");
	}

	return 0;
}
```

**Test commands:**

```
@MacBook-Air src % g++ main.cpp && ./a.out && hexdump < store.bin
51 51
0000000 21 73 74 6f 72 65 23 72 63 6f 6e 66 69 67 2e 74
0000010 78 74 14 00 00 00 00 00 01 00 44 69 73 70 6c 61
0000020 79 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0000030 00 00 00                                       
0000033
@MacBook-Air src % nc -u -v 192.168.2.120 10501 < store.bin 
```

closed 2020-07-29 by vanvught

rbarreiros, 2020-07-29:
Using the same test application, in linux, there is an endianness problem in the hexdump, although, the packet arrives correctly to orange pi, I assume NC handles the ntohs/htons still, the data is read incorrectly and all fields cleared in the remote config store on orange pi, using latest experimental (updated at the time of this test)

```
$ g++ main.cpp && ./a.out && hexdump < store.bin
51 51
0000000 7321 6f74 6572 7223 6f63 666e 6769 742e
0000010 7478 0014 0000 0000 0001 6944 7073 616c
0000020 0079 0000 0000 0000 0000 0000 0000 0000
0000030 0000 0000
0000033

$nc -u -v 192.168.1.7 10501 < store.bin
Connection to 192.168.1.7 10501 port [udp/*] succeeded!

$ echo '?version#' | nc -u -p 10501 -w 1 192.168.1.7 10501
version:[V2.7] Orange Pi Zero Compiled on Jul 29 2020 at 15:14:49

```

```
size=10
0000  3f 76 65 72 73 69 6f 6e  23 0a                    ?version #.
Run() src/remoteconfig.cpp, line 299: '\n'
Run() src/remoteconfig.cpp, line 304: ?
--> src/remoteconfig.cpp:HandleVersion:371
<-- src/remoteconfig.cpp:HandleVersion:385
size=51
0000  21 73 74 6f 72 65 23 72  63 6f 6e 66 69 67 2e 74  !store#r config.t
0010  78 74 14 00 00 00 00 00  01 00 44 69 73 70 6c 61  xt...... ..Displa
0020  79 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  y....... ........
0030  00 00 00                                          ...
Run() src/remoteconfig.cpp, line 332: !
Run() src/remoteconfig.cpp, line 340: !store#
--> src/remoteconfig.cpp:HandleTxtFile:794
--> src/remoteconfigstatic.cpp:GetIndex:70
GetIndex() src/remoteconfigstatic.cpp, line 72: nLength=1017
size=16
0000  72 63 6f 6e 66 69 67 2e  74 78 74 14 00 00 00 00  rconfig. txt.....
GetIndex() src/remoteconfigstatic.cpp, line 85: nLength=11, i=0
<-- src/remoteconfigstatic.cpp:GetIndex:86
--> src/remoteconfig.cpp:HandleTxtFileRconfig:899
        --> src/spiflashstore.cpp:Update:186
Update() src/spiflashstore.cpp, line 192: [RConfig]:12:0x40350834, nOffset=0, nDataLength=33-0, bSetList=0x0, nOffsetSetList=0
size=33
0000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
0010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
0020  00                                                .
GetStoreOffset() src/spiflashstore.cpp, line 166: nOffset=1264
Update() src/spiflashstore.cpp, line 226: m_tState=1
        <-- src/spiflashstore.cpp:Update:228
src/remoteconfigparamsdump.cpp::Dump 'rconfig.txt':
<-- src/remoteconfig.cpp:HandleTxtFileRconfig:914
<-- src/remoteconfig.cpp:HandleTxtFile:895
Flash() src/spiflashstore.cpp, line 288: m_tState=1
Flash() src/spiflashstore.cpp, line 288: m_tState=2
size=32
0000  41 76 56 10 ff ff ff ff  ff ff ff ff ff ff ff ff  AvV..... ........
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  ........ ........
```

```
Store [RConfig]:12
GetStoreOffset() src/spiflashstore.cpp, line 166: nOffset=1264
size=48
0000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
0010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
0020  00 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  ........ ........
```

vanvught, 2020-07-29:
The commit on the `experimental` branch is from a hour ago. Please check if you have https://github.com/vanvught/rpidmx512/blob/experimental/lib-remoteconfig/src/remoteconfig.cpp#L790

The store is empty as this https://github.com/vanvught/rpidmx512/blob/experimental/lib-remoteconfig/src/remoteconfig.cpp#L889 is evaluating to false. Then with `m_tRemoteConfigHandleMode = REMOTE_CONFIG_HANDLE_MODE_BIN` there is still the `remoteConfigParams.Load`
These `if `statements must be splitted.

rbarreiros, 2020-07-29:
Sorry, I was working on my repo, and failed to fetch your update into my repo... Now finally update with the latest changes (the ones you made about 2hours ago, and it's working perfectly, thanks and sorry for the dumbness !! :)

vanvught, 2020-07-29:
No worries. Thank you for reporting this issue. I am aiming for the best open source software.

commit ce6cce7f (2020-07-29): Fix #130

*. Also fixed the typo DISABLE_RDMNET_LLRP_ONY
*. Improvements in order to make it Modern C++11