vanvught/rpidmx512 #133: Follow-up for #130

vanvught, 2020-07-29:
The `if` statements as here https://github.com/vanvught/rpidmx512/blob/master/lib-remoteconfig/src/remoteconfig.cpp#L889 should be written as

```
	if (m_tRemoteConfigHandleMode == REMOTE_CONFIG_HANDLE_MODE_BIN)
             if (m_nBytesReceived == sizeof(struct TRemoteConfigParams))  {
		uint32_t nSize;
		remoteConfigParams.Builder(reinterpret_cast<const struct TRemoteConfigParams*>(m_pStoreBuffer), m_pUdpBuffer, udp::BUFFER_SIZE, nSize);
		m_nBytesReceived = nSize;
            } else {
                DEBUG_EXIT
                return;
           }
	}
```

cc: @rbarreiros

rbarreiros, 2020-07-29:
I can do that patch if you want, can have a pull ready later tonight

Rui Barreiros
Systems Designer | Producer | Mixer | Technical Director
rbarreiros@gmail.com | rbarreiros@ruibarreiros.com
https://www.ruibarreiros.com

vanvught, 2020-07-29:
> I can do that patch if you want, can have a pull ready later tonight

That would be great. Thanks!

rbarreiros, 2020-07-29:
Finished, only tested compilation

#134

vanvught, 2020-07-30:
Fixed with fc3d7ad

closed 2020-07-30 by vanvught