vanvught/rpidmx512 #137: LTC SMPTE: Implement SMPTE 2059 [PTP]

vanvught, 2020-08-01:
https://en.wikipedia.org/wiki/SMPTE_2059
https://en.wikipedia.org/wiki/Precision_Time_Protocol

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

drummerclint, 2020-08-02:
oh nice!  What do you guys want to do with PTP though?  which version do you want to implement?  Dante uses PTPv1 and Q-SYS and AES67 uses PTPv2.  If possible, add option to disable and also to change the ptp domain.  I have a bunch of dante and Q-SYS devices here at home and at the stadium so I can test for interoperability :)

rbarreiros, 2020-08-02:
Small correction, Dante software is ptpv1 only, hardware that is aes67 enabled is ptpv1 and ptpv2.

vanvught, 2020-08-02:
Hi Clint,

As I understand the PTP protocol specifications correctly, the V2 is not backwards compatible with V1. Hence I have decided to implement V2 only. 

The Orange Pi Zero (LTC SMPTE firmware)  will act as a client clock (Ordinary Clock).

As a proof-of-concept, I have just implemented the messages: SYNC (1-step and 2-step), FOLLOW_UP (1-step),  DELAY_REQ  and DELAY_RESP. And then just in the seconds precision domain. It is running fine in a Linux environment. 

Next step is moving this PoC to the OPi . There I have to implement: `gettimeofday `and `settimeofday`

The OPi PTP implementation boundaries are:

- The highest precision of the OPi system clock is microsecond.
- The OPi PHY MAC supports software time stamping only.

Yes, there will be an option for disabling PTP. For backwards compatibility, PTP will be disabled by default.  

What I need to understand yet, is how to implement the SMPTE 2059 profile. What does it exactly mean in the PTP configuration?

> also to change the ptp domain.

That is also a subject which I need to learn about.

I am eager see an interoperability with your  devices ;)

Greetings, Arjan

hippyau, 2020-08-02:
I don't really see a realistic application for integration with any
existing transport protocols like AVB or (closed source) Dante.
I mean a device that sent LTC as audio via Dante would be awesome, but I
don't think is achievable.

Used in a live setting or studio, I see a solution to tightly syncronize a
network of oPi's, with PTP.
With TC generation, by telling a distributed network of TC generators to
start presentation at an exact specific timestamp, like 10ms in the future
for example.
Maybe tight enough that even the LTC audio samples between network nodes
would be closely synchronized.

Maybe can sync ws28xx nodes leds too, or is that handled?

vanvught, 2020-08-02:
> Maybe can sync ws28xx nodes leds too, or is that handled?

Both the sACN E1.31 and Art-Net protocols have their full implementation in my firmware. Therefore also the synchronization specification. When using proper software/desk, that is using sACN E1.31 ;), then there is perfect synchronization for any output (DMX or WS28xx).

vanvught, 2020-08-02:
What I am just thinking now, a nice to have ;); that my embedded showfile player can be synchronized started with PTP (or any network SMTPE). But still, there are just 24 hrs per day, and it is just hobby time :)

hippyau, 2020-08-02:
>> But still, there are just 24 hrs per day, and it is just hobby time :)

so true :)   If we do get PTP on the oPi,  it will get used :)  if not, no
stress :)

drummerclint, 2020-08-02:
> I don't really see a realistic application for integration with any existing transport protocols like AVB or (closed source) Dante. I mean a device that sent LTC as audio via Dante would be awesome, [...]

@hippyau I don't really see an application either (Audinate is starting to support software based dante but that still requires licensing etc) but the concern I had was when both types of devices co exist on the same network, how do their ptp clocks do their thing without disturbing the other group?  Brightsign also uses ptp to sync their video wall modes and apparently it's known to cause issues with Q-SYS qlan so we keep them on separate vlans at the stadium.  

I love the idea of having a series of oPi's tightly sync'd up.  I can see different aspects of live production each having an opi in different areas of a venue in super tight sync.   

I'm not much of a programmer but I admire the work that you guys do! :) 
@vanvught I don't see any issues in that test build you sent us.

drummerclint, 2020-08-02:
> Small correction, Dante software is ptpv1 only, hardware that is aes67 enabled is ptpv1 and ptpv2.

@rbarreiros You're right! My mistake, I'm used to pre aes67 dante. ahaha

vanvught, 2020-08-04:
With reference to : https://www.smpte.org/sites/default/files/users/user27446/IEEE%201588%20%26%20SMPTE%20Montreal%202016_02_09.pdf

The SMPTE ST-2059-2 profile specifies: Domain: 127 [0..127]
That domain number is application specific and IPv4 multicast ip can be: 224.0.1.130, 131 or 132 based on hash of sub-domain?

Rui? Clint? ;)

vanvught, 2020-10-09:
With reference to : https://www.smpte.org/sites/default/files/users/user27446/IEEE%201588%20%26%20SMPTE%20Montreal%202016_02_09.pdf

The SMPTE ST-2059-2 profile specifies: Domain: 127 [0..127]
That domain number is application specific and IPv4 multicast ip can be: 224.0.1.130, 131 or 132 based on hash of sub-domain?

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87