vanvught/rpidmx512 #140: Question:  H3 SPI CS

hippyau, 2020-08-07:
Hi Arjan,
  I could only find reference to CS0 in the h2+ datasheet, as pin PA13... are the others listed here mapped to accessible oPi0 pins, or are they for H3 or something else?  Assuming if I needed more CS pins, i would be using H3_SPI_CS_NONE and doing it myself?

```
typedef enum H3_SPI_CS {
	H3_SPI_CS0 = 0,		///< Chip Select 0
	H3_SPI_CS1 = 1,		///< Chip Select 1
	H3_SPI_CS2 = 2,		///< Chip Select 2
	H3_SPI_CS3 = 3,		///< Chip Select 3
	H3_SPI_CS_NONE = 4	///< No CS, control it yourself
} h3_spi_chip_select_t;
```

Cheers :)

hippyau, 2020-08-07:
Related :) 

`#include "h3_spi.h"	// TODO Replace with hal_spi.h ?`

I don't see a function in the HAL to do a synchronous RW, or mistaken and does this return read over the *pData output data?
```
	void WriteRead(char *pData, uint32_t nLength, const bool bDoSetup = true) {
		if (bDoSetup) {
			Setup();
		}
		FUNC_PREFIX(spi_transfern(pData, nLength));
	} 
```

vanvught, 2020-08-07:
Hi Hippy,

> Assuming if I needed more CS pins, i would be using H3_SPI_CS_NONE and doing it myself?

Yes, see example here -> 
https://github.com/vanvught/rpidmx512/blob/master/lib-ws28xx/src/ws28xxmulti8x.cpp#L39
and here -> 
https://github.com/vanvught/rpidmx512/blob/master/lib-l6470/src/slushmotor.cpp#L125

> does this return read over the *pData output data?

Yes, it does.

> EDIT: Actually this is NYI for H3, I will just use the h3_spi.h api?

It is used here -> https://github.com/vanvught/rpidmx512/blob/e7acdfe7f6f9414818db7f021aca5f669b7a7824/lib-device/include/bw.h#L107

It is best to use the HAL SPI API's moving forward.

Groet, Arjan

hippyau, 2020-08-07:
Awesome, Thank you!

closed 2020-08-07 by hippyau

vanvught, 2020-08-07:
Yw! 

Somehow 'funny' , the examples given are not using the HAL SPI API. And the HAL SPI does not really work with CS_NONE. So better to use the function prefix method as shown in the examples ;)