vanvught/rpidmx512 #143: undefined reference to `memcpy'?

hippyau, 2020-08-13:
Hey Arjan,

No stress, i'm messing around, but I am getting the weirdest thing... Well I don't understand anyway.   

I know it's 3rd party, but I am trying some code from the project [microui](https://github.com/rxi/microui) which is a tiny, (C but reports C++ compilable) like 1K lines of code, it's a _very_ basic window and widget framework.  It's very lightweight, only needs memcpy, memset, strtod & qsort.  and i've read through all the code and can't see any calls that raise suspicion.  

I've had to implement `strtod` and `qsort` with some codes I found, and read them carefully and they seem all good too.

I have worked hard, and have gotten down to just this error from so many, where it can't find memcpy when linking :)
I've tried including string.h, and many other things, all day.

```
[/home/hip/dev/new/rpidmx512/lib-arm]
$DEFINES [-DORANGE_PI_ONE  -DNDEBUG -D_TIME_STAMP_YEAR_=2020 -D_TIME_STAMP_MONTH_=8 -D_TIME_STAMP_DAY_=13]
$MAKE_FLAGS [-DCONSOLE_FB]
$BUILD_DIRS [build_h3/src build_h3/src/h3]
$TARGET [lib_h3/libarm.a ]
arm-none-eabi-ar: creating lib_h3/libarm.a
arm-none-eabi-ld: build_h3/lib/microui.o: in function `mu_init(mu_Context*)':
microui.cpp:(.text+0x144): undefined reference to `memcpy'
arm-none-eabi-ld: build_h3/lib/microui.o: in function `push_layout(mu_Context*, mu_Rect, mu_Vec2)':
microui.cpp:(.text+0x1848): undefined reference to `memcpy'
make: *** [../h3-firmware-template/Rules.mk:221: build_h3/main.elf] Error 1
```

and in the map file, there are only references to `h3_memcpy`. 

I change every call to ui_memcpy() and implement that in the microui.cpp file, and it still dies here looking for memcpy? 

I have a branch in my fork here [https://github.com/hippyau/rpidmx512/tree/experimental-ui](https://github.com/hippyau/rpidmx512/tree/experimental-ui)

Any ideas?

rewolff, 2020-08-13:
Some memcpy calls are generated by the compiler. 

```
struct somestruct a, b ;

b = a;
```
Maybe more. Best course of action would be to provide a memcpy. If you're not calling it you can hope it doesn't get called too often, so an implementation would only be a few lines if you don't care about performance.

vanvught, 2020-08-13:
Hi Hippy,

`memcpy` is defined in `string.h`

[microui.c](https://github.com/rxi/microui/blob/213dba41ae0a478be6398d66d7b110b6325a9d02/src/microui.c#L25) does include <string.h> 

So I am not sure why do you get this error.

The `h3_memcpy` should be used in low-level routines only -> lib-h3

As it is all ANSI-C , then leave the extension of the files just with `*.c`

And make sure that you have 

```
#ifdef __cplusplus
extern "C" {
#endif

....

#ifdef __cplusplus
}
#endif
```

around the C prototypes. 

> Some memcpy calls are generated by the compiler.

Please be aware of the global settings: `COPS+=-nostartfiles -ffreestanding -nostdinc -nostdlib`

Greetings, Arjan

vanvught, 2020-08-13:
> Some memcpy calls are generated by the compiler.
> 
> struct somestruct a, b ;
> 
> b = a;
> Maybe more.

@rewolff That would be in C++ only. Right?

vanvught, 2020-08-13:
> (C but reports C++ compilable)

Well... it is missing 
```
#ifdef __cplusplus
extern "C" {
#endif
```

 in https://github.com/hippyau/rpidmx512/blob/experimental-ui/scratch-h3-draw/include/microui.h#L223
```
#ifdef __cplusplus
}
#endif
```

hippyau, 2020-08-14:
Thanks so much guys!

> Well... it is missing

@vanvught Thank you, I've wrapped in extern "C" for correctness, though that didn't solve the missing memcpy mystery for me today.

> Some memcpy calls are generated by the compiler.

@rewolff  mind-blown... thank you, that must be what is happening as I removed all visible calls to memcpy to test and it still complained.

> Best course of action would be to provide a memcpy.

Yes I did try this, but compiler complained it's a redefinition of existing memcpy :(  

I then realized that Arjan's memcpy() is declared inline, so perhaps is not seen externally by linker for these compiler generated memcpy calls maybe?  

In strings.h
```
inline static void* memcpy(void *__restrict__ dest, const void *__restrict__ src, size_t n) {
```

I had to change to get it available publicly and seen by the presumed compiler generated memcpy calls...

```
__attribute__((weak)) extern void* memcpy(void *__restrict__ dest, const void *__restrict__ src, size_t n) {
```
^^^ this compiles and links properly ^^^

I know Arjan won't like this because it's no longer inline and already used everywhere in his firmwares, so I try to inline it...

```
../include/string.h:56:43: error: inline function 'void* memcpy(void*, const void*, size_t)' declared weak [-Werror=attributes]
   56 | __attribute__((weak)) extern inline void* memcpy(void *__restrict__ dest, const void *__restrict__ src, size_t n) {
```

But if I don't use `__attribute__((weak))` it will not link.   I stumbled on this weak attribute while I was researching how to override existing function with my local one... ie. put my own implementation of memcpy in microui.cpp and have compiler not complain about redefinition.

vanvught, 2020-08-14:
Hi Hippy, 

This is great information. Thanks!

I am eager to see where the `memcpy()` is generated in the source code. In 1000's of lines I have written so far, I've never seen it happen ;) Will you attach the .list file ?

When you compile this code with `Os` , then the compiler might not in-line the `memcpy()`. Just check the *.params.cpp files  with their entries in the .list files. There are the visible `string.h` functions. Such as `strcasecmp`. 

Just add 

```
#if !defined(__clang__)	// Needed for compiling on MacOS
# pragma GCC push_options
# pragma GCC optimize ("Os")
#endif
```
at the top. And do not try to compile on MacOS ;)

hippyau, 2020-08-14:
Hey Arjan :)

> In 1000's of lines I have written so far, I've never seen it happen ;) 

Trust me to find a weird ass bug! hehehe... story of my life :) 

So I tried adding the GCC push_options as you suggested above, in the microui.cpp, and it still seems to call in memcpy.

I think this is the code and assembly here that shows the memcpy, there is another in `push_layout`...  it does look like structure copy?

```
build_h3/lib/microui.o: In function `mu_init':
microui.cpp:(.text+0xa8): undefined reference to `memcpy'
```

{microui.cpp}
```
void mu_init(mu_Context *ctx) {
  memset(ctx, 0, sizeof(*ctx));
  ctx->draw_frame = draw_frame;
  ctx->_style = default_style;
  ctx->style = &ctx->_style;
}
```

{orangepi_one.list} At the end can see a `<memcpy>`...   (interesting seems call to `memset(ctx,0,sizeof(*ctx))` is inlined?)
```
400016a8 <mu_init>:
400016a8:	e3011ccb 	movw	r1, #7371	; 0x1ccb
400016ac:	e3a02000 	mov	r2, #0
400016b0:	e92d4070 	push	{r4, r5, r6, lr}
400016b4:	e1a04000 	mov	r4, r0
400016b8:	e3401004 	movt	r1, #4
400016bc:	e2800020 	add	r0, r0, #32
400016c0:	f5d0f002 	pld	[r0, #2]
400016c4:	e2411020 	sub	r1, r1, #32
400016c8:	e5402020 	strb	r2, [r0, #-32]	; 0xffffffe0
400016cc:	e351000b 	cmp	r1, #11
400016d0:	e540201f 	strb	r2, [r0, #-31]	; 0xffffffe1
400016d4:	e1a03000 	mov	r3, r0
400016d8:	e540201e 	strb	r2, [r0, #-30]	; 0xffffffe2
400016dc:	e2800020 	add	r0, r0, #32
400016e0:	e540203d 	strb	r2, [r0, #-61]	; 0xffffffc3
400016e4:	e540203c 	strb	r2, [r0, #-60]	; 0xffffffc4
400016e8:	e540203b 	strb	r2, [r0, #-59]	; 0xffffffc5
400016ec:	e540203a 	strb	r2, [r0, #-58]	; 0xffffffc6
400016f0:	e5402039 	strb	r2, [r0, #-57]	; 0xffffffc7
400016f4:	e5402038 	strb	r2, [r0, #-56]	; 0xffffffc8
400016f8:	e5402037 	strb	r2, [r0, #-55]	; 0xffffffc9
400016fc:	e5402036 	strb	r2, [r0, #-54]	; 0xffffffca
40001700:	e5402035 	strb	r2, [r0, #-53]	; 0xffffffcb
40001704:	e5402034 	strb	r2, [r0, #-52]	; 0xffffffcc
40001708:	e5402033 	strb	r2, [r0, #-51]	; 0xffffffcd
4000170c:	e5402032 	strb	r2, [r0, #-50]	; 0xffffffce
40001710:	e5402031 	strb	r2, [r0, #-49]	; 0xffffffcf
40001714:	e5402030 	strb	r2, [r0, #-48]	; 0xffffffd0
40001718:	e540202f 	strb	r2, [r0, #-47]	; 0xffffffd1
4000171c:	e540202e 	strb	r2, [r0, #-46]	; 0xffffffd2
40001720:	e540202d 	strb	r2, [r0, #-45]	; 0xffffffd3
40001724:	e540202c 	strb	r2, [r0, #-44]	; 0xffffffd4
40001728:	e540202b 	strb	r2, [r0, #-43]	; 0xffffffd5
4000172c:	e540202a 	strb	r2, [r0, #-42]	; 0xffffffd6
40001730:	e5402029 	strb	r2, [r0, #-41]	; 0xffffffd7
40001734:	e5402028 	strb	r2, [r0, #-40]	; 0xffffffd8
40001738:	e5402027 	strb	r2, [r0, #-39]	; 0xffffffd9
4000173c:	e5402026 	strb	r2, [r0, #-38]	; 0xffffffda
40001740:	e5402025 	strb	r2, [r0, #-37]	; 0xffffffdb
40001744:	e5402024 	strb	r2, [r0, #-36]	; 0xffffffdc
40001748:	e5402023 	strb	r2, [r0, #-35]	; 0xffffffdd
4000174c:	e5402022 	strb	r2, [r0, #-34]	; 0xffffffde
40001750:	e5402021 	strb	r2, [r0, #-33]	; 0xffffffdf
40001754:	1affffd9 	bne	400016c0 <mu_init+0x18>
40001758:	e283100c 	add	r1, r3, #12
4000175c:	e3a02000 	mov	r2, #0
40001760:	e4c32001 	strb	r2, [r3], #1
40001764:	e1530001 	cmp	r3, r1
40001768:	1afffffc 	bne	40001760 <mu_init+0xb8>
4000176c:	e30236cc 	movw	r3, #9932	; 0x26cc
40001770:	e284500c 	add	r5, r4, #12
40001774:	e3443000 	movt	r3, #16384	; 0x4000
40001778:	e3021634 	movw	r1, #9780	; 0x2634
4000177c:	e3441001 	movt	r1, #16385	; 0x4001
40001780:	e1a00005 	mov	r0, r5
40001784:	e5843008 	str	r3, [r4, #8]
40001788:	e3a0205c 	mov	r2, #92	; 0x5c
4000178c:	ebffff05 	bl	400013a8 <memcpy>
40001790:	e5845068 	str	r5, [r4, #104]	; 0x68
40001794:	e8bd8070 	pop	{r4, r5, r6, pc}

```

Whole thing included here :)
[orangepi_one.list.gz](https://github.com/vanvught/rpidmx512/files/5074153/orangepi_one.list.gz)

hippyau, 2020-08-14:
This is the culprit...

'ctx->_style = default_style;`

if I return before this statement (to avoid -Werror=unused-variable) it compiles with with no memcpy.

```
void mu_init(mu_Context *ctx) {
  memset(ctx, 0, sizeof(*ctx));
  ctx->draw_frame = draw_frame;
  return;
  ctx->_style = default_style;
  ctx->style = &ctx->_style;
}
```

vanvught, 2020-08-14:
> Op 14 aug. 2020 om 12:59 heeft Hippy <notifications@github.com> het volgende geschreven:
> 
> ﻿
> So I tried adding the GCC push_options as you suggested above, in the microui.cpp, and it still seems to call in memcpy.
> 

Yes, it would still call memcpy. However with size optimalization and memcpy is called multiple times, there would be no in-line. Hence the linker should find the generated memcpy by the compiler.

hippyau, 2020-08-14:
a quick hack i'm using for now...

```
#if !defined (NO_INLINE_MEMCPY)
inline static void* memcpy(void *__restrict__ dest, const void *__restrict__ src, size_t n) 
#else	
__attribute__((weak)) extern void* memcpy(void *__restrict__ dest, const void *__restrict__ src, size_t n) 
#endif
```

`
#define NO_INLINE_MEMCPY
`

hippyau, 2020-08-14:
> Yes, it would still call memcpy. However with size optimalization and memcpy is called multiple times, there would be no in-line. Hence the linker should find the generated memcpy by the compiler.

Well I put it back in and it seems to generate exact same size, so I could be doing it wrong because seems no change.  
Is okay to put these in microui.cpp as is preprocessor directive, or need to go in Rules.mk?

```
build_h3/main.elf  :
section        size         addr
.init           356   1073741824
.text         73664   1073742192
.rodata        3864   1073815856
.ARM.exidx        8   1073819720
.data           104   1073819728
.bss         140748   1073823744
Total        218744
```

hippyau, 2020-08-14:
Hey, just a thought...  because it's C code causing the memcpy problem

```
#if !defined(__clang__)	// Needed for compiling on MacOS
# pragma GCC push_options
# pragma GCC optimize ("Os")
#endif
```
does this work with C++ only?

vanvught, 2020-08-14:
> does this work with C++ only?

That compiler option is for both C and C++.

Alternatively, you can provide your own `<string.h>`

The `include` directory in the project is the first entry  in the `-I` list

hippyau, 2020-08-14:
All good, thanks so much for your help @vanvught and @rewolff !!

closed 2020-08-14 by hippyau

hippyau, 2020-08-16:
Wow, this one is killing me :)  It's not entirely a memcpy problem, but some compiler issue?

`
cmd->text.pos = pos;
`

So this one is causing a "Data abort at address: 0x400002fc0"  

It's not using memcpy, instead it's doing this... and failing badly.
```
40002fac:	e5808018 	str	r8, [r0, #24]
40002fb0:	e1a0c000 	mov	ip, r0
40002fb4:	e280e008 	add	lr, r0, #8
40002fb8:	e3560000 	cmp	r6, #0
40002fbc:	e894000f 	ldm	r4, {r0, r1, r2, r3}
40002fc0:	e88e000f 	stm	lr, {r0, r1, r2, r3}
```

whereas if I break up, it seems to pass...
```
cmd->text.pos.x = pos.x;
cmd->text.pos.y = pos.y;
```  

And doing this...

```
//cmd->clip.rect = rect;
memcpy(&cmd->clip.rect, &rect, sizeof(rect));  //  it this equivalent to the assignment above, just slower?
```

This seems like the compiler, (for this platform?), doesn't know how to do this structure copy thing?   

I don't really know why,  is it not a normal kind of operation, or has something to do with platform memory management?

Will I just rewrite 200 statements to get it working, or you geniuses @vanvught @rewolff might have an idea what is going on here?  :)

vanvught, 2020-08-16:
> "Data abort at address: 0x400002fc0"

https://www.keil.com/support/docs/3080.htm

vanvught, 2020-08-16:
>  has something to do with platform memory management?

Not really. It is the code that is trying to access a wrong memory address. 

Can be that memory is not properly allocated: static or dynamically. Just check the address location for `cmd`

hippyau, 2020-08-16:
Hey Arjan,

 I don't want to bother you from making good code. 

 It's just so weird... Or more correct, I just don't understand.   The memory is there, it looks like it should work, but it doesn't.

**I am giving up on this,** so don't worry about looking into...  I think it's not good mixing this C code with lots of pointers to structures inside structures and such.  I no have the time to rewrite... not worth it.

ctx is the UI context, is a big structure with other embedded structures too.  In main.cpp, before using...
`	mu_Context *ctx = static_cast<mu_Context *>(malloc(sizeof(mu_Context)));`
at runtime, I printf...
` ctx @ 40500000c, size 269516`  seems legit.
cmd is a pointer to a structure inside ctx...
`     mu_Command *cmd = (mu_Command*) (ctx->command_list.items + ctx->command_list.idx);`
and it printf's as...
` cmd @ 40500133, size 32` which also seems legit.

but still crashes out at  0x40003348