vanvught/rpidmx512 #145: LTC SMPTE: Add support for GPS time

vanvught, 2020-08-28:
Implementation based on UART modules. For example: ATGM336H 

- [ ] Initial commit -> af5d0e6
- [ ] Change baudrate to 115200
- [ ] Use PPS; Which GPIO pin on Orange Pi Zero? -> http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/gpio-header
- [ ] Add gps.txt configuration options. Example: utc_offset

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

hippyau, 2020-08-28:
Awesome, I've ordered a module for testing.

vanvught, 2020-08-29:
[image]

[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/5145510/RemoteConfigUI.jar.zip)

vanvught, 2020-09-10:
When `source=systime` then the OLED will show the reference time source: `GPS`, `NTP` or `RTC` (and near future `PTP` #137). 

[image]

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

vanvught, 2020-09-13:
Dear all, please find attached the V2.8 with the new System Time framework and the GPS Time support. I hope you will give it a good test. Many thanks, Arjan

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/5213923/opi_emac_ltc_smpte.zip)

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

hippyau, 2020-09-13:
Nice work Arjan! Hoping my GPS module arrives in the next few days.

Legend!

hippyau, 2020-09-26:
GPS RX <-- oPi UART1 TX Pin 8  
GPS TX --> oPi UART1 RX Pin 10
GPS PPS --> oPi GPIO Pin 18 
GPS GND to oPi GND Pin 6 (or 9,14,20,25)
GPS VCC <-- oPi +V (3.3V on Pin 1 or 17,  5.0V on Pin 2 or 4)

[image]

hippyau, 2020-09-26:
What should I expect?  I get on uart:

--> src/gps.cpp:GPS:74
<-- src/gps.cpp:GPS:84

and nothing more of interest?

```
U-Boot 2018.09 (Mar 02 2019 - 08:59:42 +0100) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Zero
DRAM:  256 MiB
MMC:   SUNXI SD/MMC: 0
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
USB2:   USB EHCI 1.00
USB3:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
scanning bus 2 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
110622 bytes read in 7 ms (15.1 MiB/s)
## Booting kernel from Legacy Image at 48000000 ...
   Image Name:   http://www.orangepi-dmx.org
   Image Type:   ARM U-Boot Standalone Program (gzip compressed)
   Data Size:    110558 Bytes = 108 KiB
   Load Address: 40000000
   Entry Point:  40000000
   Verifying Checksum ... OK
EHCI failed to shut down host controller.
EHCI failed to shut down host controller.
   Uncompressing Standalone Program ... OK
--> rtc/hwclock.cpp:HwClock:46
--> rtc/hwclockrtc.cpp:RtcProbe:66
<-- rtc/hwclock.cpp:HwClock:55
No RTC connected
--> rtc/hwclock.cpp:HcToSys:76
<-- rtc/hwclock.cpp:HcToSys:78
MX25L1605D, sector size 4096, 2097152 bytes
uboot.spi
Check difference
Writing
470158 bytes written
Done
uImage
Check difference
Writing
110622 bytes written
Done
Detected MX25L1605D with sector size 4096 total 2097152 bytes
No display found
[V2.8] Orange Pi Zero Compiled on Sep 13 2020 at 11:39:36
LTC SMPTE
Network init
Network
 Hostname  : allwinner_420913
 If        : 1: eth0
 Inet      : 169.254.66.9/16
 Netmask   : 255.255.0.0
 Gateway   : 169.254.66.9
 Broadcast : 169.254.255.255
 Mac       : 02:42:3c:42:09:13
 Mode      : Z
NTP v4 Client
 Not enabled
MAX7219
 matrix [4]
Configuring Art-Net
Starting Art-Net
Node 3
 Firmware   : 1.46
 Short name : LTC SMPTE Node
 Long name  : Orange Pi Zero Art-Net 3 www.orangepi-dmx.org
Art-Net started
MIDI
 Direction    : Output
 Active sense : Disabled
 Baudrate     : 31250 (Default)
 Interface    : UART2
mDNS
 Name : allwinner_420913.local
 _apple-midi._udp.local 5004 
 _config._udp.local 10501 
AppleMIDI
 SSRC    : a9fe4209 (2852012553)
 Session : allwinner_420913
--> src/gps.cpp:GPS:74
<-- src/gps.cpp:GPS:84
RDMNet configuration
 CID : 02c00142-8470-4620-b894-06183c420913
LLRP Device configuration
 Port UDP               : 5569
 Multicast join Request : 239.255.250.133
 Multicast Response     : 239.255.250.134
RDM Device configuration
 Manufacturer Name : www.orangepi-dmx.org
 Manufacturer ID   : 5000
 Serial Number     : 3C420913
 Root label        : LTC SMPTE
 Product Category  : 0801
 Product Detail    : 0601
RDM Responder configuration
 Protocol Version 1.0
 DMX Address      : 65535
 DMX Footprint    : 0
 Personality 1 of 1 [RDMNet LLRP Only]
 Sub Devices      : 0
 Sensors          : 0
Source : LTC
 LTC output is disabled
 NTP output is disabled
 WS28xx output is disabled
```

vanvught, 2020-09-26:
> Source : LTC

The source should be: `systime`

vanvught, 2020-09-26:
https://github.com/vanvught/rpidmx512/blob/experimental/opi_emac_ltc_smpte/firmware/main.cpp#L336

```
/**
	 * The GPS Time client is running when enabled AND source = System-Time
	 * The NTP Client is stopped.
	 */
```

hippyau, 2020-09-26:
Thanks, 100% sure my ltc.txt is... 

Should see some debug messages?  GPS sentences?

hip@hip-Alienware-13:/media/hip/OPI_ZERO$ cat ltc.txt
#ltc.txt
#source=ltc
# Disable outputs #
#disable_display=0
#disable_max7219=0
#disable_ltc=0
#disable_midi=0
#disable_artnet=0
#disable_tcnet=0
#disable_rtp-midi=0
# System clock / RTC #
show_systime=1
#disable_timesync=0

source=systime 
auto_start=1

# source=internal #

fps=30

#start_hour=0
#start_minute=0
#start_second=0
#start_frame=0
#stop_hour=23
#stop_minute=29
#stop_second=59
#stop_frame=24
# NTP Server #
#ntp_enable=0
#year=20
#month=4
#day=6
# OSC Server #
#osc_enable=0
#osc_port=8000
# WS28xx display #
#ws28xx_enable=0

hippyau, 2020-09-26:
gps.txt

`
hip@hip-Alienware-13:/media/hip/OPI_ZERO$ cat gps.txt
#gps.txt
enable=1
utc_offset=10.0
`

vanvught, 2020-09-26:
> utc_offset=10.0

That is NL local time 16:12 -2 + 10 = + 8 that's midnight ;-)

hippyau, 2020-09-26:
yes it's 12:24am here right now :)

I'm just hooking up scope now...  does GPS modules send stuff without command?

vanvught, 2020-09-26:
> Should see some debug messages? GPS sentences?

There is some debug information here -> https://github.com/vanvught/rpidmx512/blob/experimental/lib-gps/src/gps.cpp#L167

You can also uncomment here -> https://github.com/vanvught/rpidmx512/blob/experimental/lib-gps/src/gps.cpp#L194

hippyau, 2020-09-26:
Thanks, i'll keep pokeing.... :)  <3

vanvught, 2020-09-26:
> oes GPS modules send stuff without command?

Yes, there is a serial stream every second. When there is a single sat , then it can just be the time. All other fields are empty.

vanvught, 2020-09-26:
@hippyau  What is your GPS module? 

I have just received another one; a u-blox NEO 6M V2 
So next is trying to get this one at 115200 baud -> https://github.com/vanvught/rpidmx512/blob/experimental/lib-gps/src/gpsconst.cpp#L34

hippyau, 2020-09-26:
Awesome.   can see stuff happening on the scope, but I now realize changes to .txt files on the SD card are not maybe imported(?)  if the firmware is not replaced... duh!  

I follow up tomorrow, very sleepy now :)

hippyau, 2020-09-26:
I have [a ATGM336H](https://www.ebay.com.au/itm/1PC-ATGM336H-GPS-Beidou-dual-mode-module-with-EEPROM-instead-of-NEO-M8N/303417356649?)

vanvught, 2020-09-26:
> I have a ATGM336H

```
gps.txt
module=ATGM336H
```

then the module will switch to 115200 baud

https://github.com/vanvught/rpidmx512/blob/experimental/lib-gps/src/gpsconst.cpp#L29

vanvught, 2020-09-26:
> but I now realize changes to .txt files on the SD card are not maybe imported(?) if the firmware is not replaced... duh!

Using SDCard ? ;-) Much much (much) easier using TFTP and the remote configuration framework

hippyau, 2020-09-26:
is weird, I can see module:

```
[V2.8] Orange Pi Zero Compiled on Sep 13 2020 at 11:39:36
LTC SMPTE
Network init
Network
 Hostname  : allwinner_420913
 If        : 1: eth0
 Inet      : 192.168.1.122/24
 Netmask   : 255.255.255.0
 Gateway   : 192.168.1.1
 Broadcast : 192.168.1.255
 Mac       : 02:42:3c:42:09:13
 Mode      : D
```

ping....
```
hip@hip-Alienware-13:~/OrangePi$ ping 192.168.1.122
PING 192.168.1.122 (192.168.1.122) 56(84) bytes of data.
64 bytes from 192.168.1.122: icmp_seq=1 ttl=64 time=0.207 ms
64 bytes from 192.168.1.122: icmp_seq=2 ttl=64 time=0.232 ms
64 bytes from 192.168.1.122: icmp_seq=3 ttl=64 time=0.227 ms
64 bytes from 192.168.1.122: icmp_seq=4 ttl=64 time=0.450 ms
64 bytes from 192.168.1.122: icmp_seq=5 ttl=64 time=0.200 ms
^C
--- 192.168.1.122 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4075ms
rtt min/avg/max/mdev = 0.200/0.263/0.450/0.094 ms

```

run the latest java posted in this thread...
```

hip@hip-Alienware-13:~/OrangePi$ sudo ./run.sh
Linux
enp2s0
	/fe80:0:0:0:f10f:a22d:2bf7:9a73%enp2s0 false false false
	/192.168.1.49 false false false
Local ip: 192.168.1.49
Interface address: 192.168.1.49
No more messages.
No more messages.
No more messages.
```

does not show up in the latest remote config jar from this thread... 
[image]

hippyau, 2020-09-26:
This is my problem.....
(fresh uboot flash and latest uImage)

I change gps.txt and ltc.txt on sd card...  does not get imported to SPI.

I can't see node appear in latest remote java config manager,

stuck with LTC as input, can't change to systime.

Play more in the PM part of today, is 1:10am here :) :+1: 

Cheers,

hippyau, 2020-09-26:
Problem is my end, something gone wrong... can't get older java config version to connect to existing wall clock here.   I'll work out, don't worry... my issue.

hippyau, 2020-09-27:
Hey Arjan,

  I have all the configuration correct I believe,  but have a boot loop with no error messages.  I will try to compile myself and see what is going on, presume the watchdog is tripping over, so maybe I have not wired correctly, buy pretty sure I have.

I can see some stuff, that changes between boots....

```
AppleMIDI
 SSRC    : c0a8017a (3232235898)
 Session : allwinner_420913
--> src/gps.cpp:GPS:74
<-- src/gps.cpp:GPS:84
unbind--> src/gps.cpp:Start:153
0x400343cc |$GPGSV,1,1,00*79|
[76]
<-- src/gps.cpp:Start:186
GPS
 Module : ATGM336H [115200]
 UTC offset : 36000 (seconds)
 Idle
 UART: 1
RDMNet configuration

```

```
AppleMIDI
 SSRC    : c0a8017a (3232235898)
 Session : allwinner_420913
--> src/gps.cpp:GPS:74
<-- src/gps.cpp:GPS:84
unbind--> src/gps.cpp:Start:153
0x400343cc |$GNGGA,,,,,,0,00,25.5,,,,,,*64|
[760]
<-- src/gps.cpp:Start:186
GPS
 Module : ATGM336H [115200]
 UTC offset : 36000 (seconds)
 Idle
 UART: 1
RDMNet configuration

```

```

U-Boot 2017.03-rc2-00252-gc987c17120-dirty (Oct 26 2019 - 15:37:55 +0200) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Zero
DRAM:  256 MiB
MMC:   SUNXI SD/MMC: 0
spi_flash_decode_fdt: Cannot decode address
SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
spi_flash_decode_fdt: Cannot decode address
SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
device 0 offset 0x180000, size 0x22000
SF: 139264 bytes @ 0x180000 Read: OK
## Booting kernel from Legacy Image at 48000000 ...
   Image Name:   http://www.orangepi-dmx.org
   Image Type:   ARM U-Boot Standalone Program (gzip compressed)
   Data Size:    110558 Bytes = 108 KiB
   Load Address: 40000000
   Entry Point:  40000000
   Verifying Checksum ... OK
   Uncompressing Standalone Program ... OK
--> rtc/hwclock.cpp:HwClock:46
--> rtc/hwclockrtc.cpp:RtcProbe:66
<-- rtc/hwclock.cpp:HwClock:55
No RTC connected
--> rtc/hwclock.cpp:HcToSys:76
<-- rtc/hwclock.cpp:HcToSys:78
MX25L1605D, sector size 4096, 2097152 bytes
Detected MX25L1605D with sector size 4096 total 2097152 bytes
No display found
[V2.8] Orange Pi Zero Compiled on Sep 13 2020 at 11:39:36
LTC SMPTE
Network init
Network
 Hostname  : allwinner_420913
 If        : 1: eth0
 Inet      : 192.168.1.122/24
 Netmask   : 255.255.255.0
 Gateway   : 192.168.1.1
 Broadcast : 192.168.1.255
 Mac       : 02:42:3c:42:09:13
 Mode      : D
NTP v4 Client
 Not enabled
MAX7219
 matrix [4]
Configuring Art-Net
Starting Art-Net
Node 3
 Firmware   : 1.46
 Short name : LTC SMPTE Node
 Long name  : Orange Pi Zero Art-Net 3 www.orangepi-dmx.org
Art-Net started
MIDI
 Direction    : Output
 Active sense : Disabled
 Baudrate     : 31250 (Default)
 Interface    : UART2
mDNS
 Name : allwinner_420913.local
 _apple-midi._udp.local 5004 
 _config._udp.local 10501 
AppleMIDI
 SSRC    : c0a8017a (3232235898)
 Session : allwinner_420913
--> src/gps.cpp:GPS:74
<-- src/gps.cpp:GPS:84
unbind--> src/gps.cpp:Start:153
<-- src/gps.cpp:Start:186
GPS
 Module : ATGM336H [9600]
 UTC offset : 36000 (seconds)
 Idle
 UART: 1
RDMNet configuration
 CID : 02c00142-8470-4620-b894-06183c420913
LLRP Device configuration
 Port UDP               : 5569
 Multicast join Request : 239.255.250.133
 Multicast Response     : 239.255.250.134
RDM Device configuration
 Manufacturer Name : www.orangepi-dmx.org
 Manufacturer ID   : 5000
 Serial Number     : 3C420913
 Root label        : LTC SMPTE
 Product Category  : 0801
 Product Detail    : 0601
RDM Responder configuration
 Protocol Version 1.0
 DMX Address      : 65535
 DMX Footprint    : 0
 Personality 1 of 1 [RDMNet LLRP Only]
 Sub Devices      : 0
 Sensors          : 0
Source : System-Time
 NTP output is disabled
 WS28xx output is disabled

{{ sits here for a couple of seconds... }}

U-Boot SPL 2017.03-rc2-00252-gc987c17120-dirty (Oct 26 2019 - 15:37:55)
DRAM: 256 MiB
Trying to boot from sunxi SPI

```

vanvught, 2020-09-27:
Hi Hippy, this is a pretty odd situation. So far, I am not able to replicate the situation. I will give it another try in the afternoon. Greets, Arjan

vanvught, 2020-09-27:
Hi Hippy, just to be sure; you are using my u-boot versions? Greets, Arjan

hippyau, 2020-09-27:
Hey Arjan,  no stress.   Yes this is your normal u-boot, with the image from the zip you posted in this issue.  
I make gps.txt exactly like yours - except offset,  and I change ltc.txt only line changed is source=systime 
I'll have a bit more of a play around and hopefully work out what is going on, and report back.

vanvught, 2020-09-27:
>  no stress

Well ;-) it should work. I am eager to know what’s wrong with that image. Or maybe we should just build another one ...

hippyau, 2020-09-27:
I put my own messages in there... It seems to go into GPSTimeClient::Run() 7 times before it crashes,  it also appears to get past GPS::Run() every time, but crashes very shortly thereafter on the 7th try...   I see there is timeouts in there, (1 * 1000)  and it seems to be okay for one second before it crashes... coincidence? 

```
 Sensors          : 0
Source : System-Time
 NTP output is disabled
 WS28xx output is disabled
GPS::Run()...
0x4003441e |$BDGSV,1,1,00*68|
GPS::Run() Returns
GPS::Run()...
0x40034470 |$GNRMC,,V,,,,,,,,,,N*4D|
GPS::Run() Returns
GPS::Run()...
0x400344c2 |$GNVTG,,,,,,,,,N*2E|
GPS::Run() Returns
GPS::Run()...
GPS::Run() Returns
GPS::Run()...
GPS::Run() Returns
GPS::Run()...
GPS::Run() Returns
GPS::Run()...
GPS::Run() Returns
GPS::Ru

```

Of course I am an idiot, how to get these debug messages printing out?  :)

`DEBUG_PUTS("(GPS::GetStatus() == GPSStatus::VALID)");`

`DEBUG_PUTS("No time update");`

vanvught, 2020-09-27:
> GPS::Run() Returns
GPS::Ru

Will you give it a try without connecting the RX on the debug UART header? 

>  how to get these debug messages printing out? :)

See https://github.com/vanvught/rpidmx512/blob/experimental/lib-gps/src/gps.cpp#L28

hippyau, 2020-09-27:
Okay thanks, i'll try that.

Is also interesting the scratch-h3-gps_time seems to work without crashing.

vanvught, 2020-09-27:
Hi Hippy, some good news. I can replicate your situation now -> I have connected the PPS. So weird .....

vanvught, 2020-09-27:
> Is also interesting the scratch-h3-gps_time seems to work without crashing.

Yes, it is. I am lost. It seems that the PPS interrupt will crash the LTC SMPTE firmware. I cannot believe, I have not tested with PPS. I am lost ;-) Anyway, some work, debugging to do ...

hippyau, 2020-09-27:
Hey Arjan,

  Ahh.. see have always had it connected.

Is weird,  I added a bunch of temporary debug messages to try and see where it leaves,  but it seems like we never leave GPSTimeClient::Run() ??

```

void GPSTimeClient::Run() {
	DEBUG_PUTS("Enter GPSTimeClient::Run()");

	GPS::Run();

	if (s_Status == Status::WAITING_TIMEOUT) {
		const uint32_t nMillis = Hardware::Get()->Millis();

		if (__builtin_expect(((nMillis - s_nLastUpdateMillis) < (3600 * 1000)), 1)) {
			DEBUG_PUTS("Disable fiq()");
			__disable_fiq(); //TODO Can this be done different? I am not happy now ;)
			DEBUG_PUTS("Leave GPSTimeClient::Run()");
			return;
		}

		if (GPS::GetStatus() == GPSStatus::VALID) {
			m_nWaitPPSMillis = nMillis;

			s_Status = Status::WAITING_PPS;
			__enable_fiq();

			DEBUG_PUTS("(GPS::GetStatus() == GPSStatus::VALID)");
			return;
		}

		DEBUG_PUTS("No Fix");

		if (GPS::IsTimeUpdated()) {
			const uint32_t nElapsedMillis = nMillis - GetTimeTimestampMillis();

			if (nElapsedMillis < (1 * 1000)) {

				struct timeval tv;
				tv.tv_sec = GPS::GetLocalSeconds();
				tv.tv_usec = 0;
				settimeofday(&tv, nullptr);

				s_nLastUpdateMillis = nMillis;
				s_Status = Status::WAITING_TIMEOUT;

				DEBUG_PRINTF("(GPS::IsTimeUpdated()) %u", nElapsedMillis);

				return;
			}
		}

		s_Status = Status::NOT_SET;

		DEBUG_PUTS("No time update");
		return;
	}

	if (s_Status == Status::WAITING_PPS) {
		const uint32_t nMillis = Hardware::Get()->Millis();

		if (__builtin_expect(((nMillis - m_nWaitPPSMillis) > (1 * 1000)), 0)) {
			// There is no PPS
			if (GPS::IsTimeUpdated()) {

				struct timeval tv;
				tv.tv_sec = GPS::GetLocalSeconds();
				tv.tv_usec = 0;
[... 148 lines cut ...]
 Product Category  : 0801
 Product Detail    : 0601
RDM Responder configuration
 Protocol Version 1.0
 DMX Address      : 65535
 DMX Footprint    : 0
 Personality 1 of 1 [RDMNet LLRP Only]
 Sub Devices      : 0
 Sensors          : 0
Source : System-Time
 NTP output is disabled
 WS28xx output is disabled
Run() src/h3/gpstimeclient.cpp, line 99: Enter GPSTimeClient::Run()
0x4003841e |$GNGLL,,,,,,V,N*7A|
Run() src/h3/gpstimeclient.cpp, line 99: Enter GPSTimeClient::Run()
0x40038470 |$GPGSA,A,1,,,,,,,,,,,,,25.5,25.5,25.5*02|
Run() src/h3/gpstimeclient.cpp, line 99: Enter GPSTimeClient::Run()
Run() src/h3/gpstimeclient.cpp, line 99: Enter GPSTimeClient
U-Boot SPL 2017.03-rc2-00252-gc987c17120-dirty (Oct 26 2019 - 15:37:55)
DRAM: 256 MiB
Trying to boot from sunxi SPI

U-Boot 2017.03-rc2-00252-gc987c17120-dirty (Oct 26 2019 - 15:37:55 +0200) Allwinner Technology

```

hippyau, 2020-09-27:
Yep, if I disconnect PPS,  it leaves... and seems to run fine.
```
....
0x40038514 |$GNGLL,,,,,,V,N*7A|
Run() src/h3/gpstimeclient.cpp, line 107: Disable fiq()
Run() src/h3/gpstimeclient.cpp, line 109: Leave GPSTimeClient::Run()
Run() src/h3/gpstimeclient.cpp, line 99: Enter GPSTimeClient::Run()
0x40038566 |$GNZDA,,,,,,*56|
Run() src/h3/gpstimeclient.cpp, line 107: Disable fiq()
Run() src/h3/gpstimeclient.cpp, line 109: Leave GPSTimeClient::Run()
Run() src/h3/gpstimeclient.cpp, line 99: Enter GPSTimeClient::Run()
0x400385b8 |$GPTXT,01,01,01,ANTENNA OK*35|
Run() src/h3/gpstimeclient.cpp, line 107: Disable fiq()
Run() src/h3/gpstimeclient.cpp, line 109: Leave GPSTimeClient::Run()
....

```

vanvught, 2020-09-27:
Hi Hippy,

> Yep, if I disconnect PPS, it leaves... and seems to run fine.

We have an issue with the FIQ. When not used, it is working fine.

Have I told you already; I am lost ;) There is an another FIQ routine enabled, which is causing an issue.

[image]

hippyau, 2020-09-27:
No worries :)

I had to remove the word 'local' to get the scratch firmware to compile,
(compiler suggested change) maybe the WrapperGetLocalSeconds is failing
here?

Cheers,

vanvught, 2020-09-27:
Got it ;( when LTC output is enabled, then the FIQ is already in use. I am not happy ;)

[image]

vanvught, 2020-09-27:
All, I need to go back to my drawing board. My apology. With the attachment here, you can work with GPS Time, only when `disable_ltc=1`

- Arjan

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

hippyau, 2020-09-27:
Thanks for your work on this Arjan, no stress.
Will disable ltc.

Maybe PPS can be a regular IRQ and just record a timestamp that Run can process.  I know IRQ is another branch instruction, but only happens once a second.

I imagine LTC would benefit greatly from FIQ due to frequency.

Cheers,
Hio

closed 2020-10-03 by vanvught

vanvught, 2020-10-03:
Dear all, please find attached the V2.8 with the new System Time framework and the GPS Time support. Although you might not have a GPS module, I hope you will give it a good test. Many thanks, Arjan

[opi_emac_ltc_smpte.zip](https://github.com/vanvught/rpidmx512/files/5321665/opi_emac_ltc_smpte.zip)

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

hippyau, 2020-10-07:
Hey Arjan :)

Thanks for this.  I've installed, and it seems to boot with/without PPS connect, and i'm getting ArtNet TC from systime source, but i really can't tell if the GPS module is working or not,  the clock is just running from 00:00:00:00.  

I don't see any kind of "GPS Module Found" message, or at least I can recognize.  This is all I see, does this look right?  I might not have a satellite?    Cheers :)

```
Hit any key to stop autoboot:  0 
spi_flash_decode_fdt: Cannot decode address
SF: Detected mx25l1605d with page size 256 Bytes, erase size 64 KiB, total 2 MiB
device 0 offset 0x180000, size 0x22000
SF: 139264 bytes @ 0x180000 Read: OK
## Booting kernel from Legacy Image at 48000000 ...
   Image Name:   http://www.orangepi-dmx.org
   Image Type:   ARM U-Boot Standalone Program (gzip compressed)
   Data Size:    110616 Bytes = 108 KiB
   Load Address: 40000000
   Entry Point:  40000000
   Verifying Checksum ... OK
   Uncompressing Standalone Program ... OK
--> rtc/hwclock.cpp:HwClock:46
--> rtc/hwclockrtc.cpp:RtcProbe:66
<-- rtc/hwclock.cpp:HwClock:55
No RTC connected
--> rtc/hwclock.cpp:HcToSys:76
<-- rtc/hwclock.cpp:HcToSys:78
MX25L1605D, sector size 4096, 2097152 bytes
Detected MX25L1605D with sector size 4096 total 2097152 bytes
No display found
[V2.8] Orange Pi Zero Compiled on Oct  3 2020 at 14:41:34
LTC SMPTE
Network init
Network
 Hostname  : allwinner_420913
 If        : 1: eth0
 Inet      : 192.168.1.122/24
 Netmask   : 255.255.255.0
 Gateway   : 192.168.1.1
 Broadcast : 192.168.1.255
 Mac       : 02:42:3c:42:09:13
 Mode      : D
NTP v4 Client
 Not enabled
MAX7219
 matrix [4]
Configuring Art-Net
Starting Art-Net
Node 3
 Firmware   : 1.47
 Short name : LTC SMPTE Node
 Long name  : Orange Pi Zero Art-Net 3 www.orangepi-dmx.org
Art-Net started
MIDI
 Direction    : Output
 Active sense : Disabled
 Baudrate     : 31250 (Default)
 Interface    : UART2
mDNS
 Name : allwinner_420913.local
 _apple-midi._udp.local 5004 
 _config._udp.local 10501 
AppleMIDI
 SSRC    : c0a8017a (3232235898)
 Session : allwinner_420913
--> src/gps.cpp:GPS:74
<-- src/gps.cpp:GPS:84
unbind--> src/gps.cpp:Start:153
<-- src/gps.cpp:Start:186
GPS
 Module : ATGM336H [9600]
 UTC offset : 36000 (seconds)
 Idle
 UART: 1
RDMNet configuration
 CID : 02c00142-8470-4620-b894-06183c420913
LLRP Device configuration
 Port UDP               : 5569
 Multicast join Request : 239.255.250.133
 Multicast Response     : 239.255.250.134
RDM Device configuration
 Manufacturer Name : www.orangepi-dmx.org
 Manufacturer ID   : 5000
 Serial Number     : 3C420913
 Root label        : LTC SMPTE
 Product Category  : 0801
 Product Detail    : 0601
RDM Responder configuration
 Protocol Version 1.0
 DMX Address      : 65535
 DMX Footprint    : 0
 Personality 1 of 1 [RDMNet LLRP Only]
 Sub Devices      : 0
 Sensors          : 0
Source : System-Time
 NTP output is disabled
 WS28xx output is disabled
 RGB panel output is disabled

```

vanvught, 2020-10-07:
Hi Hippy,

> GPS
>  Module : ATGM336H [9600]
>  UTC offset : 36000 (seconds)
>  Idle
>  UART: 1

The above is not what I would like to see ;-) The ATGM336H should have switched to 115200 baud.

It would switch to back to 9600 when there is no response -> https://github.com/vanvught/rpidmx512/blob/master/lib-gps/src/gps.cpp#L175

The GPS module is always outputting a sentence. This should be printed here -> https://github.com/vanvught/rpidmx512/blob/master/lib-gps/src/gps.cpp#L167
as there is still `#undef` NDEBUG here -> https://github.com/vanvught/rpidmx512/blob/master/lib-gps/src/gpsdump.cpp

Groet, Arjan

hippyau, 2020-10-07:
Bugger,  I can see regular serial traffic on the scope, I've checked voltages and wiring, but nothing I do seems to get firmware to do anything else.   

I can see it going crazy in the debug serial (as expected) when I load my previous compile where I added my own messages whilst messing with the PPS a few days ago, so I think comms to the GPS module should be working.  

I'll keep digging.  

Cheers :)

hippyau, 2020-10-07:
It's really odd.  I can see on the scope on the GPS UART, a packet of data being sent at bootup to module, and regular messages coming back from the module... but it is like the firmware is not seeing the messages coming back for some reason.

I have to sleep now, but will try the logic analyzer tomorrow hopefully and look at the UART contents, my scope decode is pretty useless.

vanvught, 2020-10-08:
For reference

```
[V2.8] Orange Pi Zero Compiled on Oct  3 2020 at 14:41:34
LTC SMPTE
Network init
Network
 Hostname  : allwinner_1E0BD3
 If        : 1: eth0
 Inet      : 192.168.2.147/24
 Netmask   : 255.255.255.0
 Gateway   : 192.168.2.1
 Broadcast : 192.168.2.255
 Mac       : 02:42:08:1e:0b:d3
 Mode      : D
NTP v4 Client
 Not enabled
MAX7219
 matrix [4]
Configuring Art-Net
Starting Art-Net
Node 3
 Firmware   : 1.47
 Short name : LTC SMPTE Node
 Long name  : Orange Pi Zero Art-Net 3 www.orangepi-dmx.org
Art-Net started
MIDI
 Direction    : Output
 Active sense : Disabled
 Baudrate     : 31250 (Default)
 Interface    : UART2
mDNS
 Name : allwinner_1E0BD3.local
 _apple-midi._udp.local 5004
 _config._udp.local 10501
AppleMIDI
 SSRC    : c0a80293 (3232236179)
 Session : allwinner_1E0BD3
--> src/gps.cpp:GPS:74
<-- src/gps.cpp:GPS:84
unbind--> src/gps.cpp:Start:153
0x400343c4 |$GNGGA,,,,,,0,00,25.5,,,,,,*64|
[371]
<-- src/gps.cpp:Start:186
GPS
 Module : ATGM336H [115200]
 UTC offset : 7200 (seconds)
 Idle
 UART: 1
```

commit 88dc7e06 (2020-10-03): V2.8 [Fixed #145]

*. Ran clang-tidy -checks modernize-redundant-void-arg modernize-concat-nested-namespaces modernize-use-nullptr