vanvught/rpidmx512 #149: oPi Zero GPIO to HUB-75

hippyau, 2020-10-08:
[image]

Can the OrangePi can drive these pins directly with level shifting hardware?

Cheers :)

vanvught, 2020-10-08:
> Can the OrangePi can drive these pins directly with level shifting hardware?

Yes, it can. However, I am planning to design an open source hardware board for the OPi Zero with 2x 74HCT245.
I am going to save that image. Dank :)

hippyau, 2020-10-08:
I've pulled and compiled latest experimental ltc firmware,

Should I only need to add `rgbpanel_enable=1` to ltc.txt ?

:)

vanvught, 2020-10-08:
> Should I only need to add rgbpanel_enable=1 to ltc.txt ?
Yes

The sources can only be the network protocols or systime. 

And please add: `if (tLtcDisabledOutputs.bRgbPanel)  ....`

```
		if ((!tLtcDisabledOutputs.bWS28xx) || (!tLtcDisabledOutputs.bRgbPanel)) {
			ltcDisplayRgb.Run();
		}

		if (tLtcDisabledOutputs.bRgbPanel) {
			lb.Run();
		}
```

And remove this line -> https://github.com/vanvught/rpidmx512/blob/experimental/opi_emac_ltc_smpte/firmware/main.cpp#L547

[image]

hippyau, 2020-10-09:
Thanks so much for the info...
  
I kill a OPi maybe :)  I think either it was bad from the factory (never used before), or I wired my ribbon cable up wrong because I was eager to see it in action.  I just get nothing on the UART0 from it at all, no uboot... it's dead :(   

All my others are hooked up to stuff, so I'm going to order some Zeros now.   :) 

I was in such a rush, I used the IDC cable that came with the panel, it is not copper but aluminum wire :( and didn't solder well... but this didn't kill the OPi.    

How many panels in full resolution do you think the OPi can generate at ~30FPS?  Since I don't think we could handle more than 2 or 3 panel max (?), I was thinking of making (when time) a pcb that adapts the HUB-75 directly to the OPi header,  so was small PCB that holds OPi here.  

[image]

vanvught, 2020-10-09:
> How many panels in full resolution do you think the OPi can generate at ~30FPS?

The FPS depends on the colour depth. With the current setting there is 120FPS for 64x32 led panel.

vanvught, 2020-10-09:
Hi Hip,

When you install https://github.com/vanvught/rpidmx512/tree/experimental/scratch-h3-matrix_gpio then the OLED display will show both the panel FPS and the sACN E1.31 FPS.
The colour depth is set here -> https://github.com/vanvught/rpidmx512/blob/experimental/lib-rgbpanel/include/rgbpanel.h#L35

closed 2020-10-25 by vanvught

hippyau, 2020-11-06:
Hey Arjan!! 

I have an Zero all hooked up, I used fresh pull of the above scratch-h3-matrix_gpio, but I am not sure my ACN setup is working...  I can see on scope on the clock line it's working, but no pixels on display,  I think my DMX is not getting there...

So just wanting to test rgb panel is working, I have not seen any pixels yet, so decide to try ltc firmware, but fresh pull from experimental, and I can't get opi_emac_ltc_smpte compile...

I can see lib-midi get compiled, but later I get an RtpMidiHandler problem?

```
$TARGET [lib_h3/libh3.a ]
arm-none-eabi-ar: creating lib_h3/libh3.a
[/mnt/d5e5c544-a90f-45fd-b1b3-d028b0ebdd07/OrangePi/rgbpanel/rpidmx512/lib-arm]
[src src/h3  src/debug]
$DEFINES [-DORANGE_PI  -DNDEBUG -D_TIME_STAMP_YEAR_=2020 -D_TIME_STAMP_MONTH_=11 -D_TIME_STAMP_DAY_=6 -DREMOTE_CONFIG]
$MAKE_FLAGS [-DLTC_READER -DRDMNET_LLRP_ONLY -DARM_ALLOW_MULTI_CORE -DNDEBUG -DENABLE_TC1602 -DENABLE_CURSOR_MODE -D_TIME_STAMP_YEAR_=2020 -D_TIME_STAMP_MONTH_=11 -D_TIME_STAMP_DAY_=6]
$BUILD_DIRS [build_h3/src build_h3/src/h3 build_h3/src/debug]
$TARGET [lib_h3/libarm.a ]
arm-none-eabi-ar: creating lib_h3/libarm.a
../lib-ltc/lib_h3/libltc.a(rtpmidireader.o): In function `RtpMidiReader::~RtpMidiReader()':
rtpmidireader.cpp:(.text+0x44): undefined reference to `RtpMidiHandler::~RtpMidiHandler()'
../lib-ltc/lib_h3/libltc.a(rtpmidireader.o): In function `RtpMidiReader::~RtpMidiReader()':
rtpmidireader.cpp:(.text+0x74): undefined reference to `RtpMidiHandler::~RtpMidiHandler()'
../h3-firmware-template/Rules.mk:223: recipe for target 'build_h3/main.elf' failed
make: *** [build_h3/main.elf] Error 1
hip@hip-Alienware-13:~/OrangePi/rgbpanel/rpidmx512/opi_emac_ltc_smpte$

```

Any ideas?

hippyau, 2020-11-06:
additionally, using the scratch-h3-matrix_gpio firmware, in the RC manager, I only get two files...
[image]

I'm using qlcplus and sending unicast, but can't tell if my DMX is getting there...

```
Shell() lib/h3/shell.cpp, line 74: TABLE_SIZE=14
MX25L1605D, sector size 4096, 2097152 bytes
Detected MX25L1605D with sector size 4096 total 2097152 bytes
[V0.0] Orange Pi Zero Compiled on Nov  6 2020 at 21:42:54
Network
 Hostname  : allwinner_380453
 If        : 1: eth0
 Inet      : 192.168.1.134/24
 Netmask   : 255.255.255.0
 Gateway   : 192.168.1.1
 Broadcast : 192.168.1.255
 Mac       : 02:42:04:38:04:53
 Mode      : D
PlatformInit() src/h3/rgbpanel.cpp, line 128: nBufferSize=172032
PlatformInit() src/h3/rgbpanel.cpp, line 136: 0x4050082c 0x405a883c
smp_start_core(1, core1_task)
Running
RGB led panel
 64x32x1
 Universes : 1 to 13-24
Bridge
 Firmware : 1.19
 Output
  Port  0 Universe 1   [HTP]
  Port  1 Universe 2   [HTP]
  Port  2 Universe 3   [HTP]
  Port  3 Universe 4   [HTP]
  Port  4 Universe 5   [HTP]
  Port  5 Universe 6   [HTP]
  Port  6 Universe 7   [HTP]
  Port  7 Universe 8   [HTP]
  Port  8 Universe 9   [HTP]
  Port  9 Universe 10  [HTP]
  Port 10 Universe 11  [HTP]
  Port 11 Universe 12  [HTP]
  Port 12 Universe 13  [HTP]
 Direct update : Yes

```

hippyau, 2020-11-06:
Update:  I can see R G B, Latch, OE signals on the scope, i'm confident something should be happening, maybe my panel not working :(

vanvught, 2020-11-06:
[image]
[image]

[image]

vanvught, 2020-11-06:
> additionally, using the scratch-h3-matrix_gpio firmware, in the RC manager, I only get two files...

That's 'fixed' in the latest commit -> https://github.com/vanvught/rpidmx512/blob/experimental/scratch-h3-matrix_gpio/firmware/main.cpp#L126

hippyau, 2020-11-06:
Dank!  I will try now :)

hippyau, 2020-11-06:
Hey Arjan, getting closer, but now...

```
[src src/h3  src/debug]
$DEFINES [-DORANGE_PI  -DNDEBUG -D_TIME_STAMP_YEAR_=2020 -D_TIME_STAMP_MONTH_=11 -D_TIME_STAMP_DAY_=6 -DREMOTE_CONFIG]
$MAKE_FLAGS [-DLTC_READER -DRDMNET_LLRP_ONLY -DARM_ALLOW_MULTI_CORE -DNDEBUG -DENABLE_SSD1311 -DENABLE_TC1602 -DENABLE_CURSOR_MODE -D_TIME_STAMP_YEAR_=2020 -D_TIME_STAMP_MONTH_=11 -D_TIME_STAMP_DAY_=6]
$BUILD_DIRS [build_h3/src build_h3/src/h3 build_h3/src/debug]
$TARGET [lib_h3/libdisplay.a ]
src/lcdbw.cpp: In member function 'void LcdBw::Write(const char*, uint32_t)':
src/lcdbw.cpp:161:27: error: 'micros' was not declared in this scope
  161 |  const uint32_t elapsed = micros() - m_nWriteMicros;
      |                           ^~~~~~
../h3-firmware-template/lib/Rules.mk:114: recipe for target 'build_h3/src/lcdbw.o' failed

```

hippyau, 2020-11-06:
is to do with `-DENABLE_SSD1311 -DENABLE_TC1602` ? ... I don't have those devices, where to disable?

vanvught, 2020-11-06:
> is to do with -DENABLE_SSD1311 -DENABLE_TC1602 ? ... I don't have those devices, where to disable?

It has nothing to do with the defines. And they should not be disabled.

hippyau, 2020-11-06:
I had a wiring problem :)  I now have some pixel action...  but still can't compile because of above.

vanvught, 2020-11-06:
```
src/lcdbw.cpp: In member function 'void LcdBw::Write(const char*, uint32_t)':
src/lcdbw.cpp:161:27: error: 'micros' was not declared in this scope
  161 |  const uint32_t elapsed = micros() - m_nWriteMicros;
      |                           ^~~~~~
../h3-firmware-template/lib/Rules.mk:114: recipe for target 'build_h3/src/lcdbw.o' failed
```

I was able to reproduce this error. I have committed files in order to fix the issue. I cloned 'experimental' to a different place and the compilation is running fine now.

hippyau, 2020-11-06:
I didn't see commit, will try now... :)

hippyau, 2020-11-06:
Compiles :+1:  :+1:  :+1:

vanvught, 2020-11-06:
>  I now have some pixel action...

You can change the color depth here -> https://github.com/vanvught/rpidmx512/blob/experimental/lib-rgbpanel/include/rgbpanel.h#L35
The OLED display will tell the refresh rate of the panel.

hippyau, 2020-11-06:
Yay, thanks!!!!!!!!!!!!

[image]

vanvught, 2020-11-06:
The last colon dot will be fixed soon :-)

When you still awake, can you check the clock frequency of the panel CLK?

It is on my todo list, but you have the scope ready 😀

hippyau, 2020-11-06:
Hey Arjan,  
 This is straight off PA10,  it drops a clock every now and then, but seems pretty consistently about 9MHz...

[image]

hippyau, 2020-11-06:
Hey Arjan,  I have this other small panel...

[image]

It seems to have a colour swap?  See the yellow is blue?  This wiring is the same, I have glued together dupont wires into a connector,  works as expected on big panel, but different on the small panel... 

[P2.5 Indoor Full Color LED Display Module HUB75,160mm x80mm, 64*32 Pixels,SMD RGB P2.5 LED Panel Matrix, Compatible With PIN2DMD](https://a.aliexpress.com/_mO2eMVx)

vanvught, 2020-11-06:
> This is straight off PA10, it drops a clock every now and then, but seems pretty consistently about 10MHz...

Hi Hip, thanks! 

When there is a drop, then the row address is being latched. 
The 10Mhz is what I thought it would be based on refresh rate and color depth. It relatively low frequency; I am not sure yet if this is a limitation for the Allwinner H2+/H3 SoC, or that there is lock in my multi-core code preventing higher frequency.

For the sACN firmware where we can select the panel size and and panel count, we can use this 10 MHz for setting the color depth with having around 120 FPS panel update.

vanvught, 2020-11-06:
> It seems to have a colour swap? See the yellow is blue? This wiring is the same, ...

I have ordered 3 another display's from the same seller; I really hope that the color scheme is the same for all 4. I am intending to make 2x2 64x32 display .... ;-)

vanvught, 2020-11-06:
[image]
https://github.com/vanvught/h3dmx512-zip/tree/master/eagle/Orange-Pi-Zero/RGB-panel

hippyau, 2020-11-06:
Hey, do you happen to know if JLCPCB has these components in it's assembly option?

vanvught, 2020-11-06:
> Hey, do you happen to know if JLCPCB has these components in it's assembly option?

Yes, they have. The BOM and placement file is on my GitHub. Just upload, click , click ... pay and ready ;-)

hippyau, 2020-11-06:
<3 <3 Thanks!  I'm doing it now :)

hippyau, 2020-11-07:
Hey Arjan,
  Trying to now test the scratch-h3-matrix_gpio firmware,  but I can no longer see the opi in the remote config util, but it's online.  

it's being seen in scan, `192.168.1.134,sACN E1.31,RGB Panel,13` but giving an `Invalid respone` ?

remote config util console output:
```
hip@hip-Alienware-13:~/OrangePi$ ./run.sh
Linux
enp2s0
	/fe80:0:0:0:f10f:a22d:2bf7:9a73%enp2s0 false false false
	/192.168.1.49 false false false
Local ip: 192.168.1.49
Interface address: 192.168.1.49
192.168.1.56,LTC,TimeCode,7,BigTC
Mode[0]=TimeCode
{192.168.1.56 LTC TimeCode 7} {BigTC}
192.168.1.134,sACN E1.31,RGB Panel,13
Mode[0]=RGB Panel
Invalid respone
No more messages.

```

hippyau, 2020-11-07:
Hi Arjan,

sACN is working great, I've got smooth pixel mapping...
[image]

I did notice if the sACN stream stops then starts again a bit later,  I get another `smp_start_core(1, core1_task)` message on the console and it crashes and needs hard power cycle, WDT not kicking in.    Otherwise it's been running for an hour and hasn't missed a beat!

vanvught, 2020-11-07:
Hi Hippy,

> did notice if the sACN stream stops then starts again a bit later, I get another smp_start_core(1, core1_task) message on the console and it crashes and needs hard power cycle

Yes, the 'blackout' feature is not working, yet. 

> WDT not kicking in. 

Yes, there is no WDT; that makes debugging easier.

hippyau, 2020-11-07:
Very happy, thanks Arjan!!    Side note, I can run this free pixel matrix software in Linux under WINE perfectly.... [Jinx!](http://www.live-leds.de/)

hippyau, 2020-11-07:
> It seems to have a colour swap? See the yellow is blue? This wiring is the same, I have glued together dupont wires into a connector, works as expected on big panel, but different on the small [...]

Confirming this panel has green and blue swapped - it's RBG

hippyau, 2020-11-07:
oPi Zero only hits 41 degrees Celsius when running two cores and receiving 13 sACN universes.  This is good, because in Australia it hits 40 in summer,  put it in a closed black box in the sun, will still probably keep running :)  I'll test on Christmas Day, hehehe...
 
[image]

hippyau, 2020-11-07:
> The FPS depends on the colour depth. With the current setting there is 120FPS for 64x32 led panel.

Does overall brightness also relate to colour depth?  My tiny 2.5P panel is not very bright.

vanvught, 2020-11-07:
Hi Hippy,

> Trying to now test the scratch-h3-matrix_gpio firmware, but I can no longer see the opi in the remote config util, but it's online.

Are you using the latest code ?

[image]

hippyau, 2020-11-07:
I'm using the latest RemoteConfig jar I could find, which is from August in the GPS thread.  
I compiled the firmware today.

vanvught, 2020-11-07:
> Does overall brightness also relate to colour depth? My tiny 2.5P panel is not very bright.

The RGB DMX values 0xFF, 0xFF, 0xFF are always translated to a full-on for all led's

vanvught, 2020-11-07:
> I'm using the latest RemoteConfig jar I could find, which is from August in the GPS thread.

I have not updated the .jar. There is updates sources only ;-)

Give me a second ...

vanvught, 2020-11-07:
[Uploading RemoteConfigUI.jar.zip…]()

vanvught, 2020-11-07:
[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/5504559/RemoteConfigUI.jar.zip)

hippyau, 2020-11-07:
Thank you, I can see it now :+1:   :) 
[image]

vanvught, 2020-11-07:
Thank you Hippy!!

When you would like to see 'strange' things happening; put a full 13 Universe image on the panel. Stop outputting, wait for the timeout, the blackout. And then send Universe 13 (or the last Universe) only .. :-)

hippyau, 2020-11-08:
I'm trying to hack in a 5x8 font, but not really winning.  I can see I need to >> 3 to get to the 5 bits, but just getting gobbly gook.   Work in progress,  you might see at a glance where i'm going wrong.   Plus, keep in mind I'll clean up once it's working :) 

[ Trying to hack in a 5x8 font... WIP](https://github.com/hippyau/rpidmx512/commit/a9b76953fbcbe4c49c4ac994a804d488488a75f9)

hippyau, 2020-11-17:
Yay, my boards arrived!   Just FYI,  JLC warned me that U1 and U2 placing was possibly an issue.  I said build anyway, and they look great.     One question:  is the +5v an output?

[image]

vanvught, 2020-11-17:
> One question: is the +5v an output?

No, that is input. Back powering the Orange Pi Zero on the header based on the recommendation for a Raspberry Pi HAT

rewolff, 2020-11-17:
.... But in practise if you have  good reason, you could also power the orange pi itself. That would make this connector either unnecessary or an output. But keep in mind that the reason of that rpi recommendation is that you might have a fuse on the "xxx-pi" board, so that you can trigger that fuse if you draw too much.
So best is to follow the recommendations and power everything properly fused from the hat.

vanvught, 2020-11-17:
I am using these boards for debugging and testing my firmware and make it open source hardware (https://github.com/vanvught/h3dmx512-zip/tree/master/eagle/Orange-Pi-Zero/RGB-panel), therefore if someone has its own requirements, they can simply change the board.

hippyau, 2020-11-18:
Hey Arjan!  

I got my boards in the mail :)   Thanks so much for creating this board!!  I've built up one with OLED display too...

[image]

It's awesome!  I do have a tiny hick-up,  I have a bit of small flicker.  

I hooked up the scope to the clock lines and they look pretty much the same.

The only thing 'major' besides the whole hat itself, that is different between my two test units is the OLED display.   Same configs, same firmware - one is wired direct to the panel, one has a hat + oled.   

foolishly I solder the OLED on, is there a config to force disable it? I presume the OLED is detected and only used when connected?    

Have you seen a slight flicker?

Not stressed at all, direct connect works great, **love it :)** 

[image]
(hat)

[image]
(direct)

hippyau, 2020-11-18:
I've looked at the data-sheet and I am kind of sure these are the same, kind of? 
[image]
(My part left, your part Right)

vanvught, 2020-11-18:
Hi Hippy,

> foolishly I solder the OLED on, is there a config to force disable it? I presume the OLED is detected and only used when connected?

Yes, when a supported OLED is detected, then it is used.

> Have you seen a slight flicker?

I have a major backlog for things to do (most time consuming is #152, I am in direct contact with the company, but so little support; I cannot get the OLED to work) and so little free time; I still have to solder the boards ;-)

For the 245, I would expect to see AHCT, the fast version. But as your scope is showing fine signals, it would be ok.
I really do hope that I can find some time in the weekend for building one board. 

But there is also still this empty page -> http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/ws28xx-configuration/prototype-boards/multi/8-output ;-(

> Not stressed at all, direct connect works great, love it :)

:-)

hippyau, 2020-11-18:
Thanks man...  I just built another board without OLED and have no flicker :) the hardware works great!!  <3

I'll have a look and see if I can halve the update speed of the OLED maybe and that will probs solve it.   not a big problem :)

> and so little free time; I still have to solder the boards ;-)

I am still to assemble my LTC boards - i've had all the parts for over a month now hehe!   

Very happy with the support!  Stay well and sane, don't worry about me, I get there eventually :) just like to document my journey!

hippyau, 2020-11-18:
[observation] Very interesting... board with OLED still fickers when display on or off, using the gui to toggle, the OLED display does blank when off, but the display still flickers.  The new board without any OLED display is with no visible flicker.  

[image]

vanvught, 2020-11-18:
> board with OLED still fickers when display on or off,

When the OLED display is off, then there is still I2C activity. The I2C on the header is also GPIOA, which is controlling the panel. So, it might be a good idea not outputting the Timecode to OLED when this panel is active. And just showing the static information on the OLED.

Fact; in my current setup there is no OLED connected and there is no flickering ....

hippyau, 2020-11-18:
Thanks, i've made a hack for above UI contol,  when display is off... display is not updated.   Works for my board with OLED display.     No stress :1st_place_medal:

hippyau, 2020-11-21:
Show message blocks TC somehow... lol, is my code no doubt! 
[unlisted youtube video showing issue]( https://youtu.be/1nxXxa8Fqh0 )