vanvught/rpidmx512 #151: Ethernet Link Up/Down [Question]

hippyau, 2020-10-09:
Hey Arjan,

 I've noticed a reboot is required an ethernet cable is not present bootup to get hear the network,  is that correct?  

Does firmware look once and once only for network?

Cheers,
  Hip

vanvught, 2020-10-09:
Hi Hip,

> I've noticed a reboot is required an ethernet cable is not present bootup to get hear the network, is that correct?

No, reboot is not required. When you boot without the ethernet cable, then the link is up when the cable is connected.

> Does firmware look once and once only for network?

Yes. Just at startup. 

Groet, Arjan

closed 2020-10-09 by hippyau

soerenpedersen, 2021-03-09:
> > Does firmware look once and once only for network?
> 
> Yes. Just at startup.

Hello Arjan,

Perhaps this should be changed? We have some environments where the DHCP is out of our control (and is quite slow to boot). So I could imagine that e.g. after a power surge the Orange Pi's would boot - fail to acquire a DHCP lease and be stuck with their locally assigned IP. Obviously this is not desirable. Attempting DHCP renewal or rebooting at X interval would be great for this (X perhaps even configurable) :-)

Best regards,
Søren