vanvught/rpidmx512 #152: LTC SMPTE: Add support for EA OLEDM204 [SSD1311]

vanvught, 2020-10-23:
Examples availability:

- https://www.distrelec.nl/nl/alphanumeric-oled-display-yellow-electronic-assembly-ea-oledm204-lga/p/30117338
- https://www.tme.eu/nl/details/eaoledm204-lga/alfanumerieke-oled-displays/electronic-assembly/ea-oledm204-lga/

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

vanvught, 2020-11-14:
I have bought 2 EA OLEDM204 at https://www.reichelt.nl and both do not work -> they are not recognized on the I2C bus. I am already in direct contact with ELECTRONIC ASSEMBLY GmbH. Any input here is much appreciated. Thanks, Arjan

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

vanvught, 2020-11-26:
The intensity(/contrast) can be set now.

[image]

[orangepi_zero.img.gz](https://github.com/vanvught/rpidmx512/files/5604556/orangepi_zero.img.gz)

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

wyged, 2020-11-27:
Hi Arjan. I'm struggling here to understand how to install the new
firmware you send me yesterday, on the SD Card. The file you sent me is
called " orangepi_zero.img.gz ", do I have to rename this as " uImage "???
the previous versions are named " orangepi_zero.uImage.gz " and they worked
fine.

Please Help!!!

*Chip Vicol*
Pandora's Box / Widget Designer Programmer
Cell: +40 735 024 656
Email: chipvicol@gmail.com

hippyau, 2020-11-27:
Hi Chip,

 Yes, you should rename this to `uImage` and copy to the SD.   The SD card
should have had a uboot image written to it previously?

Uboot will look for uImage on the SD, boot it, and if it is different to
the version in the SPI flash, it will copy it to the SPI flash, so you can
boot without the SD.

Cheers,
  Hippy

wyged, 2020-11-27:
Did that, and still doesn't work. Is working fine with previous versions.
Realy confused here

*Chip Vicol*
Pandora's Box / Widget Designer Programmer
Cell: +40 735 024 656
Email: chipvicol@gmail.com

hippyau, 2020-11-27:
Do you have a usb<>serial device you can connect to the debug output, that
info is really helpful.

Can you see any other signs of life?  Can you see the firmware show up in
the Remote Configuration UI?

I don't know anything about this particular firmware (I don't have that
display) so Arjan might have some idea what is going on,  maybe the new
intensity setting he added should be tried?

Presumably it was working for him.  I'll shut up now, because I don't know
why 😀

Cheers,
  Hippy

vanvught, 2020-11-27:
I have attached the wrong file in previous append ;-(

Verstuurd vanaf mijn iPhone

vanvught, 2020-11-27:
[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/5607259/orangepi_zero.uImage.gz)

hippyau, 2020-11-27:
Chip, how good is this service....   Arjan is a legend!   <3

wyged, 2020-11-27:
It is working now!!!!!!🥰THANK YOU!Sent from my Huawei phone-------- Original message --------From: Arjan <notifications@github.com>Date: Fri, Nov 27, 2020, 12:18 PMTo: vanvught/rpidmx512 <rpidmx512@noreply.github.com>Cc: Chip Vicol <chipvicol@gmail.com>, Mention <mention@noreply.github.com>Subject: Re: [vanvught/rpidmx512] LTC SMPTE: Add support for EA OLEDM204 [SSD1311] (#152)
orangepi_zero.uImage.gz

wyged, 2020-11-27:
Sent from my Huawei phone-------- Original message --------From: Arjan <notifications@github.com>Date: Fri, Nov 27, 2020, 12:18 PMTo: vanvught/rpidmx512 <rpidmx512@noreply.github.com>Cc: Chip Vicol <chipvicol@gmail.com>, Mention <mention@noreply.github.com>Subject: Re: [vanvught/rpidmx512] LTC SMPTE: Add support for EA OLEDM204 [SSD1311] (#152)
orangepi_zero.uImage.gz

wyged, 2020-11-27:
Yes.
Arjan is a legend!
Never seen some so dedicated to his work/hobby!

Thank you, Arjan!

@hippy
What do you mean by "service"???
If you are talking about this project... It is amazing!
Working now on drawing the final schematics with balanced LTC IN and OUT
with ESD protection, and hopefully add PoE.

*Chip Vicol*
Pandora's Box / Widget Designer Programmer
Cell: +40 735 024 656
Email: chipvicol@gmail.com

hippyau, 2020-11-27:
In service, I mean exactly what you mean, dedication.
You have a photo of your EA OLEDM204?  I'd like to see if I should order one.

Absolutely too right...  amazing!   You've see his LTC board, yes?  It suited my need, i've built Arjans board and have balanced LTC,  but interested in your ESD,  i'm not so knowledgeable is those areas yet.

wyged, 2020-11-27:
Sent from my Huawei phone-------- Original message --------From: Hippy <notifications@github.com>Date: Fri, Nov 27, 2020, 1:03 PMTo: vanvught/rpidmx512 <rpidmx512@noreply.github.com>Cc: Chip Vicol <chipvicol@gmail.com>, Mention <mention@noreply.github.com>Subject: Re: [vanvught/rpidmx512] LTC SMPTE: Add support for EA OLEDM204 [SSD1311] (#152)
In service, I mean exactly what you mean, dedication.
You have a photo of your EA OLEDM204?  I'd like to see if I should order one.
Absolutely too right...  amazing!   You've see his LTC board, yes?  It suited my need, i've built Arjans board and have balanced LTC,  but interested in your ESD,  i'm not so knowledgeable is those areas yet.

wyged, 2020-11-27:
@hippyYou can look and find me on Facebook "chip vicol". I would be able to share more pictures with you there...Sent from my Huawei phone-------- Original message --------From: Hippy <notifications@github.com>Date: Fri, Nov 27, 2020, 1:03 PMTo: vanvught/rpidmx512 <rpidmx512@noreply.github.com>Cc: Chip Vicol <chipvicol@gmail.com>, Mention <mention@noreply.github.com>Subject: Re: [vanvught/rpidmx512] LTC SMPTE: Add support for EA OLEDM204 [SSD1311] (#152)
In service, I mean exactly what you mean, dedication.
You have a photo of your EA OLEDM204?  I'd like to see if I should order one.
Absolutely too right...  amazing!   You've see his LTC board, yes?  It suited my need, i've built Arjans board and have balanced LTC,  but interested in your ESD,  i'm not so knowledgeable is those areas yet.

vanvught, 2020-11-27:
[image]

[image]

hippyau, 2020-11-27:
Cheers Arjan,  that's pretty cool...   much more readable for my tired eyes :)

hippy, 2020-11-27:
I wondered wtf this repo was. You want hippyau not hippy. 

>  @hippy What do you mean by "service"??? If you are talking about this project... It is amazing!

hippyau, 2020-11-27:
Hey hippy,  lol!!

[image]

Is this you by any chance, there is too many hippys? :)

vanvught, 2020-11-29:
[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/5612778/orangepi_zero.uImage.gz)

Some minor fixes. Close to perfection ;-)

vanvught, 2020-11-30:
[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/5618752/orangepi_zero.uImage.gz)

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

commit 603eebac (2020-12-04): V2.8 [Fixed #152]
closed 2020-12-04 by vanvught