vanvught/rpidmx512 #153: LTC SMPTE:  If no source is selected within 10 seconds, the node will start with previous setting.

vanvught, 2020-10-25:
This applies for when using the MCP buttons option. 

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

vanvught, 2020-10-31:
Please will you test the attached firmware. Many thanks in advance, Arjan

cc @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/5469423/orangepi_zero.uImage.gz)

wyged, 2020-10-31:
Hi Arjan. The function works as it should. Thanks.
Do you happen to know why I'm getting an error when trying to upload the
uImage via TFTP? The OrangePI is getting into TFTP mode, the firewall is
off...
Please see attached screenshot.

Thanks
*Chip Vicol*
Pandora's Box / Widget Designer Programmer
Cell: +40 735 024 656
Email: chipvicol@gmail.com

vanvught, 2020-11-01:
Thanks Chip.

 I have sent you an email for resolving your TFTP issue. 

When we have completed #152 then I will release the new firmware on https://github.com/vanvught/h3dmx512-zip

closed 2020-11-01 by vanvught

drummerclint, 2020-11-01:
This is perfect!  Seems to work fine.

vanvught, 2020-11-01:
Thank Clint!