vanvught/rpidmx512 #154: Feature: possible to add : between HH:MM:SS:FF?

drummerclint, 2020-11-01:
Hey Arjan, 
Is it possible to add : or ; delimiters between the HH:MM:SS:FF on the large max7219 8x8 dot module display?

vanvught, 2020-11-02:
Hi Clint,

This is a great enhancement. Thanks. Yes it is possible. I will have to do some creative font modifications and we might even can have flashing ":"

Arjan

drummerclint, 2020-11-02:
Hey Arjan,

That'd be great, I'm glad you agree!  maybe use ; or , for the frame to denote if it is drop frame.

Clint

vanvught, 2020-11-03:
Hi Clint,

The , will be used for drop same, similar to https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/ltcoutputs.cpp#L123

Arjan

hippyau, 2020-11-05:
> While non-drop timecode is displayed with colons separating the digit pairs—"HH:MM:SS:FF"—drop-frame is usually represented with a semicolon (;) or period (.)  - [wikipedia](https://en.wikipedia.org/wiki/SMPTE_timecode)

a `,` is easier to distinguish on a small screen though. :+1:

vanvught, 2020-11-05:
I noticed some inconsistency in how this character is shown on the several displays. I will fix that. And this character is controlled from  https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/h3/ltcoutputs.cpp#L123 only.

I have currently set the DF character to ;
The NDF character is set to :

vanvught, 2020-11-14:
Hi Clint, please find attached the updated firmware.

I am not that enthusiastic about the blink effect.

All, please let me know your thought's. Dank, Arjan

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/5540972/orangepi_zero.uImage.gz)

drummerclint, 2020-11-17:
Thanks Arjan!
It looks okay, I really like the systime format.  Is there a way to clump the numbers between the :  closer together? for example 21 :57 :16 would be 21 : 57 : 16, although i get with certain digits like the 4 and 0, it takes up  7 of the 8 columns
Dank!

vanvught, 2020-11-17:
Hi Clint,

As the modules have a 8x8 led matrix I am using the CP437 font 8x8 bits. The placement of a character is fixed within each of the 8 modules. In CP437, the last column for the digits are zero's, hence I can use that column for showing the colon. 

Are you happy with this colon solution? Or should I revert to none? 

Thanks, Arjan

> Dank!

:-)

drummerclint, 2020-11-18:
Hey Arjan,

Definitely better than no colon.  
Obrigado! :)

commit 56af4816 (2020-11-21): Fixed #154

*. Ran clang-tidy -checks modernize-redundant-void-arg,modernize-concat-nested-namespaces,modernize-use-nullptr,modernize-use-auto,modernize-use-default-member-init
commit 61eb0da9 (2020-11-22): V2.8 [Fixed #154]

*. Ran clang-tidy -checks modernize-redundant-void-arg,modernize-concat-nested-namespaces,modernize-use-nullptr,modernize-use-auto,modernize-use-default-member-init
closed 2020-11-22 by vanvught