vanvught/rpidmx512 #156: RemoteConfigUI - firmware update workflow (& low priority request)

hippyau, 2020-11-07:
To update firmware, follow this procedure:

1.  Open Remote Config UI  ->  `$ java -jar RemoteConfigUI.jar`
2.  Select target device from list
3.  Using menus, `Action -> TFTP On/Off`,  confirm On.
4.  Using menus, `Run -> TFTP Client`
5.  Use "`Select uImage`" button to select the firmware to upload
6.  Use "`TFTP put`" button to upload firmware
7.  If success, text will turn green.
8.  Using menus, `Action -> TFTP On/Off`, confirm Off (firmware is now written to SPI flash)
9.  Right click on device, or use menus Action -> Reboot to start the new firmware.

It'd be awesome if the TFTP client would turn on TFTP and then turn off after upload, then reboot, so the firmware is update is one operation.  If we forget one step, it can leave beginners scratching their heads. 

I get the TFTP client might get used for other files in the future, so maybe a separate firmware update routine would be good.

vanvught, 2020-11-07:
For reference, the steps described is exactly the flow as used when automating using scripting -> https://github.com/vanvught/h3dmx512-zip/blob/master/do-tftp.sh

hippyau, 2020-11-10:
Annoying for you and me maybe, but RemoteConfigUI not allow reuse port of remote config messages, and do-tftp.sh fails when RC is open, because netcat faults on binding `nc: bind failed: Address already in use` presumably when trying to enable TFTP.   

Low priority.

vanvught, 2020-11-10:
>  Annoying for you ...

Well not really, the RemoteConfigUI is an AS-IS application demonstrating the remote configuration framework. I do my build workflow completely from the command-line.

hippyau, 2020-11-11:
Understood,  if I knew how to compile it, I could have a crack at it.   Pretty sure setsocketopt with SO_REUSEADDR would do the trick.  I do compile and do-tftp on the command line, but editing the .txt files is so much easier with the UI :)

vanvught, 2020-11-11:
> netcat faults on binding nc: bind failed: Address already in use

It seems to me that `netcat` is not doing the re-use

vanvught, 2020-11-11:
It is set in my code -> https://github.com/vanvught/rpidmx512/blob/master/RemoteConfigUI/src/org/orangepi/dmx/RemoteConfig.java#L893

hippyau, 2020-11-11:
ahh, okay thanks... doh!   never mind, sorry to waste your time.  `netcat` might require recompile to accomplish [https://superuser.com/questions/1321883/how-to-make-nc-listen-on-a-port-without-setting-so-reuseport](https://superuser.com/questions/1321883/how-to-make-nc-listen-on-a-port-without-setting-so-reuseport).

vanvught, 2020-11-15:
-> [http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/remote-configuration/example-java-ui/workflow-firmware-installation](http://www.orangepi-dmx.org/orange-pi-dmx512-rdm/remote-configuration/example-java-ui/workflow-firmware-installation)

vanvught, 2020-11-15:
New Menu Item

[image]

After TFTP On, the TFTP Client is shown

[image]

After finishing the TFTP transfer, there is a TFTP Off and the Reboot

[image]

and waiting for the Node to be back and showing the (new) firmware version

[image]

vanvught, 2020-11-15:
[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/5542787/RemoteConfigUI.jar.zip)

drummerclint, 2020-11-18:
This is awesome!

vanvught, 2020-11-18:
@drummerclint 
> This is awesome!

Thanks! We are close to perfection.

hippyau, 2020-11-18:
Look at all the work you have done!  👏🏻
[image]

drummerclint, 2020-11-18:
> @drummerclint
> 
> > This is awesome!
> 
> Thanks! We are close to perfection.

Indeed!  Can we have an option to download/backup a config from an opi to load onto another?

vanvught, 2020-11-19:
@drummerclint 
> Can we have an option to download/backup a config from an opi to load onto another?

Yes, we can :-) I need to fix some details. 

[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/5568936/RemoteConfigUI.jar.zip)

[image]
[image]
[image]
[image]
[image]

closed 2020-11-21 by vanvught