vanvught/rpidmx512 #157: Stop on error during compile?

hippyau, 2020-11-10:
How can I get my build_script.sh to stop compilation on first file with errors?  

It would just speed up my workflow a lot,  I am a "test vert often" kind of programmer, and when compile sometimes an error early on scrolls past and I don't see it because of the noise... and at the end some 20s later it fails trying to link firmware... I scratch the head sometimes :)

hippyau, 2020-11-13:
I have done light investigation and subsequently and naively tried to add  `|| exit 1` on the end of these statements in Rules.mk but is has not worked.  I assumed this would return an exit on error from what i've read, but I don't think it effective here, and might need to go somewhere else.

```
define compile-objects
$(BUILD)$1/%.o: $1/%.c
	$(CC) $(COPS) -c $$< -o $$@ || exit 1
	
$(BUILD)$1/%.o: $1/%.cpp
	$(CPP) $(COPS) $(CPPOPS)  -c $$< -o $$@ || exit 1
	
$(BUILD)$1/%.o: $1/%.S
	$(CC) $(COPS) -D__ASSEMBLY__ -c $$< -o $$@ || exit 1 
endef
```

vanvught, 2020-11-13:
Hi Hippy, 

I am not in favor modifying the make files. The solution should be found in the return code, the result of a make.

- Arjan

rewolff, 2020-11-13:
Make by default stops on error. You can ask it to continue with unrelated objects-to-be-built with the -k option on make. I have recently added flags I often use on make (-j 5) to "MAKEFLAGS" in my environment. This results in that argument being passed to all makes that I start. If you've done that with -k that would explain the current behavior. 

The -j 5 option could also be the cause: This will use multiple cores in your machine to compile the sources. Up to 5 jobs can be started. Still make will try to stop ASAP. So starting job 1,2,3,4,5 and then job 1 errors out, job 2,3,4,5 will continue and may generate warnings but job 6 should not be started. 

Another explanation would be that your compiler doesn't return error when it detects a problem with your sources. This might be caused by a wrapper-script that ends in "exit 0".

hippyau, 2020-11-14:
Thanks for the replies,

The script basically `make --quiet -f Makefile.H3 clean && make --quiet -f Makefile.H3 `

I notice that sans the --quiet option, it generates a lot more noise but does appear to stop faster - or my eyes playing tricks on me.   This is never going to work the way I'd like, because I presume it will keep going building libs even if one fails, until another lib relies on that lib.

It's not a big drama, just a slight inconvenience really.   Thanks again.

closed 2020-11-14 by hippyau

vanvught, 2020-11-14:
On my wishlist is converting the make files into CMake. Then it is also possible, easier to build out of the source tree. Just a wish ;-)