vanvught/rpidmx512 #161: widget_mode=3 (sniffer) ignored on Raspberry Pi Zero?

sdbbs, 2020-11-17:
Hi,

I have a setup with: 

* Raspberry Pi 3B+ running Raspbian Stretch, with OLA built from git
* [RDM isolated board with USB (FT245RL)](http://www.orangepi-dmx.org/raspberry-pi-rdm-controller/rdm-isolated-break-out-board), driven by a Raspberry Pi Zero, and connected to the Rpi 3B+ over USB

The SD card on the Raspberry Pi Zero, has been previously formatted for Raspbian (so it still shows "RECOVERY" and "boot" partitions); I have simply extracted the files in the zip linked on [4. Orange Pi Zero / Raspberry Pi : DMX USB Pro](http://www.orangepi-dmx.org/raspberry-pi-rdm-controller), the Raspberry Pi SDCard [zip-file](https://github.com/vanvught/rpidmx512-zip/blob/master/rpi_dmx_usb_pro.zip?raw=true), to the "boot" partition. Then I added a file `params.txt` manually on the boot partition, with contents as on http://www.orangepi-dmx.org/raspberry-pi-rdm-controller/configuration - except with `widget_mode=2`.

When I boot this system, and attach a HDMI monitor/display to the RPi Zero, I get this written on screen:

```none
[V2.3] Pi 3 Model B+ 1GB Compiled on Oct 5 2019 at 12:27:44
RDM Controller with USB [Compatible with Enttec USB Pro protocol], Widget mode : 2 (RDM)
Device UUID : 5000:eb500653, Label : Raspberry Pi DMX USB Pro
```

However, if I then power off the system, and replace `widget_mode=2` in `params.txt` on "boot" partition of the SD card with `widget_mode=3`, then insert the SD card in the RPi  Zero and boot again - then again I get exactly the same screen text as above, that is "Widget mode : 2 (RDM)", which means the `widget_mode=3` was ignored.

Where have I gone wrong in setting up this system? And what can I do, do get the widget in Widget Mode 3 (RDM Sniffer)?

vanvught, 2020-11-17:
Hi sdbbs, 

All files should be in the root directory of the FAT formatted partition. Please make that there is no `#` before `widget_mode=3`

[image]

> [V2.3] Pi 3 Model B+ 1GB Compiled on Oct 5 2019 at 12:27:44

That's not an Zero, but a Model 3 instead.

Please note that the preferred ARM board is the Orange Pi Zero LTS 256MB -> https://github.com/vanvught/rpidmx512-zip/issues/6#issuecomment-640098142

When you have an Orange Pi Zero LTS 256MB, then it would be interesting to know how the RDM discovery is going when using the Art-Net firmware. With this configuration, the discovery process is completely done in the Zero firmware and the result is then send to OLA. 

Thanks, Arjan

vanvught, 2020-11-29:
When you need further assistance, then please open an issue here -> https://github.com/vanvught/rpidmx512-zip

closed 2020-11-29 by vanvught