vanvught/rpidmx512 #166: LTC SMPTE:  Add support for MIDI Clock (BPM) view and generation

vanvught, 2020-12-22:
- When the source is MIDI or the source is rtpMIDI then show the BPM (MIDI Clock). 
- When the source is not MIDI and the source is not rtpMIDI, then make it possible to generate MIDI Clock.
- The MIDI System Realtime commands Clock, Start, Stop and Continue are controlled by means of plain text UDP and OSC commands.

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

brunchboy, 2020-12-22:
Nice, I will point people at this when they want a way to generate MIDI clock from Pioneer DJ equipment.

vanvught, 2020-12-22:
> Nice, I will point people at this when they want a way to generate MIDI clock from Pioneer DJ equipment.

We can also look at TCNet data packets. There is the BPM value available and then we can use this value for generating the MIDI clock automatically.

vanvught, 2020-12-23:
[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/5735090/orangepi_zero.uImage.gz)

[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/5735089/RemoteConfigUI.jar.zip)
[image]
[image]

commit 08faf342 (2020-12-24): Fixed #166
closed 2020-12-24 by vanvught