vanvught/rpidmx512 #179: Network: DHCP Client

vanvught, 2021-03-09:
Follow-up for https://github.com/vanvught/rpidmx512/issues/151#issuecomment-794105766
> > > Does firmware look once and once only for network?
> > 
> > 
> > Yes. Just at startup.
> 
> Hello Arjan,
> 
> Perhaps this should be changed? We have some environments where the DHCP is out of our control (and is quite slow to boot). So I could imagine that e.g. after a power surge the Orange Pi's would boot - fail to acquire a DHCP lease and be stuck with their locally assigned IP. Obviously this is not desirable. Attempting DHCP renewal or rebooting (if board has local IP) at X interval would be great for this (X perhaps even configurable) :-)
> 
> Best regards,
> Søren

cc: @soerenpedersen

vanvught, 2021-03-10:
Value in minutes between 0 and 5. Where 0 is no retry.

[image]

cc: @soerenpedersen

soerenpedersen, 2021-03-10:
Looks good! Looking forward to testing it out :-)

vanvught, 2021-03-10:
@soerenpedersen Hi Søren, herewith test firmware.

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/6118554/orangepi_zero.uImage.gz)

hippyau, 2021-03-10:
Can the MAC detect link up/down?

hippyau, 2021-03-10:
> Can the MAC detect link up/down?
>
> Yes. Just give it a try with disconnecting the Ethernet cable while
> booting. As soon as the DHCP goes is in 'renew' then plug in the Ethernet
> cable and you have DHCP running immediately.

Thanks, I was more thinking a link-goes-up event could triggers the DHCP
client.

vanvught, 2021-03-10:
The case for DHCP client retry is when there is a ink-up, but the DHCP is not reachable/available, yet.

vanvught, 2021-03-11:
> Perhaps it should be enabled by default when DHCP is configured?

That would break the IP Addressing Modes as described here -> https://github.com/vanvught/rpidmx512/issues/117#issuecomment-630290528

hippyau, 2021-03-11:
> then defaulting to the 2.x.y.z (based on MAC address as specified in Art-Net 4 "IP address configuration – Static Addressing").

I suggest is bad practice,  2.x.x.x is an internet route-able IP range,  feel sorry for the people on it :)   Should use 10.x,x,x /wherever possible/.

vanvught, 2021-03-11:
> I suggest is bad practice, 2.x.x.x is an internet route-able IP range, feel sorry for the people on it :) Should use 10.x,x,x /wherever possible/.

2.x.x.x is part of the Art-Net standard. And all my firmware is 100% compliant with any standards used in the lighting industry.

hippyau, 2021-03-11:
Understood,  but they also support 10.x.x.x and this complies with a
preceding network RFC-5735.   in my eyes btw! I don't really care.... not
your problem.... just saying :)
Hope you are well :)

soerenpedersen, 2021-03-11:
> > Perhaps it should be enabled by default when DHCP is configured?
> 
> That would break the IP Addressing Modes as described here -> [#117 (comment)](https://github.com/vanvught/rpidmx512/issues/117#issuecomment-630290528)

Ah, I see. Looking forward to testing out your firmware some more when my RJ45 connectors finally arrive for the 8x 4 board. :-)

vanvught, 2021-03-11:
>  8x 4 board. :-)

Cool! With feedback from other users I have made some great updates to the 8x 4U firmware. I hope to publish it this weekend.

I am also working on an simpler board with the 5M80ZE64C5N CPLD. The BOM is € 6,04

[image]

commit 8da1b174 (2021-03-13): Fixed #179
closed 2021-03-13 by vanvught