vanvught/rpidmx512 #184: Multiple SPI (WS2801/APA102/SK9822) ledstrips

Raaff, 2021-04-23:
Hi Arjan,

Is my assumption correct that the implementation of the Pixel8x prototype boards and firmware is only usable with 1 pin ledstrips (WS2812/SK6812), and not with ledstrips with a separate clock pin (WS2801/APA102/SK9822)?

Cheers!
Raaf

vanvught, 2021-04-23:
Hi Raaf,

That's correct; the current firmware is designed for RTZ signal type of chip's/led's only. 

The 74-series logic board (https://github.com/vanvught/h3dmx512-zip/tree/master/eagle/Orange-Pi-Zero/Pixel8x/P4) can only do RTZ.

The CPLD board (https://github.com/vanvught/h3dmx512-zip/tree/master/eagle/Orange-Pi-Zero/Pixel8x/5M80ZE64/P1) might be able to generate both Clock and Data signal. As there is little to no demand for this solution, I have not investigated the possibilities, yet.

Greets, Arjan

Raaff, 2021-04-23:
Thanks!
I'll be using SK9822 leds in a project because of their higher pwm-frequency and the possibility to dim them flicker free using the global brightness bits. I'll probably daisy chain more of them then.

vanvught, 2021-04-23:
I am already thinking on how to support the Clock/Data chips with the CPLD board. Just brainstorming; maybe we just need to expose the SPI Clock / 8 signal to an output of the CPLD. And a minor change in the H3 firmware.

vanvught, 2021-04-24:
Hi Raaf,

I have got the WS2801 working for the Pixel 8x CPLD board. 

For the APA102/SK9822, currently I have the GlobalBrightness set in the properties file. Is this ok with you? Or do we need to assign a DMX slot? 

Greets, Arjan

Raaff, 2021-04-24:
Wow, fantastic!

A DMX slot for the GlobalBrightness would be great, as this allows current control dimming (instead of PWM dimming) on the SK9822. This is really useful when using ledstrips as video lights, as PWM dimming can result in banding on video.

Did you change the schematics for this, or is it the same prototype board but with half the number of connectable led-strips when using SPI-ledstrips?

Cheers!
Raaf

vanvught, 2021-04-25:
Hi Raaf,

> Did you change the schematics for this, or is it the same prototype board but with half the number of connectable led-strips when using SPI-ledstrips?

The output is still 8x with a single clock, hence all outputs are updated in sync.

For the CPLD board, the CLOCK will be connected to the header on the right. From that header we can make a break-out board for the led-strips connections. The orange arrow shows the connection to made. Unfortunately, this board needs to be ordered with PCBWay as JLCPCB cannot source from DigiKey yet. JLCBC has not the CPLD in stock and also not the voltage regulator.

[image]

I have got also the 74-series logic board working. Using the DIP version for easy hacking. Added a single AND gateway for the clock signal. And getting the data from the RJ-45 output. I will update the SMD version and order that one at JLCPCB quickly. 

[image]
[image]

[Short demo](https://drive.google.com/file/d/1oS2GzIvA8L2NianoTtWsU3wNLy_8aAtX/view?usp=sharing)

Greets, Arjan

vanvught, 2021-05-02:
Waiting for new boards.

- 74 logic
[image]

- CPLD with break-out
[image]
[image]

Raaff, 2021-05-02:
Nice! Could you push (the CLPD version) to h3dmx512-zip git? I want to order that one too.

vanvught, 2021-05-09:
Hi Raaff,

> Could you push (the CLPD version) to h3dmx512-zip git? 

Just done. Added files for both jlcpcb.com (€35.29) and pcbway.com (€67.11)

- jlcpcb.com does not have the CPLD in stock
- pcbway.com is expensive with SMT assembly

[https://github.com/vanvught/h3dmx512-zip/tree/master/eagle/Orange-Pi-Zero/Pixel8x/5M80ZE64/P2](https://github.com/vanvught/h3dmx512-zip/tree/master/eagle/Orange-Pi-Zero/Pixel8x/5M80ZE64/P2)

vanvught, 2021-05-09:
Note: When pcbway.cpm can not source the original CPLD then replace it with : https://www.digikey.nl/product-detail/en/intel/5M80ZE64I5N/544-2817-ND/2660734
Digi-Key Part Number: 544-2817-ND
Manufacturer Part Number: 5M80ZE64I5N
Price: €2,05

vanvught, 2021-05-16:
The 74-logic prototype is working fine with SK9822. Working on the finishing touch for the code. Waiting for the CPLD board to receive. 

[image]

Raaff, 2021-05-16:
Looks promising! I'm waiting for the CPLD-boards as well. I ordered the PCB and the components separately, as I was afraid for availability issues. Soldering will be a challenge though.

vanvught, 2021-05-27:
I've got it working with the updated CPLD board. However, major GND issues with the yellow expansion board; just using one pin on the header for GND wasn't a good idea. Hence some GND taken from the RJ45 connector. 

[image]

vanvught, 2021-05-31:
[opi_emac_artnet_pixel_multi.zip](https://github.com/vanvught/rpidmx512/files/6571756/opi_emac_artnet_pixel_multi.zip)
[opi_emac_e131_pixel_multi.zip](https://github.com/vanvught/rpidmx512/files/6571757/opi_emac_e131_pixel_multi.zip)

vanvught, 2021-06-06:
Enhancement published [https://github.com/vanvught/h3dmx512-zip](https://github.com/vanvught/h3dmx512-zip)

Raaff, 2021-07-08:
Great, thanks a lot!

My CPLD-board is finished and it is able to output to 8 lines to SK9822 simultaneously with the test-patterns. On art-net output, I only get it working when I select only 1 output in devices.txt (active_out=1). If I increase the number of outputs (led_count=60, start_uni_port_1=1, start_uni_port_2=2, active_out=2), I get no clock or data output. I tried the opi_emac_artnet_pixel_multi-firmwares of May 31 and June 6, and even if I select a different pixel type (WS2812), it only works on either multiple strings with a test-pattern or a single strip on art-net.

Did you test CPLD-board also with art-net?
Maybe i configured devices.txt wrong, I've attached those files with the serial output of the Orange Pis below
[devices.txt, serial out.zip](https://github.com/vanvught/rpidmx512/files/6784290/devices.txt.serial.out.zip)

[image]

vanvught, 2021-07-08:
Hi Raaff! Great to so the board.

> Did you test CPLD-board also with art-net?

Just recently, and I have found that there was something wrong with the build system which I've fixed now.

I've attached the latest build. Please let me know how this works for you.

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/6784361/orangepi_zero.uImage.gz)

Raaff, 2021-07-08:
Yes, this fixes it! Fantastic.

BTW, is it intended behavior that it needs art-net data on all configured universes before it outputs pixel data at all? e.g. when configured for 8 universes on 8 ports, it won't output pixel data if only 7 art-net universes are received. (Not that it's a problem)

vanvught, 2021-07-08:
> Yes, this fixes it! Fantastic.

 :-)

> BTW, is it intended behavior that it needs art-net data on all configured universes before it outputs pixel data at all? e.g. when configured for 8 universes on 8 ports, it won't output pixel data if only 7 art-net universes are received. 

There are 2 scenario's with sending DMX universes (both sACN / Art-Net):

1. The pixels are updated when a `sync` is received. Preferred option for highest possible FPS.
2. The pixels are updated when the `last` Universe is received. That is the last Universe of the last active port. 
 
Thank you! ;-)

closed 2021-07-08 by vanvught

Raaff, 2021-07-08:
Ah, sounds obvious, as the strips need to be updated in sync.

Cheers :)

vanvught, 2021-07-08:
> Ah, sounds obvious, as the strips need to be updated in sync.

With my design, all the ports are updated without any jitter between the ports -> all ports are updated in a single SPI data stream. Hence I need a `sync` or a `last` universe for this single update process.