vanvught/rpidmx512 #187: LTC SMPTE: Network stops when source=ltc and ntp_enable=1

vanvught, 2021-05-28:
Reported by: @dmpw-jan

cc: @hippyau @rbarreiros @brunchboy @wyged @drummerclint

vanvught, 2021-05-30:
@dmpw-jan @hippyau @rbarreiros @brunchboy @wyged @drummerclint @ferrandis87

All, please will you give this build a good test. Many thanks in advance. Arjan

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/6566074/orangepi_zero.uImage.gz)

brunchboy, 2021-05-30:
I’ve tried to explain this before, I can’t test your builds, I don’t own any suitable hardware. I point my users at your project in my documentation, so some of them might be able to help (you’ve already tagged Clint, who is one who I know does have the right hardware), but tagging me accomplishes nothing. If you would like to create a topic in the Beat Link Trigger [Zulip stream](https://deep-symmetry.zulipchat.com/join/lp5i6yolkay7erzjzodx365q/) that would seem a valid place to make such requests as well (and perhaps make more people aware of this integration option).

vanvught, 2021-05-30:
@dmpw-jan @hippyau @rbarreiros @wyged @drummerclint @ferrandis87

All, please will you give this build a good test. Many thanks in advance. Arjan

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/6566650/orangepi_zero.uImage.gz)

vanvught, 2021-05-30:
@dmpw-jan @hippyau @rbarreiros @wyged @drummerclint @ferrandis87

All, please will you give this build a good test. Many thanks in advance. Arjan

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/6566909/orangepi_zero.uImage.gz)

vanvught, 2021-05-31:
@dmpw-jan @hippyau @rbarreiros @wyged @drummerclint @ferrandis87

All, please will you give this build a good test. Many thanks in advance. Arjan

This would be the last one here before publication.

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/6571755/orangepi_zero.uImage.gz)

vanvught, 2021-06-06:
Fix published [https://github.com/vanvught/h3dmx512-zip](https://github.com/vanvught/h3dmx512-zip)

closed 2021-06-06 by vanvught