vanvught/rpidmx512 #192: Compiling LTC SMTPE

hippyau, 2021-07-03:
Hey Arjan,

 Fresh clone today, and build is failing...

```
In file included from firmware/main.cpp:71:
./include/source.h: In static member function 'static void Source::Show(ltc::source, bool)':
./include/source.h:42:59: error: conversion from 'int' to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
   42 |    Display::Get()->SetCursorPos(Display::Get()->getCols() - 3, 3);
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
In file included from ../lib-rdm/include/rdmdeviceresponder.h:33,
                 from ../lib-rdmnet/include/rdmnetdevice.h:32,
                 from ../lib-rdmnet/include/rdmnetllrponly.h:31,
                 from firmware/main.cpp:88:
../lib-rdm/include/rdmdevice.h: In member function 'uint16_t RDMDevice::CalculateChecksum()':
../lib-rdm/include/rdmdevice.h:163:14: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
  163 |    nChecksum += static_cast<uint16_t>(m_tRDMDevice.aDeviceRootLabel[i]);
      |    ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../lib-rdmnet/include/rdmnetdevice.h:32,
                 from ../lib-rdmnet/include/rdmnetllrponly.h:31,
                 from firmware/main.cpp:88:
../lib-rdm/include/rdmdeviceresponder.h: In member function 'uint16_t RDMDeviceResponder::CalculateChecksum()':
../lib-rdm/include/rdmdeviceresponder.h:244:13: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
  244 |   nChecksum += m_tRDMDeviceInfo.current_personality;
      |   ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
../h3-firmware-template/Rules.mk:235: recipe for target 'build_h3/firmware/main.o' failed
make: *** [build_h3/firmware/main.o] Error 1
```

vanvught, 2021-07-03:
Hi Hip, which branch are working with `master` or `development`? What is your cross-compiler version? Dank, Arjan

hippyau, 2021-07-03:
Hey mate,  ```gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)```  on  `master`

vanvught, 2021-07-03:
Thanks Hip. Will work on it after the morning coffee

vanvught, 2021-07-03:
```
arjanvanvught@MacBook-Air ~ % cd tmp
arjanvanvught@MacBook-Air tmp % git clone https://github.com/vanvught/rpidmx512
Cloning into 'rpidmx512'...
remote: Enumerating objects: 41272, done.
remote: Counting objects: 100% (3988/3988), done.
remote: Compressing objects: 100% (2298/2298), done.
remote: Total 41272 (delta 2636), reused 2442 (delta 1438), pack-reused 37284
Receiving objects: 100% (41272/41272), 104.25 MiB | 2.35 MiB/s, done.
Resolving deltas: 100% (30081/30081), done.
arjanvanvught@MacBook-Air tmp % cd rpidmx512/opi_emac_ltc_smpte 
arjanvanvught@MacBook-Air opi_emac_ltc_smpte % make -f Makefile.H3 clean && make -f Makefile.H3
...
...
...
arm-none-eabi-objdump -D build_h3/main.elf | arm-none-eabi-c++filt > orangepi_zero.list
arm-none-eabi-size -A -x build_h3/main.elf
build_h3/main.elf  :
section         size         addr
.init          0x164   0x40000000
.text        0x31b10   0x40000168
.rodata       0x4360   0x40031c78
.ARM.exidx       0x8   0x40035fd8
.data           0x6c   0x40035fe0
.bss         0x2e7b8   0x40038000
Total        0x64800

arm-none-eabi-objcopy build_h3/main.elf -O binary orangepi_zero.img	
gzip -n -c orangepi_zero.img > orangepi_zero.img.gz 
mkimage -n 'http://www.orangepi-dmx.org' -A arm -O u-boot -T standalone -C none -a 0x40000000 -d orangepi_zero.img orangepi_zero.uImage
Image Name:   http://www.orangepi-dmx.org
Created:      Sat Jul  3 11:02:43 2021
Image Type:   ARM U-Boot Standalone Program (uncompressed)
Data Size:    221260 Bytes = 216.07 KiB = 0.21 MiB
Load Address: 40000000
Entry Point:  40000000
mkimage -n 'http://www.orangepi-dmx.org' -A arm -O u-boot -T standalone -C gzip -a 0x40000000 -d orangepi_zero.img.gz orangepi_zero.uImage.gz
Image Name:   http://www.orangepi-dmx.org
Created:      Sat Jul  3 11:02:43 2021
Image Type:   ARM U-Boot Standalone Program (gzip compressed)
Data Size:    116404 Bytes = 113.68 KiB = 0.11 MiB
Load Address: 40000000
Entry Point:  40000000
arjanvanvught@MacBook-Air opi_emac_ltc_smpte % 
```

hippyau, 2021-07-03:
Weird, something has changed between December and now...  I just compiled the stuff I was doing with the smpte and rgbpanel and it worked fine,  but can't get this latest to build...

```
hip@hip-Alienware-13:~/OrangePi/july21/rpidmx512/opi_emac_ltc_smpte$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
hip@hip-Alienware-13:~/OrangePi/july21/rpidmx512/opi_emac_ltc_smpte$ git pull
Already up to date.
```
```
hip@hip-Alienware-13:~/OrangePi/july21/rpidmx512/opi_emac_ltc_smpte$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/share/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: /mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/src/gcc/configure --target=arm-none-eabi --prefix=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native --libexecdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/lib --infodir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-native/arm-none-eabi --build=x86_64-linux-gnu --host=x86_64-linux-gnu --with-gmp=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-mpfr=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-mpc=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-isl=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-libelf=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Arm Embedded Toolchain 9-2020-q2-update' --with-multilib-list=rmprofile,aprofile
Thread model: single
gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)
```

History:
```
 2005  git pull
 2006  cd opi_emac_ltc_smpte/
 2007  make -f Makefile.H3 clean && make -f Makefile.H3

```

```
hip@hip-Alienware-13:~/OrangePi/july21/rpidmx512/opi_emac_ltc_smpte$ make -f Makefile.H3 clean && make -f Makefile.H3
[/mnt/d5e5c544-a90f-45fd-b1b3-d028b0ebdd07/OrangePi/july21/rpidmx512/opi_emac_ltc_smpte]
[../lib-remoteconfig ../lib-ltc ../lib-tcnet ../lib-midi ../lib-input ../lib-osc ../lib-ws28xxdisplay ../lib-ws28xx ../lib-rgbpanel ../lib-rdmnet ../lib-rdm ../lib-rdmsensor ../lib-rdmsubdevice ../lib-gps ../lib-artnet4 ../lib-artnet ../lib-e131 ../lib-uuid ../lib-spiflashinstall ../lib-spiflashstore ../lib-spiflash ../lib-network ../lib-lightset ../lib-properties ../lib-display ../lib-device ../lib-hal ../lib-c++ ../lib-debug ../lib-h3 ../lib-c ../lib-arm]
$PLATFORM_LIBGCC [-L /usr/share/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/thumb/v7ve+simd/hard/armv7-a/cortex-a7/hardfp/vfpv4 -L /usr/share/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/thumb/v7ve+simd/hard]
make -f Makefile.H3 clean 'PLATFORM=ORANGE_PI' 'MAKE_FLAGS=-DNODE_LTC_SMPTE -DNODE_ARTNET -DNODE_RDMNET_LLRP_ONLY -DARM_ALLOW_MULTI_CORE -DNDEBUG -DENABLE_SSD1311 -DENABLE_TC1602 -DENABLE_CURSOR_MODE -D_TIME_STAMP_YEAR_=2021 -D_TIME_STAMP_MONTH_=7 -D_TIME_STAMP_DAY_=3' -C ../lib-remoteconfig 
make[1]: Entering directory '/mnt/d5e5c544-a90f-45fd-b1b3-d028b0ebdd07/OrangePi/july21/rpidmx512/lib-remoteconfig'
[/mnt/d5e5c544-a90f-45fd-b1b3-d028b0ebdd07/OrangePi/july21/rpidmx512/lib-remoteconfig]
[src src/h3 src/h3/tftp src/debug]
$DEFINES [-DORANGE_PI  -DNDEBUG -D_TIME_STAMP_YEAR_=2021 -D_TIME_STAMP_MONTH_=7 -D_TIME_STAMP_DAY_=3 -DREMOTE_CONFIG]
$MAKE_FLAGS [-DNODE_LTC_SMPTE -DNODE_ARTNET -DNODE_RDMNET_LLRP_ONLY -DARM_ALLOW_MULTI_CORE -DNDEBUG -DENABLE_SSD1311 -DENABLE_TC1602 -DENABLE_CURSOR_MODE -D_TIME_STAMP_YEAR_=2021 -D_TIME_STAMP_MONTH_=7 -D_TIME_STAMP_DAY_=3]
$BUILD_DIRS [build_h3/src build_h3/src/h3 build_h3/src/h3/tftp build_h3/src/debug]
$TARGET [lib_h3/libremoteconfig.a ]
rm -rf build_h3
rm -rf lib_h3
....
....
[/mnt/d5e5c544-a90f-45fd-b1b3-d028b0ebdd07/OrangePi/july21/rpidmx512/opi_emac_ltc_smpte]
[... cut ...]
../lib-rdm/include/rdmdeviceresponder.h:244:13: error: conversion from 'int' to 'short unsigned int' may change value [-Werror=conversion]
  244 |   nChecksum += m_tRDMDeviceInfo.current_personality;
      |   ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
../h3-firmware-template/Rules.mk:235: recipe for target 'build_h3/firmware/main.o' failed
make: *** [build_h3/firmware/main.o] Error 1
hip@hip-Alienware-13:~/OrangePi/july21/rpidmx512/opi_emac_ltc_smpte$ 

```

hippyau, 2021-07-03:
Tried build_all,   getting these `error: conversion from 'int' to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]` all over the place :)

```
src/h3/shellcmd.cpp: In member function 'uint32_t Shell::hexadecimalToDecimal(const char*, uint32_t)':
src/h3/shellcmd.cpp:176:35: error: conversion from 'int' to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
  176 |   const uint8_t nNibble = c > '9' ? (c | 0x20) - 'a' + 10 : (c - '0');
      |                           ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/h3/shellcmd.cpp:176:35: error: conversion from 'int' to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
src/h3/shellcmd.cpp: In member function 'void Shell::CmdSet()':
src/h3/shellcmd.cpp:262:34: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
  262 |  uint32_t nLength = nArgv0Length + file::length::EXT;
      |                     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
src/h3/shell.cpp: In member function 'uint16_t Shell::ValidateCmd(uint32_t, shell::CmdIndex&)':
src/h3/shell.cpp:122:13: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
  122 |    return i + 1;
      |           ~~^~~
src/h3/shell.cpp: In member function 'void Shell::ValidateArg(uint16_t, uint32_t)':
src/h3/shell.cpp:155:30: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
  155 |     m_nArgvLength[j - 1] = i - nArgvStart;
      |                            ~~^~~~~~~~~~~~
cc1plus: all warnings being treated as errors
src/h3/shell.cpp:169:28: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
  169 |   m_nArgvLength[j - 1] = i - nArgvStart;
      |                          ~~^~~~~~~~~~~~
../h3-firmware-template/lib/Rules.mk:115: recipe for target 'build_h3/src/remoteconfigparams.o' failed

```

hippyau, 2021-07-03:
Can see conversion was enabled in May... Don't know why it is not hassling you though?

https://github.com/vanvught/rpidmx512/commit/86b10839f68eea4005c3fa1e3fb1d335fd0b898c#diff-b5a87bd3da8a9e1d6d5b0fea880c32a7e40cc98dff7028d3c428f3ad7ab20ca7L155

vanvught, 2021-07-03:
I am trying to understand why the `src/h3/shellcmd.cpp` does not give errors in my development system.

hippyau, 2021-07-03:
fyi, not limited to ```shellcmd.cpp```, there are others...

```In file included from src/remoteconfigparams.cpp:41:
../lib-properties/include/propertiesbuilder.h: In instantiation of 'bool PropertiesBuilder::Add(const char*, T, bool, uint32_t) [with T = bool; uint32_t = unsigned int]':
../lib-properties/include/propertiesbuilder.h:37:39:   required from here
../lib-properties/include/propertiesbuilder.h:55:11: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
   55 |   m_nSize += static_cast<uint16_t>(i);
      |   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```
src/h3/shell.cpp:155:30: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
  155 |     m_nArgvLength[j - 1] = i - nArgvStart;

```
```
n file included from src/remoteconfigparams.cpp:41:
../lib-properties/include/propertiesbuilder.h: In instantiation of 'bool PropertiesBuilder::Add(const char*, T, bool, uint32_t) [with T = bool; uint32_t = unsigned int]':
../lib-properties/include/propertiesbuilder.h:37:39:   required from here
../lib-properties/include/propertiesbuilder.h:55:11: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
   55 |   m_nSize += static_cast<uint16_t>(i);
      |   ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

vanvught, 2021-07-03:
I am using 10.x versions. But that should not matter? 

arjanvanvught@MacBook-Air ~ % arm-none-eabi-g++ --version
**arm-none-eabi-g++ (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)**
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

arjanvanvught@MacBook-Air ~ % ssh pi@nuc-i7
pi@nuc-i7's password: 
Welcome to Ubuntu 20.04.2 LTS (GNU/Linux 5.4.0-77-generic x86_64)

pi@nuc-i7:~$  arm-none-eabi-g++ --version
**arm-none-eabi-g++ (GCC) 10.2.0**
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

hippyau, 2021-07-03:
I commented out -Wconversion in `Rules.mk` and `lib/Rules.mk`, and also `-Wsign-conversion` in `lib/Rules.mk` and compiles a dream...

```
src/h3/shellcmd.cpp: In member function 'void Shell::CmdSet()':
src/h3/shellcmd.cpp:262:34: error: conversion to 'uint32_t' {aka 'unsigned int'} from 'int' may change the sign of the result [-Werror=sign-conversion]
  262 |  uint32_t nLength = nArgv0Length + file::length::EXT;
      |                     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors

```

vanvught, 2021-07-03:
> I am trying to understand why the src/h3/shellcmd.cpp does not give errors in my development system.

https://github.com/vanvught/rpidmx512/blob/master/lib-ltc/src/ltcdisplayrgbsetrgb.cpp#L108

`const auto nNibble = c > '9' ? static_cast<uint8_t>((c | 0x20) - 'a' + 10) : static_cast<uint8_t>(c - '0');`

and

src/h3/shellcmd.cpp:176:35: error: conversion from 'int' to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
  176 |   const uint8_t nNibble = c > '9' ? (c | 0x20) - 'a' + 10 : (c - '0');
      |                           ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

but all fine for the compile ....

```
arm-none-eabi-objdump -D build_h3/main.elf | arm-none-eabi-c++filt > orangepi_zero.list
arm-none-eabi-size -A -x build_h3/main.elf
build_h3/main.elf  :
section         size         addr
.init          0x164   0x40000000
.text        0x324b4   0x40000168
.rodata       0x4573   0x4003261c
.ARM.exidx       0x8   0x40036b90
.data           0x6c   0x40036b98
.bss         0x2e7b8   0x40038000
Total        0x653b7

arm-none-eabi-objcopy build_h3/main.elf -O binary orangepi_zero.img
gzip -n -c orangepi_zero.img > orangepi_zero.img.gz 
mkimage -n 'http://www.orangepi-dmx.org' -A arm -O u-boot -T standalone -C none -a 0x40000000 -d orangepi_zero.img orangepi_zero.uImage
Image Name:   http://www.orangepi-dmx.org
Created:      Sat Jul  3 11:34:34 2021
Image Type:   ARM U-Boot Standalone Program (uncompressed)
Data Size:    224260 Bytes = 219.00 KiB = 0.21 MiB
Load Address: 40000000
Entry Point:  40000000
mkimage -n 'http://www.orangepi-dmx.org' -A arm -O u-boot -T standalone -C gzip -a 0x40000000 -d orangepi_zero.img.gz orangepi_zero.uImage.gz
Image Name:   http://www.orangepi-dmx.org
Created:      Sat Jul  3 11:34:34 2021
Image Type:   ARM U-Boot Standalone Program (gzip compressed)
Data Size:    117998 Bytes = 115.23 KiB = 0.11 MiB
Load Address: 40000000
Entry Point:  40000000
pi@nuc-i7:/development/workspace/opi_emac_ltc_smpte$
```

vanvught, 2021-07-03:
I am confused ;-)

hippyau, 2021-07-03:
¯\_(ツ)_/¯  is pretty weird.  I have a typical compiler setup, nothing unusual.   I might try looking to update my compiler, could be compiler bug.

vanvught, 2021-07-03:
> ¯_(ツ)_/¯ is pretty weird. I have a typical compiler setup, nothing unusual. I might try looking to update my compiler, could be compiler bug.

I am about to install a new Xubuntu with your `gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)`

vanvught, 2021-07-03:
Got it replicated.

```
In file included from firmware/main.cpp:69:
./include/source.h: In static member function 'static void Source::Show(ltc::source, bool)':
./include/source.h:42:59: error: conversion from 'int' to 'uint8_t' {aka 'unsigned char'} may change value [-Werror=conversion]
   42 |    Display::Get()->SetCursorPos(Display::Get()->getCols() - 3, 3);
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
In file included from ../lib-rdm/include/rdmdeviceresponder.h:33,
                 from ../lib-rdmnet/include/rdmnetdevice.h:32,
                 from ../lib-rdmnet/include/rdmnetllrponly.h:31,
                 from firmware/main.cpp:86:
../lib-rdm/include/rdmdevice.h: In member function 'uint16_t RDMDevice::CalculateChecksum()':
../lib-rdm/include/rdmdevice.h:163:14: error: conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} may change value [-Werror=conversion]
  163 |    nChecksum += static_cast<uint16_t>(m_tRDMDevice.aDeviceRootLabel[i]);
      |    ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../lib-rdmnet/include/rdmnetdevice.h:32,
                 from ../lib-rdmnet/include/rdmnetllrponly.h:31,
                 from firmware/main.cpp:86:
../lib-rdm/include/rdmdeviceresponder.h: In member function 'uint16_t RDMDeviceResponder::CalculateChecksum()':
../lib-rdm/include/rdmdeviceresponder.h:244:13: error: conversion from 'int' to 'short unsigned int' may change value [-Werror=conversion]
  244 |   nChecksum += m_tRDMDeviceInfo.current_personality;
      |   ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make: *** [../h3-firmware-template/Rules.mk:235: build_h3/firmware/main.o] Error 1
pi@debian-parallels:/tmp/rpidmx512/opi_emac_ltc_smpte$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pi@debian-parallels:/tmp/rpidmx512/opi_emac_ltc_smpte$ 
```

vanvught, 2021-07-03:
Hi Hip, please will you give the `development` branch a good test? Dank, Arjan

hippyau, 2021-07-05:
Confirmed ```development``` branch is now compiling with -Wconversion and -Wsignconversion :+1:

commit 23d548b7 (2021-07-03): Fixed #192
closed 2021-07-05 by hippyau