vanvught/rpidmx512 #194: RGB Panel: Some weird flickering not present previously

hippyau, 2021-07-03:
Hi Arjan,

I notice some ghost pixels in the latest version that were not present in the firmware I was previously using (Version 2.8 compiled November 12 2020).   See that line in the middle.

Can see in the image the line of pixels, and also on the last row of `A` and `r` there are blue pixels.  When running TC, there is row offset by a pixel,  kind of like tearing in the image.

Running older firmware (experimental branch) the issue is not present.
 
Also, a small issue...  Master brightness not applied to whole panel, just the Digit/Colon

Only the top line responded to Master brightness setting.  Hard to see, but top line is master brightness at 8%,  other lines are not effected.

[image]

hippyau, 2021-07-03:
I just realized the whole line is missing too :)  

This is older firmware... nice & sharp.   Exactly the same text file configs.

[image]

vanvught, 2021-07-03:
Same here with latest build; ghost pixels not that sharp.

> Also, a small issue... Master brightness not applied to whole panel, just the Digit/Colon

Works as designed ;-)

[image]

hippyau, 2021-07-03:
That is my legacy :)

vanvught, 2021-07-03:
Got it..fixed!

closed 2021-07-03 by vanvught

hippyau, 2021-07-03:
Confirmed, and working perfectly....  you are a legend!

[image]

Hahaha! :100:

commit 64ce599c (2021-07-03): Fixed #194