vanvught/rpidmx512 #197: [super, absolutely lowest priority] RemoteUI to default to a physical wire / radio interface...

hippyau, 2021-07-15:
Minor annoyance,  interface defaults to docker in this example,  maybe because alphabetic reasons(?), but great is a 'real' interface was recognized before virtual ones (tricky!). 

[image]

```
hip@hip-Alienware-13:~$ ifconfig
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:20:b4:5e:ff  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

enp2s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.49  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::f10f:a22d:2bf7:9a73  prefixlen 64  scopeid 0x20<link>
        ether 34:e6:d7:3e:b6:95  txqueuelen 1000  (Ethernet)
        RX packets 13354587  bytes 8623509305 (8.6 GB)
        RX errors 0  dropped 34213  overruns 0  frame 0
        TX packets 5318554  bytes 594621946 (594.6 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 19  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 9526728  bytes 2619457002 (2.6 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 9526728  bytes 2619457002 (2.6 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

vanvught, 2021-07-15:
Hi Hip, it is fixed in yesterday's commit -> https://github.com/vanvught/h3dmx512-zip
I've removed the Network selection ;-)

closed 2021-07-15 by vanvught

hippyau, 2021-07-15:
Wow, you read my mind :)