vanvught/rpidmx512 #223: Portability issues

rbalykov, 2021-11-04:
Hello all.

First, have you seen this: https://github.com/Allwinner-Homlet/H3-BSP4.4-brandy
I think u-boot with netboot support gonna be much better than dancing with SD flashing.

Second, why not using CMSIS and/or RTOS? When using multicore CPU, semaphores and mutexes become real must-have.

vanvught, 2021-11-04:
There is no portability issue.  The firmware runs on Allwinner SoC H2+/H3, Broadcom SoC's and on Linux/MacOS. And soon on the GigaDevice GD32F20x family (https://github.com/vanvught/GD32F20x-Bootloader-TFTP). There is a very lightweight hardware abstraction layer. 

> I think u-boot with netboot support gonna be much better than dancing with SD flashing.

The SDCard is just used once -> https://www.orangepi-dmx.org/orange-pi/remote-configuration-management-for-spi-flash/spi-flash-boot-process

> why not using CMSIS 

Not an added value. It is more nice to have.

> or RTOS?

Not needed. All firmware is single core. Except for the LTC where a second core used when using a RGB panel.

> , semaphores and mutexes become real must-have.

Not with the firmware where there is no multi-core interaction.

closed 2021-11-04 by vanvught

vanvught, 2021-11-04:
There is no portability issue. The firmware runs on Allwinner SoC H2+/H3, Broadcom SoC's and on Linux/MacOS. And soon on the GigaDevice GD32F20x family (https://github.com/vanvught/GD32F20x-Bootloader-TFTP). There is a very lightweight hardware abstraction layer.

I think u-boot with netboot support gonna be much better than dancing with SD flashing.
The SDCard is just used once -> https://www.orangepi-dmx.org/orange-pi/remote-configuration-management-for-spi-flash/example-java-ui/workflow-firmware-installation

why not using CMSIS
Not an added value. It is more nice to have. Moreover, Allwinner does not provide a CMSIS implementation.

or RTOS?
Not needed. RTOS is just giving overhead. And introducing third party libraries.

, semaphores and mutexes become real must-have.
Not with the firmware where there is no multi-core interaction.