vanvught/rpidmx512 #227: Using the OrangePiZero as a recorder then as player

MarcHagen, 2021-12-04:
Hi,

I've been fiddling around with the project and works great as a Artnet node.
But in a upcoming hobby project i would like to have a "headless" dmx player to spuw out DMX .
There is a player option in the current project! Great!

Now. how to record then? I have a light console where i have to program the show in. and then record the physical dmx output to a file.
Is this possible in the current project state?

hippyau, 2021-12-04:
I have not done myself, but believe you would  playback OLA recorder files
(part of the OLA project...
https://docs.openlighting.org/ola/man/man1/ola_recorder.1.html).

These can be triggered by OSC or ArtNet with the DMX Showfile Player
firmware.... https://www.orangepi-dmx.org/8-dmx-showfile-player

MarcHagen, 2021-12-04:
Yes i found that part, i just need a way to record from a DMX input.

hippyau, 2021-12-04:
Yeah, currently you use OLA recorder on a Linux installation to capture
your show...

You can then playback these files back on the OrangePi.

Recording would be a nice feature @Arjan van Vught
***@***.***> :)

MarcHagen, 2021-12-04:
Mm yeah well that the issue, i dont have a dmx usb interface (except  the orangepi) nor have i a linux install.
But if its not currently in the project then i have to find another solution,

but this would indeed be a good feature for the project :)

hippyau, 2021-12-04:
In a spot, without a USB device you can no doubt capture (network) ArtNet
or sACN with OLA recorder to file, is configured correctly

mrv96, 2023-12-02:
Upvote

vanvught, 2023-12-07:
It is on my radar and todo-list. I hope to find some good free time soon. Thanks. Arjan.

vanvught, 2024-05-25:
URL http://allwinner_34020e.local./showfile

[image]
[image]

closed 2024-05-25 by vanvught

MarcHagen, 2024-05-25:
That is one massive update! Thank you very much ❤️