vanvught/rpidmx512 #228: ArtNet Output only works when active_out set to 1 with 8X4 CPLD Board

mattg66, 2021-12-04:
I have 8 WS2815 strips connected to the CPLD board and the test patterns works correctly across all of the strips, however when trying to control via ArtNet, no output is presented to anything unless the active_out parameter is commented out and only the first output is used, in which case only the strip connected to the first output does anything as expected.

Any ideas what the problem could be?

vanvught, 2021-12-04:
Hi Matt,

There is output when the last universe is received **or** when then there is an ArtSync received. The latter is the preferred option when designing the perfect light show.

The `active_out` parameter tells how many ports are in use; how many ports have connected strips.

When all 8 ports have connected strips then `active_out=8` should be used.

With Art-Net limitation where universes are divided in pages, the last universe to be send (when not using ArtSync) is easily to see with the latest sample Java UI.

Hope this helps. Greetings, Arjan

[image]

mattg66, 2021-12-04:
Hi Arjan,

Thanks for this, the problem was because i disabled the last universe from MagicQ.
Expect a donation over the next few days and some photos of the result.

Thanks

closed 2021-12-04 by mattg66