vanvught/rpidmx512 #234: Library Not Found

kenchung285, 2022-02-14:
Excuse me, I'd like to learn how to write I2C with RPI 4 with the files "pca9685.h" and "pca9685.cpp" written by you. But I cannot figure out the following problems : 

1. In rpidmx512/lib-pca9685/pca9685.cpp, you include header files "hal_i2c.h" and "hal_gpio.h". Where can I download these two header files on my RPI 4 ?
2. I cannot find the function definition of "FUNC_PREFIX()" and "i2c_write()" in line 423, rpidmx512/lib-pca9685/pca9685.cpp

Thank you very much for answering my questions.

vanvught, 2022-02-14:
> In rpidmx512/lib-pca9685/pca9685.cpp, you include header files "hal_i2c.h" and "hal_gpio.h". Where can I download these two header files on my RPI 4 ?

The include file can be found here https://github.com/vanvught/rpidmx512/tree/master/lib-hal/include
And are included in de make file https://github.com/vanvught/rpidmx512/blob/master/lib-pca9685/Makefile.Linux#L4

> I cannot find the function definition of "FUNC_PREFIX()" and "i2c_write()" in line 423, rpidmx512/lib-pca9685/pca9685.cpp

The `FUNC_PREFIX(x) ` is defined here https://github.com/vanvught/rpidmx512/blob/master/lib-hal/include/linux/hal_api.h#L30

closed 2022-02-14 by vanvught

hippyau, 2022-02-15:
Hi @kenchung285,  you will find this is not typical Arduino or Linux/RPI compatible code, but specifically for the bare-metal OS Arjan has provided here. 

It will need some adaption to work with other platforms I2C libraries.

kenchung285, 2022-02-16:
Hi @vanvught @hippyau , I see, and thank you for your answers. Actually I've seen someone using the code for RPI and pca9685, so I consider that it's compatible with those RPi with an OS supported by the bcm2835 library.