vanvught/rpidmx512 #243: oPi LTC integration with TimeLord

drummerclint, 2022-04-29:
I know this is more a timelord thing but I noticed that Timelord can send data to the opi ltc.  I noticed the menu option but haven't had any success.  My opi is configured for the (8) 8x8 matrix led display, is it possible that will work in this case?  If so, how would I configure it? @hippyau 
Thanks for the amazing work!  I'm excited to be able to have shows to use this stuff on!

hippyau, 2022-05-02:
Hi @drummerclint  :)

Well, it worked with the UDP messages. Is your oPI on the same network range?  It should have just worked,  I used it with WS28xx though the messages were not particular to that implementation.  
Unfortunately, to be honest I had a major IT disaster and lost two months of coding on TimeLord and that also lost me the oPI extention.  I have not worked on TL6 for a while now :( 

Unless anything radical changed on the oPI though, I thought it should work, because it's just sending UDP messages that the LTC firmware would listen to.

drummerclint, 2022-05-02:
Hey @hippyau! :)

I'm so sorry to hear about your disaster.  That's devesting :( 

I have both the timelord PC and the opi on the same network segment.  The only way I've been able to transmit tc to it is via rtp-midi.  Both artnet tc and opi smpte don't work.  I have a feeling there's something deeper preventing this.  I've busted out wireshark and it appears those opi packets aren't getting sent out to the opi's ip, which is set correctly in the settings.  Is there a specific port the OSC server needs to be set to (mine is on 8000).   I updated the software from previous versions and I'm wondering if there's something going on in the config.  I've noticed the config is stored in the Public directory as opposed to my actual user directory.  I'm tempted to reinstall the software cleanly.  Do I need to do anything in order to maintain the license? I've tested artnet timecode and neither opi or artnettimecodeviewer are receiving.  TL is running as my user and also as administrator.  Firewall is disabled.
I've included a logfile.  Please let me know if I'm missing something, also if there's a better place we can discuss.

Thanks so much!

[TimeLord-Logfile.txt](https://github.com/vanvught/rpidmx512/files/8601267/TimeLord-Logfile.txt)

vanvught, 2022-08-04:
@hippyau Can I assist here?

hippyau, 2022-08-15:
I would have used the default port for LTC UDP messages, I recall sending !start commands etc to the OrangePI via UDP string.

Beyond that I'd have to reverse engineer the binary because of an IT disaster that happened just about that time I published the latest installer, and lost a bit of code - including this code :( .

is there a .txt file option to enable these UDP messages, because it did work... but there could be several things going on here like network or I did something dumb, and nobody but me had tested :D

drummerclint, 2022-08-17:
I noticed that none of the networking was working with that version of timelord on my computer.  I have a new laptop now and I'll try it again with my new one.  Even unlicensing the software didn't work :/  I had to reinstall an older version to unregister the software.  I made sure all firewalls were turned off and everything too.

vanvught, 2023-04-29:
Please re-open if when needed. Thanks, Arjan

closed 2023-04-29 by vanvught