vanvught/rpidmx512 #246: Implement the showfile player for all Art-Net / sACN E1.31 firmware [DMX / Pixel]

vanvught, 2022-06-08:
@batecam79

mrv96, 2023-12-02:
Upvote

vanvught, 2023-12-07:
It is on my radar and todo-list. I hope to find some good free time soon. Thanks. Arjan.

vanvught, 2024-02-13:
I can proudly announce that I have implemented the show file player for all Art-Net / sACN E1.31 firmware.
The latest development build is attached. Also the update Remote Configuration UI.
There is also a web page at `/showfile`.
Make sure you have a SDCard inserted at startup. There are some sample files in `showfiles.zip`
Please let me know when there are any issues. Thanks, Arjan

[image]
[image]
[image]
[image]

[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/14270867/RemoteConfigUI.jar.zip)
[firmware-e131.zip](https://github.com/vanvught/rpidmx512/files/14270868/firmware-e131.zip)
[firmware-artnet.zip](https://github.com/vanvught/rpidmx512/files/14270869/firmware-artnet.zip)
[showfiles.zip](https://github.com/vanvught/rpidmx512/files/14270870/showfiles.zip)

hippyau, 2024-02-13:
> <https://github.com/vanvught/rpidmx512/assets/4319913/2e5db232-9243-4d60-89fe-e9f4991439a0> Scherm.afbeelding02.31.png
> <https://github.com/vanvught/rpidmx512/assets/4319913/e9d05663-7f28-4cfb-9803-55706a2a6f1e> Scherm.afbeelding01.56.png
> <https://github.com/vanvught/rpidmx512/assets/4319913/c1e455b4-e73e-4a8c-97ed-78eaff421ea2> Scherm.afbeelding04.50.png

Awesome :) 😀  I'll check it out!

closed 2024-02-23 by vanvught