vanvught/rpidmx512 #261: build_rpi-firmware.sh - compile error- In member function 'void Dmx::ClearData(uint32_t)

netmindz, 2023-04-14:
fetch-loop-arrays  -pedantic -fno-exceptions -fno-unwind-tables -fno-rtti -fno-threadsafe-statics -std=c++11 -nostdinc++ -c src/rpi/dmx.cpp -o build/src/rpi/dmx.o
src/rpi/dmx.cpp:360:59: error: FP registers might be clobbered despite 'interrupt' attribute: compile with '-mgeneral-regs-only' [-Werror=attributes]
  360 | static void __attribute__((interrupt("FIQ"))) fiq_dmx(void) {
      |                                                           ^
src/rpi/dmx.cpp: In member function 'void Dmx::ClearData(uint32_t)':
src/rpi/dmx.cpp:743:36: error: expression does not compute the number of elements in this array; element type is 'Data', not 'uint32_t' {aka 'unsigned int'} [-Werror=sizeof-array-div]
  743 |         auto i = sizeof(s_DmxData) / sizeof(uint32_t);
      |                  ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
src/rpi/dmx.cpp:743:38: note: add parentheses around 'sizeof (uint32_t)' to silence this warning
  743 |         auto i = sizeof(s_DmxData) / sizeof(uint32_t);
      |                                      ^~~~~~~~~~~~~~~~
      |                                      (               )
src/rpi/dmx.cpp:80:20: note: array 's_DmxData' declared here
   80 | static struct Data s_DmxData[buffer::INDEX_ENTRIES] ALIGNED;
      |                    ^~~~~~~~~
cc1plus: all warnings being treated as errors
make[1]: *** [../firmware-template-bcm/lib/Rules.mk:110: build/src/rpi/dmx.o] Error 1

netmindz, 2023-04-14:
If I suppress the warnings as errors then it fails with

make[1]: Leaving directory '/home/will/netmindz/rpidmx512/lib-arm'
arm-none-eabi-ld build/vectors.o   build/firmware/main.o -Map kernel.map -T ./../firmware-template-bcm/memmap -o build/main.elf -L../lib-widget/lib -L../lib-dmx/lib -L../lib-rdm/lib -L../lib-usb/lib -L../lib-network/lib -L../lib-display/lib -L../lib-hal/lib -L../lib-network/lib -L../lib-properties/lib -L../lib-device/lib -L../lib-hal/lib -L../lib-c++/lib -L../lib-debug/lib -L../lib-c/lib -L../lib-bcm2835/lib -L../lib-arm/lib -lwidget -ldmx -lrdm -lusb -lnetwork -ldisplay -lhal -lnetwork -lproperties -ldevice -lhal -lc++ -ldebug -lc -lbcm2835 -larm -L /usr/lib/gcc/arm-none-eabi/12.2.0/arm/v5te/hard/armv7-a/cortex-a7/hardfp/vfpv4 -L /usr/lib/gcc/arm-none-eabi/12.2.0/arm/v5te/hard -lgcc
arm-none-eabi-ld: warning: build/main.elf has a LOAD segment with RWX permissions
arm-none-eabi-ld: ../lib-network/lib/libnetwork.a(network.o): in function `Network::MacAddressCopyTo(unsigned char*)':
network.cpp:(.text+0x30): undefined reference to `mac_address_get(unsigned char*)'
arm-none-eabi-ld: ../lib-hal/lib/libhal.a(ff.o): in function `f_printf':
ff.c:(.text+0x33d8): undefined reference to `strlen'
make: *** [../firmware-template-bcm/Rules.mk:133: build/main.elf] Error 1
Error :  ../rpi_dmx_usb_pro

vanvught, 2023-04-14:
Please see https://github.com/vanvught/rpidmx512-zip/issues/5#issuecomment-536272513

closed 2023-04-14 by vanvught

vanvught, 2023-04-15:
Hi,

Please check the `experimental` branch. That will build with `build_rpi-firmware.sh`

@netmindz

netmindz, 2023-04-16:
Nope, same need for suppression of warnings and same error about strlen

netmindz, 2023-04-16:
`diff --git a/lib-bcm2835/src/irq_timer.c b/lib-bcm2835/src/irq_timer.c
index 30fbe2a5..406c41d2 100644
--- a/lib-bcm2835/src/irq_timer.c
+++ b/lib-bcm2835/src/irq_timer.c
@@ -37,7 +37,10 @@
 static /*@null@*/ thunk_irq_timer_t timer1_func = NULL;
 static /*@null@*/ thunk_irq_timer_t timer3_func = NULL;
 
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wattributes"
 static void __attribute__((interrupt("IRQ"))) irq_timer_handler(void) {
+#pragma GCC diagnostic pop
        dmb();
 
        const uint32_t irq_timer_micros = BCM2835_ST->CLO;
diff --git a/lib-dmx/src/rpi/dmx.cpp b/lib-dmx/src/rpi/dmx.cpp
index 010921bc..3cce48db 100644
--- a/lib-dmx/src/rpi/dmx.cpp
+++ b/lib-dmx/src/rpi/dmx.cpp
@@ -357,7 +357,10 @@ void fiq_dmx_out_handler(void) {
        }
 }
 
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wattributes"
 static void __attribute__((interrupt("FIQ"))) fiq_dmx(void) {
+#pragma GCC diagnostic pop
        dmb();
 
 #ifdef LOGIC_ANALYZER
@@ -740,7 +743,10 @@ uint32_t Dmx::GetUpdatesPerSecond(__attribute__((unused))uint32_t nPortIndex) {
 }
 
 void Dmx::ClearData(__attribute__((unused))uint32_t nPortIndex) {
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wsizeof-array-div"
        auto i = sizeof(s_DmxData) / sizeof(uint32_t);
+#pragma GCC diagnostic pop
        auto *p = reinterpret_cast<uint32_t *>(s_DmxData);
 
        while (i-- != 0) {
`

vanvught, 2023-04-16:
Working fine with:

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GCC) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
/workspace/scripts$ ./build_rpi-firmware.sh 
[../rpi_dmx_usb_pro]
[../lib-widget/lib/libwidget.a ../lib-dmx/lib/libdmx.a ../lib-rdm/lib/librdm.a ../lib-usb/lib/libusb.a ../lib-network/lib/libnetwork.a ../lib-display/lib/libdisplay.a ../lib-hal/lib/libhal.a ../lib-network/lib/libnetwork.a ../lib-properties/lib/libproperties.a ../lib-device/lib/libdevice.a ../lib-hal/lib/libhal.a ../lib-c++/lib/libc++.a ../lib-debug/lib/libdebug.a ../lib-c/lib/libc.a ../lib-bcm2835/lib/libbcm2835.a ../lib-arm/lib/libarm.a]

arm-none-eabi-ld build7/vectors.o   build7/firmware/main.o -Map kernel7.map -T ./../firmware-template-bcm/memmap -o build7/main.elf -L../lib-widget/lib7 -L../lib-dmx/lib7 -L../lib-rdm/lib7 -L../lib-usb/lib7 -L../lib-network/lib7 -L../lib-display/lib7 -L../lib-hal/lib7 -L../lib-network/lib7 -L../lib-properties/lib7 -L../lib-device/lib7 -L../lib-hal/lib7 -L../lib-c++/lib7 -L../lib-debug/lib7 -L../lib-c/lib7 -L../lib-bcm2835/lib7 -L../lib-arm/lib7 -lwidget -ldmx -lrdm -lusb -lnetwork -ldisplay -lhal -lnetwork -lproperties -ldevice -lhal -lc++ -ldebug -lc -lbcm2835 -larm -L /opt/gnuarm-hardfp/lib/gcc/arm-none-eabi/10.2.0/armv7-a/cortex-a7/hardfp/vfpv4 -L /opt/gnuarm-hardfp/lib/gcc/arm-none-eabi/10.2.0 -lgcc
arm-none-eabi-objdump -D build7/main.elf | arm-none-eabi-c++filt > kernel7.list
arm-none-eabi-size -A build7/main.elf
build7/main.elf  :
section        size     addr
.init           340    32768
.text         62928    33120
.rodata        6000    96048
.ARM.exidx        8   102048
.data           180   102056
.bss          49264   114688
Total        118720
```

```
arm-none-eabi-objcopy build7/main.elf -O binary kernel7.img
/workspace/scripts
-rwxrwxr-x 1 pi pi 69468 apr 16 15:23 ./rpi_dmx_usb_pro/kernel7.img
-rwxrwxr-x 1 pi pi 69100 apr 16 15:23 ./rpi_dmx_usb_pro/kernel.img
```

netmindz, 2023-04-16:
Hi @vanvught   - Your changes fix the warnings as errors, but still have the strlen error, so guessing I've got a different version of one of the dependencies, can you post your binaries somewhere for me to test?

netmindz, 2023-04-16:
arm-none-eabi-gcc --version
arm-none-eabi-gcc (Fedora 12.2.0-1.fc37) 12.2.0

vanvught, 2023-04-17:
I have installed Fedora 37. And I am to reproduce the `strlen` error.

```
arm-none-eabi-ar: creating lib7/libarm.a
arm-none-eabi-objdump -D lib7/libarm.a | arm-none-eabi-c++filt > lib7/lib7.list
make[1]: Leaving directory '/mnt/workspace/lib-arm'
arm-none-eabi-ld build/vectors.o   build/firmware/main.o -Map kernel.map -T ./../firmware-template-bcm/memmap -o build/main.elf -L../lib-widget/lib -L../lib-dmx/lib -L../lib-rdm/lib -L../lib-usb/lib -L../lib-network/lib -L../lib-display/lib -L../lib-hal/lib -L../lib-network/lib -L../lib-properties/lib -L../lib-device/lib -L../lib-hal/lib -L../lib-c++/lib -L../lib-debug/lib -L../lib-c/lib -L../lib-bcm2835/lib -L../lib-arm/lib -lwidget -ldmx -lrdm -lusb -lnetwork -ldisplay -lhal -lnetwork -lproperties -ldevice -lhal -lc++ -ldebug -lc -lbcm2835 -larm -L /usr/lib/gcc/arm-none-eabi/12.2.0/arm/v5te/hard/armv7-a/cortex-a7/hardfp/vfpv4 -L /usr/lib/gcc/arm-none-eabi/12.2.0/arm/v5te/hard -lgcc
arm-none-eabi-ld: warning: build/main.elf has a LOAD segment with RWX permissions
arm-none-eabi-ld: ../lib-hal/lib/libhal.a(ff.o): in function `f_printf':
ff.c:(.text+0x33d8): undefined reference to `strlen'
make: *** [../firmware-template-bcm/Rules.mk:136: build/main.elf] Error 1
Error :  ../rpi_dmx_usb_pro

[@fedora scripts]$arm-none-eabi-gcc --version
arm-none-eabi-gcc (Fedora 12.2.0-1.fc37) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
```

netmindz, 2023-04-17:
Reproduction is half way to solving :)

netmindz, 2023-04-17:
Ok, that compiles for me now, just need to test actual output

netmindz, 2023-04-17:
Sadly neither QLC+ nor OLA happy, OLA gives

plugins/usbpro/UsbProWidgetDetector.cpp:323: USB Widget didn't respond to messages, esta id 0, device id 0
plugins/usbpro/UsbProWidgetDetector.cpp:326: Is device in USB Controller mode if it's a Goddard?
plugins/usbpro/WidgetDetectorThread.cpp:389: trying stage 1 for 0x56048b44b800
plugins/usbpro/WidgetDetectorThread.cpp:386: no more detectors to try for  0x56048b44b800

netmindz, 2023-04-17:
[image]

netmindz, 2023-04-17:
[image]

vanvught, 2023-04-17:
> Sadly neither QLC+ nor OLA happy, OLA gives

Please make sure you are powering the RPi from the BitWizard board.

Verify that the correct USB device is configured in OLA (QLC+).

The image at [https://github.com/vanvught/rpidmx512-zip](https://github.com/vanvught/rpidmx512-zip) is known as working perfectly.

netmindz, 2023-04-18:
So I have the board mounted like this
[image]
And I connect my laptop to the usb port on the add-on board yes?

netmindz, 2023-04-19:
The power led on the pi is bright, but very very dim on the add-on board. Is that normal?

commit 302aa4f0 (2023-04-16): Fixed #261

arm-none-eabi-objcopy build/main.elf -O binary kernel.img
arm-none-eabi-gcc -mfpu=neon-vfpv4 -march=armv7-a -mtune=cortex-a7 -DRPI2 -DBARE_METAL -DNO_EMAC -DSD_WRITE_SUPPORT -DNO_EMAC -DRDM_CONTROLLER -DUPDATE_CONFIG_FILE -DCONSOLE_FB -DDISABLE_RTC -DENABLE_MMU -DNDEBUG -I../include -I./include -I../lib-widget/include -I../lib-dmx/include -I../lib-rdm/include -I../lib-usb/include -I../lib-network/include -I../lib-display/include -I../lib-hal/include -I../lib-network/include -I../lib-properties/include -I../lib-device/include -I../lib-hal/include -I../lib-c++/include -I../lib-debug/ [... cut]