vanvught/rpidmx512 #269: LTC internal generator issue

pablogfdez, 2023-09-21:
Hi.

I hope you can help me.

I've connected a front panel based on MCP23017 to use it as source select as described in your website.

The problem is that when i select the "Internal" source from this controls the counter sometimes slows down and time showed don't correspond with "real" time.

This happens randomly, but more often when "LTC" was previously selected.

When i select "internal" by editting "ltc.txt" file it works as expected.

The assembly has both audio and midi pcbs fitted.

Any idea on what could be going on?

Regards.
--Pablo

vanvught, 2023-09-21:
Pablo, 

Which version are you using?

> This happens randomly, but more often when "LTC" was previously selected.

That is pretty weird. It does not sound logical.

- Arjan

pablogfdez, 2023-09-21:
Thank you for your fast response.

I'm using the latest compiled image available in this git.

I know it's weird, but please see the attached video.

I'll try to test with a different orange pi zero this weekend to discard any problem with it.

https://github.com/vanvught/rpidmx512/assets/145710427/426cceef-31fb-4461-90e2-342c4cf2209a

vanvught, 2023-09-21:
Hi Pablo,

Thanks for the video. Nice setup!

I will work on this issue in the weekend.

- Arjan

pablogfdez, 2023-09-21:
Thanks. Feel free to ask if you need me to do any test.

All hardware is based in your gerbers.
Front and back panels are 2.5mm laser cut acrylic attached to an aluminum enclosure. I want to build all the system based on a single PCB in the future. I'll send project files to you if it works but at this moment I'm pretty busy with my own projects.

--Pablo

vanvught, 2024-08-30:
@pablogfdez Hi Pablo, please check https://github.com/vanvught/h3dmx512-zip for the latest firmware. Re-open this issue when needed. Thanks, Arjan

closed 2024-08-30 by vanvught