vanvught/rpidmx512 #270: DMX Merger

pilo1337, 2023-09-22:
Hi, 

great project! i build a sACN Node and it works flawlessly!
My only question:

does it work to get dmx input on Port A and Merge it with the sACN Data and transmit it on port B?

I tried to set both ports to uni 1, one as input one as output, but this seems to does not work.

Can you help me?

vanvught, 2023-09-23:
Hi @pilo1337 ,

Thanks for the compliments. Much appreciated!

> does it work to get dmx input on Port A and Merge it with the sACN Data and transmit it on port B?

Not, yet. I am aware that other commercial products do provide this functionality. It is my policy not just copying commercial products. But instead working with the requirements from my users. 

> Can you help me?

Yes. I am working on a major upgrade of the firmware. This enhanced would be a great addition to it.

This would be the generic implementation for both sACN E1.31 and Art-Net 4 node:

```
Port A has DMX512 input assigned with Universe 1 and its protocol sACN E1.31
Then when port B is output and has also Universe 1 with protocol sACN E1.31 -> DMX512 and sACN E1.31 are merged into sACN E1.31 output.
The merging algorithm is set on Port B.

For the 4 port firmware (the Orange Pi One), then this merging applies for port A/B and C/D.
```

Correct?

Thanks, Arjan

pilo1337, 2023-09-24:
yeah, this would be awesome!

vanvught, 2023-10-01:
Hi @pilo1337 , here you are. There is also an updated Remote Configuration UI attached. There is debug output. So when there are any issues, then please attach the uart0 debug output. Thanks. Arjan 

[image]

[opi_emac_e131_dmx_multi.zip](https://github.com/vanvught/rpidmx512/files/12777557/opi_emac_e131_dmx_multi.zip)
[opi_emac_artnet_dmx_multi.zip](https://github.com/vanvught/rpidmx512/files/12777559/opi_emac_artnet_dmx_multi.zip)
[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/12777560/RemoteConfigUI.jar.zip)

pilo1337, 2023-10-02:
i will give it a shot, thanks!

pilo1337, 2023-10-04:
unfortunately, i got no success:
U-Boot SPL 2018.09 (Mar 02 2019 - 08:59:42 +0100)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2018.09 (Mar 02 2019 - 08:59:42 +0100) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Zero
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
USB2:   USB EHCI 1.00
USB3:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
scanning bus 2 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
157140 bytes read in 9 ms (16.7 MiB/s)
Wrong Image Format for bootm command
ERROR: can't get kernel image!
=>

i extracted the zip files and the gz files and put them onto sd card, but no boot with any file.. what i am doing wrong?

vanvught, 2023-10-04:
> i extracted the zip files and the gz files and put them onto sd card, but no boot with any file.. what i am doing wrong?

Basically we use the SDCard just once. Afterwards we can just TFTP the new image to the Orange Pi. 

Or use the automate Workflow option in the Java UI , or the manually workflow  -> https://www.orangepi-dmx.org/orange-pi/remote-configuration-management-for-spi-flash/example-java-ui/workflow-firmware-installation

closed 2023-10-07 by vanvught