vanvught/rpidmx512 #272: Implement an Art-Net node with PWM led driver PCA9685

vanvught, 2023-10-30:
- [ ] lib-lib-pca9685 -> major cleanup
- [ ] lib-pca9685dmx -> major cleanup
- [ ] lib-configstore -> add class StorePCA9685
- [ ] lib-remoteconfig -> add support for configuration file pca9685.txt
- [ ] Create project `opi_emac_artnet_rdm_pca9685`
- [ ] Update Java Remote Configuration UI ([https://github.com/vanvught/Remote-Config-UI](https://github.com/vanvught/Remote-Config-UI))

@ icecaster

vanvught, 2023-10-31:
Hi Tim,

Here you are. 

See the SDCard section in https://github.com/vanvught/h3dmx512-zip for how to create the SDCard for the initial installation. 

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/13217720/orangepi_zero.uImage.gz)

[RemoteConfigUI.jar.zip](https://github.com/vanvught/rpidmx512/files/13217780/RemoteConfigUI.jar.zip)

The remote config UI is not finished; the Wizard is not implemented, yet.

For testing, I have connected 5 RGB led's. 

[image]

https://github.com/vanvught/rpidmx512/assets/4319913/7311cf80-1ae4-4355-ab8a-d15b203a9ad0

@icecaster

icecaster, 2023-10-31:
Looks great!

The PCA9685 handles 12 bit, should probably use 2x artnet channels for fine dimming control?

vanvught, 2023-11-01:
> The PCA9685 handles 12 bit, should probably use 2x artnet channels for fine dimming control?

I've just polished the 2 years old code ;-) which is just using a single channel. I will add the 12-bit support to my ToDo list.

Groet, Arjan

vanvught, 2023-11-03:
Hi Tim,

I have added a new property `#use_8bit=0`

So per default it is 2 dmx channels per pwm channel:
```
Network init
Waiting for PHY auto negotiation to complete
Network started
Detected MX25L1605D with sector size 4096 total 2097152 bytes
[V1.0] Orange Pi Zero Compiled on Nov  3 2023 at 19:15:33
Art-Net 4 PCA9685
Network [D]
 Hostname  : allwinner_240813
 IfName    : 1: eth0 02:42:14:24:08:13
 Primary   : 192.168.2.147/24 (HTTP only 2.36.8.19)
 Gateway   : 192.168.2.1
 Broadcast : 192.168.2.255
Configuring MDNS
mDNS
 allwinner_240813.local.
 _config._udp.local. 10501 
 _tftp._udp.local. 69 
 _http._tcp.local. 80 node=Art-Net 4 PCA9685
NTP v4 Client
 Not enabled
Configuring Art-Net
RDM Device configuration
 Manufacturer Name : http://www.orangepi-dmx.org
 Manufacturer ID   : 5000
 Serial Number     : C0A80293
 Root label        : Orange Pi Zero RDM Device
 Product Category  : 0100
 Product Detail    : 0004
RDM Responder configuration
 Protocol Version 1.0
 DMX Address      : 1
 DMX Footprint    : 30
 Personality 1 of 1 [PCA9685]
 Sub Devices      : 0
 Sensors          : 1
Art-Net 4 V1.58
 Long name  : PCA9685
 Output
  Port 0  1    HTP Art-Net RDM
Bridge
 Firmware : 1.27
PCA9685 LED 16-bit
 Board instances: 1
 Channel count: 15
 DMX footprint: 30
 DMX start address: 1
Starting Art-Net
Art-Net started
```

[image]

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/13253415/orangepi_zero.uImage.gz)

@icecaster

vanvught, 2023-11-17:
Hi Tim,

I've fixed the web-interface. 
[orangepi_zero.img.gz](https://github.com/vanvught/rpidmx512/files/13396832/orangepi_zero.img.gz)
Please use the UI in https://github.com/vanvught/rpidmx512/issues/272#issuecomment-1787373055 for upgrading the firmware.

Groet, Arjan
@icecaster

vanvught, 2024-01-06:
Hi Tim, I have updated the Remote Configuration manager and created a new release -> https://github.com/vanvught/Remote-Config-UI/releases/tag/V25

Groet, Arjan

[image]

vanvught, 2024-01-13:
Published https://github.com/vanvught/h3dmx512-zip

closed 2024-01-13 by vanvught

icecaster, 2024-01-19:
Amazing, thank you for your work!