vanvught/rpidmx512 #273: Orange Pi Zero requires MagicQ Reduced Rate Output when full 8x4 functionality is being used

mattg66, 2023-11-04:
Hi,
With a single output running x4, operation is as expected with smooth pixel output, however when all 8 outputs are being utilised x4, the output is stuttery and updates a few times a second. When reduced rate output mode is enabled in MagicQ, the issue is reduced however it is still choppy and not as fluid as would be expected. I have faced this problem for a few years now and it still seems to be present.
I have verified it is not magicq at fault as it is always sending 32 universes of artnet traffic to the orange pi, just changing the output active out is enough to introduce the issue.

Please let me know if you want any other information from me.

vanvught, 2023-11-04:
Hi Matt,

Thank for reporting this issue. Please give the attached firmware a try.

Groet, Arjan

```
Note:
*. Before upgrading the firmware it is advised to make a backup of the configuration:
  1. https://www.orangepi-dmx.org/orange-pi/remote-configuration-management-for-spi-flash
Or
  2. Use the latest sample UI -> https://github.com/vanvught/Remote-Config-UI/releases

*. After the upgrade, then restore the saved configuration.
```

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/13257029/orangepi_zero.uImage.gz)

mattg66, 2023-11-05:
Hi, Unfortunately the issue still remains, however it does seem better than before but the output is still stuttering. Interestingly whilst the orange pi was receiving artnet, i was unable to upload the firmware. not sure if this is relevant or not.

Cheers,
Matt

vanvught, 2023-11-05:
Hi Matt,

That is somehow odd. I have tested with 8x 600 pixels with 44 FPS and it is going smoothly. I am using MapMapper.

Cheers, Arjan

vanvught, 2023-11-05:
I will think about how I can replicate the issue you have.

mattg66, 2023-11-08:
Would you be able to test on your end using MagicQ? I have it set to artnet unicast v4 and continuous mode and I am using the test function on each universe to test the output

mattg66, 2023-11-12:
I have been playing around and it works for periods perfectly smoothly, then it randomly starts dropping 'frames' and goes down to a few FPS, then after about 10 or so seconds it works fine again until it starts lagging again.

mattg66, 2023-11-12:
I have tried sACN as well with the same issue, it seems that there is a fundamental incompatibility with how the library is handling network lighting protocols and the way MagicQ sends the data, although no other fixture or artnet node i have used has exhibited any issues when used in conjunction with MagicQ

vanvught, 2023-11-12:
Hi Matt, thanks for the additional information. The firmware is successfully tested with MadMapper, Lightjams and MADRIX. And desks like GrandMA. Also, the firmware is fully compatible with the Art-Net 4 standard and fully compatible with the sACN E1.31 standard. I will work on a tool in order to diagnose why MagicQ is different with respect to the others.

mattg66, 2023-11-15:
I have tested with xLights and it does seem to work fine on that, so yes it must be some weird incompatibility between opidmx and MagicQ, the reasons behind I am now very curious as it is my favourite lighting control solution.

vanvught, 2023-11-24:
Hi Matt, thanks for the update. Good to know that xLights is working. Fixing the issue with MagicQ is on my radar and todo-list. I hope to find some good free time soon. Thanks. Arjan.

vanvught, 2024-01-12:
Hi Matt,

When your time permits, then please give that attached firmware a try. I have made some changes which improves the DMA transfers. I hope this is helps with MagicQ.

Thanks, Arjan

[orangepi_zero.uImage.gz](https://github.com/vanvught/rpidmx512/files/13917289/orangepi_zero.uImage.gz)

vanvught, 2024-08-30:
Hi Matt, please check https://github.com/vanvught/h3dmx512-zip for the latest firmware. Re-open this issue when needed. Thanks, Arjan

closed 2024-08-30 by vanvught