vanvught/rpidmx512 #281: FPP and Xlights performance

msmorul, 2024-08-13:
Using the 8x4 CLPD board with FPP test sequences and Xlights test panel, there is no output (build-in sequences work fine). The underlying issue appears to be the transmission rate from both of these is high enough that it collides w/ the DMA transfer from the PI.

Adding in barriers around spi_dma_tx_start to [Update](https://github.com/vanvught/rpidmx512/blob/master/lib-ws28xx/src/h3/ws28xxmulti.cpp#L374) as was done in [Blackout](https://github.com/vanvught/rpidmx512/blob/master/lib-ws28xx/src/h3/ws28xxmulti.cpp#L382) solved the problem.

vanvught, 2024-08-13:
Hi @msmorul Mike, thanks for reporting this issue. I will fix it as soon as possible on the development branch. Thanks again! Arjan

vanvught, 2024-08-30:
Firmware with fix is available -> https://github.com/vanvught/h3dmx512-zip

closed 2024-08-30 by vanvught