vanvught/rpidmx512 #286: UTC Offset not working

dirkdamerau, 2024-12-01:
I'm using an Orange Pi Zero H3 (allwinner) with the **opi_emac_ltc_smpte.zip** uImage. Configuration is set to "systime" for a NTP Client to LTC Output Bridge. ntp_server is set and working. The only issue is that none of the UTC Offset settings does anything. I always get only UTC time on the LTC Output. Tried the Setting in the network.txt as well as gps.txt as well as env.txt. I did an initial config using an SD Card, after that the SD Card was removed. Config changes are done by Remote Config Manager v27. Changing other settings works as expected.

vanvught, 2024-12-02:
The implementation is using 
 https://github.com/vanvught/rpidmx512/blob/7a83cc4cea07117dc9aa846e3b32ce395589f44e/lib-ltc/src/arm/systimereader.cpp#L249

> time() returns the time as the number of seconds since the Epoch, 1970-01-01 00:00:00 +0000 (UTC).

I have added a new property `utc_offset` in `ltc.txt`:
[image]

Please give the attached updated firmware a good test. Thanks, Arjan
[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/17980945/orangepi_zero.uImage.gz)

dirkdamerau, 2024-12-02:
Hey Arjan, thanks for your fast fix. This patch works ! I tried positive as well as negative values, all working. Thank again ! Best Dirk

vanvught, 2024-12-04:
Hi Dirk, thank you! Groeten, Arjan

closed 2024-12-04 by vanvught