vanvught/rpidmx512 #289: Setting timecode thorough OSC not working

kyleawayan, 2024-12-06:
Hi,

Thank you for the amazing project. I plan to use it with Beat Link Trigger.

However, I cannot set the timecode through OSC. For context, my hostname of the device is `iz_one` and I am using the internal timecode as a source. I have an Orange Pi Zero LTS AllWinner H3 512MB.

When I send `/iz_one/tc/start/01/00/00/00` to it, the timecode starts and counts up from 00:00:00:00, not 01:00:00:00. I confirmed this using an LTC audio input to Resolume, and it also shows on the OLED screen I connected to the Orange Pi. Trying out different times doesn't work either. The same applies for `/goto/hh/mm/ss/ff`.

However, setting the start via your Remote Configuration app works fine.

vanvught, 2024-12-06:
Hi Kyle,

Thank you for reporting this bug. 

Please find attached the updated firmware with the fix.

Thanks, Arjan

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18035990/orangepi_zero.uImage.gz)

kyleawayan, 2024-12-06:
Hi Arjan,

Thank you for the quick fix! It works and I confirmed it also works with the Beat Link Trigger SMPTE example!

closed 2024-12-07 by vanvught