vanvught/rpidmx512 #290: How to set up MAX7219 7 segment display w/ LTC?

kyleawayan, 2024-12-07:
I ordered this display from Amazon: https://www.amazon.com/dp/B01D0WSCJA?ref=ppx_yo2ov_dt_b_fed_asin_title

I connected it to the Orange Pi, following the LTC GPIO headers on your page: https://www.orangepi-dmx.org/orange-pi/gpio-header. Here's how I connected it from the Orange Pi -> the MAX7219 display:

- 5V -> VCC
- GND -> GND
- SPI1 MOSI (Pin 19) -> DIN
- SPI1 CS0 (Pin 24) -> CS
- SPI1 SCLK (Pin 23) -> CLK

However, upon booting up, the display doesn't show anything. I made sure to set `max7219_type` to `7segment`.

I see that your page on the MAX7219 is WIP, https://www.orangepi-dmx.org/9-ltc-smpte-timecode-reader-converter-generator/displays/max7219, so I wanted to know if there is any other configuration I have to do for the display to show the current timecode? I currently use it in internal mode, as described in #289. If not, then I may have to order new displays or see if I can test them out with another device.

vanvught, 2024-12-07:
Hi Kyle,

Your setup is oké. I have verified with the build attached in #289 and it is working fine.

Groet, Arjan

https://github.com/user-attachments/assets/08921556-0218-4508-95fe-ad04fb517d49

```
#ltc.txt
source=internal
# Disable outputs #
#disable_display=0
#disable_max7219=0
```

```
#ldisplay.txt
# OLED SSD1306/11 #
#oled_intensity=127
# Rotary control #
#rotary_fullstep=0
# MAX7219 #
max7219_type=7segment
#max7219_intensity=4
```

kyleawayan, 2024-12-07:
Thank you for confirming and testing! I got it work, turns out I need to improve on my soldering skills 😆

closed 2024-12-07 by kyleawayan