vanvught/rpidmx512 #291: rtpMIDI and Apple Network MIDI only receive SysEx instead of quarter frames

kyleawayan, 2024-12-12:
Hi,

I noticed that when using rtpMIDI or Apple Network MIDI (both tested on Mac and Windows), the only messages sent from the Orange Pi are system exclusive MIDI messages. I think these aren't enough to fill the MTC standard I guess, as the MTC is not being received correctly in grandMA3. I used the Protokol software to monitor the MIDI messages on both Mac and Windows.

When using different MIDI timecode generator software, I see that they are sending quarter frames, and it also gets received correctly in grandMA3.

I would've liked to use hardwired MIDI--and I saw your board schematics on your website--however I have no experience in building boards so I was hoping network MIDI can be an alternative.

Also just another note, on your [successfully tested page](https://www.orangepi-dmx.org/9-ltc-smpte-timecode-reader-converter-generator/successfully-tested) for the LTC project, I saw that the Roland UM-ONE MIDI interface mk2 was listed. I tried plugging mine in to the USB of the Orange Pi, but it didn't work. I realized that you may have meant that it works with your MIDI board, to plug it in to a computer. I thought it was directly supported with the Orange Pi so I can output MIDI timecode through that USB adapter straight away. 

I may recommend adding a note that USB -> MIDI is not directly supported and requires a custom board/connection to the UART2 pins. It would be awesome though if this a USB to MIDI interface was directly supported!

vanvught, 2024-12-12:
Hi Kyle,

For USB (MIDI) on the Orange Pi Zero, please see https://github.com/vanvught/rpidmx512/issues/125

I have successfully used the Roland UM-ONE MIDI connected to macOS. What is special for this interface are the Active Sense messages, which are supported in my firmware. 

Thanks, Arjan

vanvught, 2024-12-12:
Hi Kyle,

> I noticed that when using rtpMIDI or Apple Network MIDI (both tested on Mac and Windows), the only messages sent from the Orange Pi are system exclusive MIDI messages. 

That is working fine with all the applications I have on macOS. I have never had the opportunity to test it with grandMA3 (or any grandMA :-). I will work on this as an enhancement. 

Thanks, Arjan

[image]

vanvught, 2024-12-12:
Hi Kyle,

Please find attached the updated firmware with the enhancement implemented. I have done some primarily tests. Let me know when there any issues.

@dirkdamerau When your time permits, then please give this firmware a good test as well.

Thanks, Arjan

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18114127/orangepi_zero.uImage.gz)

kyleawayan, 2024-12-13:
Thank you for the quick update again! It works perfectly in grandMA3 onPC via rtpMIDI on Windows! Thank you so much!

commit 37bac76c (2024-12-14): Fixed #291
closed 2024-12-14 by vanvught