vanvught/rpidmx512 #292: LTC SMPTE: Improve NTP Client synchronise process.

vanvught, 2024-12-15:
Follow up for #287  @dirkdamerau

Place holder; more details will be added.

vanvught, 2024-12-22:
Hi @dirkdamerau Dirk,

I have added a new state : "LOCKED". The NTP client will run with POLL_SECONDS_MIN (8 seconds) until the NTP offset with the server is less than 1 milliseconds. This is it the "LOCKED" state. Now it is running with POLL_SECONDS_MAX (approx 1 hour and 8 minutes). With the new system time implemented in #287, this should be a pretty accurate NTP based clock. 

Groeten, Arjan 

 
[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18222010/orangepi_zero.uImage.gz)

dirkdamerau, 2024-12-28:
Dear Ajan, I checked the image above as well as the one you uploaded today @ #294. I do not see if and when the locked state is reached. I can confirm that the sync to the NTP Server works but I still see missing and missorderd frames in the LTC Output. It happens randomly between 17s and 3:01m. So to say once and a while every few minutes. Most of the time it follows this pattern:

00000000   23:48:06:20 | 31310952 31312871  
00000000   23:48:06:21 | 31312872 31314791  
00000000   23:48:06:22 | 31314792 31316711  
00000000   23:48:06:23 | 31316712 31318631  
00000000   23:48:06:24 | 31318632 31320551  
#DISCONTINUITY
00000000   23:48:07:01 | 31320552 31322471  
00000000   23:48:07:02 | 31322472 31324390  
00000000   23:48:07:03 | 31324391 31326307  
00000000   23:48:07:04 | 31326308 31328230  
00000000   23:48:07:05 | 31328231 31330150  
00000000   23:48:07:06 | 31330151 31332070  
00000000   23:48:07:07 | 31332071 31333990  
00000000   23:48:07:08 | 31333991 31335910  
00000000   23:48:07:09 | 31335911 31337830  
00000000   23:48:07:10 | 31337831 31339750  
00000000   23:48:07:11 | 31339751 31341670  
00000000   23:48:07:12 | 31341671 31343590  
00000000   23:48:07:13 | 31343591 31345510  
00000000   23:48:07:14 | 31345511 31347430  
00000000   23:48:07:15 | 31347431 31349350  
00000000   23:48:07:16 | 31349351 31351270  
00000000   23:48:07:17 | 31351271 31353190  
00000000   23:48:07:18 | 31353191 31355109  
00000000   23:48:07:19 | 31355110 31357029  
00000000   23:48:07:20 | 31357030 31358949  
00000000   23:48:07:21 | 31358950 31360869  
00000000   23:48:07:22 | 31360870 31362789  
00000000   23:48:07:23 | 31362790 31364709  
00000000   23:48:07:24 | 31364710 31366629  
#DISCONTINUITY
00000000   23:48:07:23 | 31366630 31368549  
#DISCONTINUITY
00000000   23:48:08:00 | 31368550 31370469  
00000000   23:48:08:01 | 31370470 31372389  
00000000   23:48:08:02 | 31372390 31374309  
00000000   23:48:08:03 | 31374310 31376229

At first there is one frame missing and a second or two later there is a doubled missordered. So the overall accuracy is ok from my standpoint but the LTC output is not consistent. If running "internal" it is still pretty accurate with no errors like in the last test. Something keep the LTC Output unstable when NTP Client is running.

It is ok for me looking at it but it could fail if I use it to trigger something based on it as time of day.

vanvught, 2024-12-29:
Dear Dirk, thank you so much for the feedback. 

The discontinuity is always around the last frame. I have some thoughts why it could be; the LTC is continues running while the source local time get changed by the NTP Client. This happens in the beginning and should not happen anymore in the locked state. 

I will work first on #296; that should be a quick win. 

I am eager to get this discontinuity issue solved. 

Thanks, Arjan

@hippyau @rbarreiros

closed 2025-02-07 by vanvught