vanvught/rpidmx512 #294: Various Issues with LTC IN Timecode

joakyhsf69, 2024-12-27:
Hello, I haven't used the device for some time, but upon picking it up and updating it to the latest version (4.9), I have encountered several issues that have prevented me from using it.

The main problem is that with LTC IN, the MTC output, both RTP-MIDI and DIN, does not work, and the Artnet output doesn’t work either. I am unsure if anything else has stopped functioning, as I have no way to verify. With the internal generator, both DIN MIDI and Artnet do work, so it doesn’t seem to be a hardware issue.

The case with the RTP output is even stranger, as it doesn’t work with either LTC IN or the internal generator. The MA2, which I am using to receive the TC, shows the numbers but doesn’t turn green, as if the message isn’t arriving correctly. This happens with both LTC IN and the generator

vanvught, 2024-12-27:
See the latest build here https://github.com/vanvught/rpidmx512/issues/293#issuecomment-2559921736

The MA2 issue is solved withhttps://github.com/vanvught/rpidmx512/issues/291

joakyhsf69, 2024-12-27:
Yes, with the latest version you provided, the issue with RTP-MIDI is resolved, but the LTC input still does not work. I can see the numbers on the display, but nothing is output, neither via MIDI DIN nor RTP-MIDI. However, LTC IN does now output through Artnet.

With this new version, I see the message 'No NTP Client' on the display. Previously, I could see the IP address, which was much more useful.

joakyhsf69, 2024-12-27:
After several tests, it only sends one packet through MIDI when the LTC starts; then it stops and doesn’t continue sending MIDI. Through Artnet, it does seem to work fine

vanvught, 2024-12-27:
> With this new version, I see the message 'No NTP Client' on the display. Previously, I could see the IP address, which was much more useful.

The IP address is still at row 3. The NTP status message is at row 4.

[image]

vanvught, 2024-12-27:
> After several tests, it only sends one packet through MIDI when the LTC starts; then it stops and doesn’t continue sending MIDI. 

Fixed in attached firmware.
[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18261177/orangepi_zero.uImage.gz)

@dirkdamerau @kyleawayan

joakyhsf69, 2024-12-27:
I have a display with only two rows.

[image0.jpeg]

IMG_0612.jpeg (view on web)<https://github.com/user-attachments/assets/00777ec3-53af-4bc7-a171-66b88b6a6253>

joakyhsf69, 2024-12-27:
With this firmware, the MIDI output seems to work, both DIN and RTP-MIDI, with LTC IN. However, approximately once a minute, the MA3 experiences a clock jump, which is reflected in the status monitor.

It’s worse with the internal generator, where this happens every 2 seconds, both via RTP-MIDI and MIDI DIN—the input clock is inconsistent. Through Artnet, it works fine, both with the internal generator and LTC IN

vanvught, 2024-12-27:
MA2 works fine here -> https://github.com/vanvught/rpidmx512/issues/291#issuecomment-2542424982

closed 2024-12-27 by vanvught

joakyhsf69, 2024-12-27:
This is what happens with the internal generator on any of the MIDI outputs with MA3: 

https://github.com/user-attachments/assets/4394ccbb-ca42-438c-91d7-4b7dc898e6d3

vanvught, 2024-12-27:
> This is what happens with the internal generator on any of the MIDI outputs with MA3:

Please explain in detail, for a none MA user, what we are seeing? Thanks. 

With reference to #291 Hi Kyle, when your time permits, then please add your comments. Thanks.  
@kyleawayan 
@hippyau

joakyhsf69, 2024-12-27:
What I’m doing is generating a timecode and sending it first via rtpMidi and then through MIDI DIN with a USB adapter to the MA3, which receives the timecode through slot 1. The MA3 detects a jump in the timecode every few seconds and displays it on the system monitor below. You can also see the timecode received via Art-Net there, which has no jumps and works smoothly. The same happens with MA2—it occasionally skips—but these events are not shown on the system monitor and are less noticeable. With the generator, it happens much more often than when converting from LTC to MIDI, but it still happens, about once every minute.

vanvught, 2024-12-27:
I understand what you are doing. But we need the details; exactly the 'few seconds'. Then what is jumping; the values. Thanks.

I've spent a lot of time analysing the Wireshark capture. There are 4 RTP packets between each Art-Net package. Reading this capture I could not find any jump. Please tell me where there is a jump.  Thanks. 
[Uploading rtp-midi.pcapng.zip…]()

@kyleawayan @hippyau @dirkdamerau

joakyhsf69, 2024-12-27:
This is what appears on the monitor. I suppose the last values are frames, but I’m not sure. Every 2 seconds, a new message appears showing a jump forward and backward in those “frames.”

18h46m46.501s  RT : Timecode signal on Slot 1 jumped: 1980  -> 1925
18h46m48.435s  RT : Timecode signal on Slot 1 jumped: 2035  -> 2100
18h46m50.500s  RT : Timecode signal on Slot 1 jumped: 2100  -> 2045
18h46m52.434s  RT : Timecode signal on Slot 1 jumped: 2155  -> 2220
18h46m54.501s  RT : Timecode signal on Slot 1 jumped: 2220  -> 2165
18h46m56.435s  RT : Timecode signal on Slot 1 jumped: 2270  -> 2340
18h46m58.500s  RT : Timecode signal on Slot 1 jumped: 2340  -> 2285
18h47m00.434s  RT : Timecode signal on Slot 1 jumped: 2395  -> 2460
18h47m02.500s  RT : Timecode signal on Slot 1 jumped: 2460  -> 2405
18h47m04.434s  RT : Timecode signal on Slot 1 jumped: 2515  -> 2580
18h47m06.500s  RT : Timecode signal on Slot 1 jumped: 2580  -> 2525
18h47m08.434s  RT : Timecode signal on Slot 1 jumped: 2635  -> 2700
18h47m10.500s  RT : Timecode signal on Slot 1 jumped: 2700  -> 2645
18h47m12.434s  RT : Timecode signal on Slot 1 jumped: 2755  -> 2820
18h47m14.500s  RT : Timecode signal on Slot 1 jumped: 2820  -> 2765
18h47m16.434s  RT : Timecode signal on Slot 1 jumped: 2875  -> 2940
18h47m18.500s  RT : Timecode signal on Slot 1 jumped: 2940  -> 2885
18h47m20.434s  RT : Timecode signal on Slot 1 jumped: 2995  -> 3060
18h47m22.500s  RT : Timecode signal on Slot 1 jumped: 3060  -> 3005
18h47m24.434s  RT : Timecode signal on Slot 1 jumped: 3115  -> 3180
18h47m26.500s  RT : Timecode signal on Slot 1 jumped: 3180  -> 3125
18h47m28.434s  RT : Timecode signal on Slot 1 jumped: 3235  -> 3300
18h47m30.500s  RT : Timecode signal on Slot 1 jumped: 3300  -> 3245
18h47m32.434s  RT : Timecode signal on Slot 1 jumped: 3355  -> 3420

¿Te gustaría agregar algo más al mensaje?

vanvught, 2024-12-27:
> 18h46m46.501s RT : Timecode signal on Slot 1 jumped: 1980 -> 1925
18h46m48.435s RT : Timecode signal on Slot 1 jumped: 2035 -> 2100
18h46m50.500s RT : Timecode signal on Slot 1 jumped: 2100 -> 2045
18h46m52.434s RT : Timecode signal on Slot 1 jumped: 2155 -> 2220
18h46m54.501s RT : Timecode signal on Slot 1 jumped: 2220 -> 2165
18h46m56.435s RT : Timecode signal on Slot 1 jumped: 2270 -> 2340
18h46m58.500s RT : Timecode signal on Slot 1 jumped: 2340 -> 2285
18h47m00.434s RT : Timecode signal on Slot 1 jumped: 2395 -> 2460
18h47m02.500s RT : Timecode signal on Slot 1 jumped: 2460 -> 2405
18h47m04.434s RT : Timecode signal on Slot 1 jumped: 2515 -> 2580
18h47m06.500s RT : Timecode signal on Slot 1 jumped: 2580 -> 2525
18h47m08.434s RT : Timecode signal on Slot 1 jumped: 2635 -> 2700
18h47m10.500s RT : Timecode signal on Slot 1 jumped: 2700 -> 2645
18h47m12.434s RT : Timecode signal on Slot 1 jumped: 2755 -> 2820
18h47m14.500s RT : Timecode signal on Slot 1 jumped: 2820 -> 2765
18h47m16.434s RT : Timecode signal on Slot 1 jumped: 2875 -> 2940
18h47m18.500s RT : Timecode signal on Slot 1 jumped: 2940 -> 2885
18h47m20.434s RT : Timecode signal on Slot 1 jumped: 2995 -> 3060
18h47m22.500s RT : Timecode signal on Slot 1 jumped: 3060 -> 3005
18h47m24.434s RT : Timecode signal on Slot 1 jumped: 3115 -> 3180
18h47m26.500s RT : Timecode signal on Slot 1 jumped: 3180 -> 3125
18h47m28.434s RT : Timecode signal on Slot 1 jumped: 3235 -> 3300
18h47m30.500s RT : Timecode signal on Slot 1 jumped: 3300 -> 3245
18h47m32.434s RT : Timecode signal on Slot 1 jumped: 3355 -> 3420

Yes, I saw that on the video as well. But it does not tell much. At least it does not tell me what is really happening.
We really need more details. Please check also a Wireshark capture. Again MA works fine here -> https://github.com/vanvught/rpidmx512/issues/291#issuecomment-2542424982

@kyleawayan @hippyau @dirkdamerau

kyleawayan, 2024-12-27:
The rtpMIDI works good enough for me, I've been able to sync my light shows in grandMA3 along with Resolume and my Pioneer DJ controller with Beat Link Trigger. However I have not opened the system monitor while running this, so I may be getting these jumps too, but not noticing them.

Does MA3 show the same logs when using another MIDI timecode software? I remember I had some similar issue a while ago with Depence's MIDI timecode out--though I don't think I was getting jumps every 2 seconds.

Also FYI @vanvught, the MA3 onPC is free to download on macOS and Windows, and MIDI input works without hardware. This can probably help with debugging if needed.

vanvught, 2024-12-27:
> the MA3 onPC is free to download on macOS and Windows, and MIDI input works without hardware. This can probably help with debugging if needed.

@kyleawayan Hi Kyle, thanks. I will give it a try. 

When your time permits, please give the latest release V5.0 -> https://github.com/vanvught/h3dmx512-zip/blob/master/opi_emac_ltc_smpte.zip a try. Just for a sanity check to make sure I haven't broken anything.

It's past midnight, I am going to sign-off.

joakyhsf69, 2024-12-27:
> Does MA3 show the same logs when using another MIDI timecode software? I remember I had some similar issue a while ago with Depence's MIDI timecode out--though I don't think I was getting jumps every 2 seconds.

The jumps occur every 2 seconds when I use the internal generator, both with Midi Din and rtpMidi. If I convert from LTC to MTC, the jumps are significantly reduced to about 1 every 60 seconds. So even with the same hardware, the jumps vary depending on the source of the original timecode.

vanvught, 2024-12-27:
> The jumps occur every 2 seconds when I use the internal generator, both with Midi Din and rtpMidi. If I convert from LTC to MTC, the jumps are significantly reduced to about 1 every 60 seconds. So even with the same hardware, the jumps vary depending on the source of the original timecode.

I really would like to see the timecode when it jumps. And it would be great to see the same in a Wireshark capture.

vanvught, 2024-12-27:
@joakyhsf69 For reference, please also try with `source=systime` Thanks.

> It's past midnight, I am going to sign-off. 

Now for real.

joakyhsf69, 2024-12-28:
> I really would like to see the timecode when it jumps. And it would be great to see the same in a Wireshark capture.

Sorry, I’ll be away for a few days. When I can, I’ll test the source=systime and try capturing with Wireshark. Thanks

rbarreiros, 2024-12-28:
> The jumps occur every 2 seconds when I use the internal generator, both with Midi Din and rtpMidi. If I convert from LTC to MTC, the jumps are significantly reduced to about 1 every 60 seconds. So [...]

I just jumped in mid conversation, but, are the FPS the same on both sides ? I.E. GMA3 TC @ something like 30fps and the receiver at some other differente FPS like 24, it'll drop/ignore 4 frames? It's the only thing I'm imagining that'll make frames skip, different FPS settings.

joakyhsf69, 2024-12-29:
> @joakyhsf69 For reference, please also try with `source=systime` Thanks.

Hello, I just got back. I'm going to try capturing the RTP-MIDI traffic with Wireshark, but I already tested using source=systime. It works the same as with source=ltc, with a jump every minute, not every two seconds. I also tested with source=artnet, and it works best there—it doesn't jump and is very stable, both with RTP-MIDI out and MIDI DIN out. In summary:

source=systime      drop every 60s
source=Ltc             drop every 60s
source=artnet        no drop, works fine
source=internal      drop every 2s

Always with the framerate set to 25fps, the MA3 automatically syncs with the framerate, no need to do anything.

joakyhsf69, 2024-12-29:
I'm not sure if I did it right, but here are the Wireshark captures of the rtpMIDI interface with source=internal and source=systime.
[Wireshark Captures.zip](https://github.com/user-attachments/files/18269304/Wireshark.Captures.zip)

vanvught, 2024-12-29:
Hi, thanks for the feedback.

kyleawayan, 2025-01-05:
Can confirm with the production firmware, I'm getting timecode jumps also. This is with `source=internal` and outputting timecode via rtpMIDI to MA3 onPC.

And I think it's part of the [MIDI timecode specification to tell MA3 the FPS of the timecode](https://en.wikipedia.org/wiki/MIDI_timecode), so changing a setting in MA3 to expect a certain FPS may not be the problem (and I also can't find an option to change it).