vanvught/rpidmx512 #296: LTC SMPTE: NTP Client: Status LOCKED is not shown

vanvught, 2024-12-28:
Follow-up for #292 
>  I do not see if and when the locked state is reached. 

Also #294 
> With this new version, I see the message 'No NTP Client' on the display. Previously, I could see the IP address, which was much more useful.

vanvught, 2024-12-30:
All, are we close to the perfect firmware?

- The NTP Client LOCKED status is now shown
- There is only the 'FTP Client' message when the `source=systime`
- I have redesigned the MIDI quarter-frame messages -> there are now completely in sync with frame updates #294 
- I have made changes to `source=systime` #292 

Groeten, Arjan

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18275175/orangepi_zero.uImage.gz)

@rbarreiros @hippyau @dirkdamerau @kyleawayan @joakyhsf69

[image]
[image]

joakyhsf69, 2024-12-30:
I'm sorry to say this, but with this new firmware, both source=systime and source=internal are still jittery; in fact, it's now more noticeable, and the jumps can clearly be seen on the MA3 clock. What's worse is that source=ltc doesn't output via MIDI, RTP, or DIN at all—only through Artnet.

vanvught, 2024-12-30:
@joakyhsf69 

> both source=systime and source=internal are still jittery; in fact, it's now more noticeable, and the jumps can clearly be seen on the MA3 clock. 

Please check a new Wireshark capture; there is perfect RTP MIDI output. 

> What's worse is that source=ltc doesn't output via MIDI, RTP, or DIN at all—only through Artnet.

It's working fine for me. There is output, perfect output.

@rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2024-12-30:
See Wireshark below. Perfect RTP in sync with Art-Net

[image]

@rbarreiros @hippyau @dirkdamerau @kyleawayan @joakyhsf69

vanvught, 2024-12-30:
grandMA3 onPC is looking good.

[image]

vanvught, 2024-12-30:
> when the source is set to LTC, the MA3 no longer throws an error, not even every 60 seconds, so that issue has been resolved.

Cool! @rbarreiros @hippyau @dirkdamerau @kyleawayan

joakyhsf69, 2024-12-30:
> grandMA3 onPC is looking good -> input is LTC
> 
> <img alt="Scherm­afbeelding 2024-12-30 om 20 35 01" width="1175" src="https://private-user-images.githubusercontent.com/4319913/399359744-fb41371f-4249-47c0-a292-1e9526d28a08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1ODgzNjcsIm5iZiI6MTczNTU4ODA2NywicGF0aCI6Ii80MzE5OTEzLzM5OTM1OTc0NC1mYjQxMzcxZi00MjQ5LTQ3YzAtYTI5Mi0xZTk1MjZkMjhhMDgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMzBUMTk0NzQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGExOTYyZjY0YmRmMGQ4NGRkZTZmMmU2NGY4MzZmZmY5YTkyYzU5NGU5NzIyMzA3YjQzNjk4OTgwNGU5YzQ0ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.3TMhSQpTxbJGL14PglSbCms3zfsXh7iqUwaX4o7mDE8">

Indeed, now with source=LTC it works great.

vanvught, 2024-12-30:
> > However, now with the source set to rtpMIDI or Artnet, it only sends zero through MIDI.
> 
> Will investigate now @joakyhsf69

Fixed.

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18277466/orangepi_zero.uImage.gz)

Next 
> Now, with this new firmware, the source=internal and source=systime both have jumps in MA3, although not in MIDI-OX. Previously, there were no jumps with source=systime.

That's a challenge; working with MIDI-OX and not working with MA3. It's evening in NL. I will take a coffee break and give it some thoughts for what could happen here.

@rbarreiros @hippyau @dirkdamerau @kyleawayan

joakyhsf69, 2024-12-30:
Ok, there’s no rush. In fact, what I use the most is source=LTC, but I’m running tests because I like the project and want it to be as good as possible. Thanks again!

vanvught, 2024-12-30:
@joakyhsf69  Thank you for doing all the testing!

joakyhsf69, 2024-12-30:
> > > However, now with the source set to rtpMIDI or Artnet, it only sends zero through MIDI.
> > 
> > 
> > Will investigate now @joakyhsf69
> 
> Fixed.

Yes, now source=rtpMIDI and source=Artnet are sending the code correctly, and it seems there are no jumps in MA3, great. However, source=MIDI is still sending zeros through rtpMIDI. In MIDI-OX, I can see the messages, but the clock doesn't run. Through Artnet, it works fine.

vanvught, 2024-12-31:
@joakyhsf69 
> That's a challenge; working with MIDI-OX and not working with MA3.

Please give` source=systime` a test with 24 FPS and 30 FPS. That is working. MA3 does not like the 25 FPS 🤔

It is 01:47 local time. Time for some sleep.

vanvught, 2024-12-31:
> However, source=MIDI is still sending zeros through rtpMIDI

Fixed. Not able to test it. 

> that's a challenge; working with MIDI-OX and not working with MA3. 

Fixed. And successfully tested.

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18282397/orangepi_zero.uImage.gz)

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan
Happy New Year!

joakyhsf69, 2025-01-02:
> Fixed. Not able to test it.
>
>
> Fixed. And successfully tested.
>
> [orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18282397/orangepi_zero.uImage.gz)
>
> @joakyhsf69 @rbarreiros @hippyau [...]

Happy New Year to everyone! I'm sorry, but is it possible that you haven't uploaded the firmware version that fixes these last two things? With source=midi, I’m still getting zeros in the RTP-MIDI output, and I only receive the first frame. Additionally, with source=midi and source=systime, it works in MA3 at 24/29.97/30 frames, but not at 25 — it’s still the same there. Or is it something I haven't done correctly? Thanks Again

vanvught, 2025-01-02:
>  is it possible that you haven't uploaded the firmware version that fixes these last two things?

It would be an unfortunate mistake on my part. I will give it a check tomorrow. Hopefully then I can also fix the 25FPS issue with `source=internal`.

vanvught, 2025-01-03:
@joakyhsf69  I've checked the previous attached firmware and it has the fixes.

I have attached the latest build. I did some major refactoring. This helps managing the source code easier.

Still the issue with `source=internal`. At my side, grandMA3 onPC does not like the RTP-MIDI at all 🤷‍♂️. After some more coffee, I will continue work on this issue.

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18299610/orangepi_zero.uImage.gz)

@rbarreiros @hippyau @dirkdamerau @kyleawayan

kyleawayan, 2025-01-05:
Hi all,

I have some time tonight and tomorrow with my setup. I tried out the latest firmware in the above message but actually cannot get MA3 to understand the timecode at all with an Art-Net input.

I am currently using the setup:
```
Depence R3 -> (Art-Net Timecode) -> Orange Pi -> (rtp-MIDI) -> grandMA3 onPC on Windows
```

I see the timecode reflecting correctly on the Orange Pi, but not in MA3. Using the [Protokol](https://hexler.net/protokol) app, I see that quarter frames are being successfully received on my Windows machine. Some weird observation: I was able to see the MA3 clock viewer jump to a certain time. It sometimes happens when I change the MIDI settings to "None" then back to my rtpMIDI session. Overall, the timecode isn't lighting green nor counting up.

However testing with `source=internal`, it is reflected in MA3... and it also jumps a lot.

kyleawayan, 2025-01-05:
Also one more thing I noticed with using `source=internal` with MA3 via rtpMIDI...

In the firmware above in this thread, stopping the internal timecode does not stop it in MA3. Compared to the production firmware, stopping the internal timecode also stops it in MA3 (the clock goes from green to grey).

vanvught, 2025-01-05:
Continue in #297

closed 2025-01-05 by vanvught