vanvught/rpidmx512 #297: LTC SMPTE: grandMA3 and RTP-MIDI output

vanvught, 2025-01-05:
This is a follow-up for the issues reported in #291  #294 #296 

Successfully tested:

- [ ] `source=ltc`
- [ ] `source=artnet`
- [ ] `source=midi`
- [ ] `source=tcnet`
- [ ] `source=internal`
- [ ] `source=systime`
- [ ] `source=etc`

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan 

https://github.com/vanvught/rpidmx512/issues/296#issuecomment-2571543954

>  I tried out the latest firmware in the above message but actually cannot get MA3 to understand the timecode at all with an Art-Net input.

https://github.com/vanvught/rpidmx512/issues/296#issuecomment-2571546294

> Also one more thing I noticed with using source=internal with MA3 via rtpMIDI...
> 
> In the firmware above in this thread, stopping the internal timecode does not stop it in MA3. Compared to the production firmware, stopping the internal timecode also stops it in MA3 (the clock goes from green to grey).

vanvught, 2025-01-05:
Testing  `source=artnet`

Windows 10 Pro Art-Net -> Orange Pi Zero RTP-MIDI -> macOS grandMA3 onPC

https://github.com/user-attachments/assets/0d83cb7e-0ddb-4923-9549-6cb7a85c7590

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-05:
Testing  `source=internal`

LTC Generator Orange Pi Zero RTP-MIDI -> macOS grandMA3 onPC

Note: When stopped, there is no output anymore.

https://github.com/user-attachments/assets/0c63923c-8c37-44a5-81c1-5e01a017d6b7

Test UI for checking error's in Art-Net Timecode (will be open source) ->
[ArtNetTimeCode.jar.zip](https://github.com/user-attachments/files/18310112/ArtNetTimeCode.jar.zip)

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-05:
Testing  `source=tcnet' FPS25 is **not working**. FPS30 is fine.

macOS TCNet -> Orange Pi Zero RTP-MIDI -> macOS grandMA3 onPC

https://github.com/user-attachments/assets/05711187-7307-4d3e-9c31-f2d89e88143f

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-05:
Dear all, please find attached the firmware which I have successfully tested -> https://github.com/vanvught/rpidmx512/issues/297#issue-2769219406

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18310233/orangepi_zero.uImage.gz)

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-05:
Dear all, please find attached the firmware which I have successfully tested -> https://github.com/vanvught/rpidmx512/issues/297#issue-2769219406

`version:[V5.0] Orange Pi Zero Compiled on Jan  5 2025 at 20:35:13`
[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18311476/orangepi_zero.uImage.gz)

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-05:
Testing  `source=etc` FPS25

source=systime [GD32F407](https://www.gd32-dmx.org) ETC output -> Orange Pi Zero RTP-MIDI -> macOS grandMA3 onPC

ETC UDP Tester -> https://github.com/ETCLabs/UDPMidiTest/tree/master with issue https://github.com/ETCLabs/UDPMidiTest/issues/1

https://github.com/user-attachments/assets/16d4b20a-af77-4a95-9074-5b78095a8bd7

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-05:
Dear all, please find attached the firmware which I have successfully tested -> https://github.com/vanvught/rpidmx512/issues/297#issue-2769219406

`version:[V5.0] Orange Pi Zero Compiled on Jan  5 2025 at 21:00:27`
[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18311562/orangepi_zero.uImage.gz)

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

kyleawayan, 2025-01-05:
Hi Arjan,

Thank you for investigating. I just installed the new firmware. Running with `source=internal` at 24fps, it looks like everything is working smoothly. In MA3, using rtpMIDI, I haven't got any timecode jump logs running the past 4 minutes so far. Also, the timecode start/stop is reflected correctly. I am also testing with Resolume, which takes a SMPTE audio input, and it looks like that is still good too.

However now going back to my other setup with `source=artnet` (I should've noted Depence and grandMA3 are running on the same computer, if that makes any difference)
```
Depence R3 -> (30fps Art-Net Timecode) -> Orange Pi -> (rtp-MIDI) -> grandMA3 onPC on Windows
```

grandMA3 doesn't respond to the timecode at all. I see the MIDI messages flowing in though on the Protokol app though.

However, I think this is a fault on Depence, as I'm actually able to test Art-Net timecode output with another software. With `source=artnet` using the [Timecode Expert](https://txl20.com/timecode-expert/) to output Art-Net timecode (using a separate computer though), everything works fine. Even the starting and stopping in MA3.

So looks like we are good on `source=artnet`--will just have to find another alternative for sending timecode out from Depence.

Thank you!

kyleawayan, 2025-01-05:
Okay was just experimenting around, and

```
Depence R3 -> (30fps Art-Net Timecode) -> grandMA3 onPC on separate MacBook
```

works fine. Although Depence doesn't reflect the start/stop to MA3, the clock just stays green the whole time. Maybe this could be part of the problem?

Not sure if the Art-Net timecode spec has start/stop commands. If I were to guess, Depence isn't sending a start/stop packet, making the rtpMIDI timecode never count up--because there was no start command.

Would it be possible to have an option in `ltc.txt` or something, to ignore these start/stop commands for softwares that don't support it, like Depence?

Note I do want to make clear that my Orange Pi is understanding the timecode from Depence, it's just grandMA3 or Timecode Expert can't read the MIDI messages from it for some reason.

hippyau, 2025-01-05:
There is no start/stop.  Sou ds like Depence be sending the same frame over
and over.  Try the Mute on Idle option in Depence.

kyleawayan, 2025-01-05:
Okay one more thing to add before I log off for now, I do notice the quarter messages coming in using `source=artnet` and Art-Net timecode from Depence, look a little strange:

https://github.com/user-attachments/assets/dbce7719-52d9-44aa-9d51-c85584ee1507

Compared to `source=internal` where the MIDI timecode works correctly:

https://github.com/user-attachments/assets/12a78146-1622-4046-a6ef-e02be5fda63f

Hopefully we will find a way to get all these softwares working together.

kyleawayan, 2025-01-05:
> There is no start/stop. Sou ds like Depence be sending the same frame over and over. Try the Mute on Idle option in Depence.

Hi,

I tried that option but no difference. I even restarted Depence and my Orange Pi, and MIDI quarter frames are still being sent even when the Depence sequence is paused.

vanvught, 2025-01-06:
Testing  `source=artnet` FPS30
```
macOS ShowKontrol Art-Net -> Orange Pi Zero RTP-MIDI -> macOS grandMA3 onPC
                                                 LTC -> Audio USB -> macOS TimeCode Monitor
```

Skipping in Art-Net TimeCode is now also working.

https://github.com/user-attachments/assets/acd271b0-58b8-4a6c-9947-580cdb38043d

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-06:
Hi Kyle, Hip,

Thank you for the detailed feedback.

> will just have to find another alternative for sending timecode out from Depence.

I have added some more functionality. It should not break other's. I have done a quick test here -> https://github.com/vanvught/rpidmx512/issues/297#issuecomment-2573390994 The output from Showkontrol is not perfect. But it gives an idea that it works with grandMA3. As I have a license for Showkontrol, I will have a chat with the developers. 

Thanks, Arjan

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18321776/orangepi_zero.uImage.gz)

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

joakyhsf69, 2025-01-09:
Hi!. This is the best firmware I’ve tested so far. However, I’ve detected some new issues, most of them with source=MIDI. RTPMidi output still doesn’t work. Additionally, when you cut the signal, the clock keeps running until you send another signal. 
I’ve also noticed another problem that I hadn’t seen before. In all sources, 24fps doesn’t work on LTC out—the timecode output is erratic. It seems to be a general issue with that output, regardless of the source; at other framerates, it appears to work fine. I tested it using the SMPTE input of Resolume Arena and a synchronized video to check if it ran smoothly. Of course, in Resolume, I selected the corresponding framerate according to what was being sent. The jumps are already visible on the clock. 
Issues like the generator at 25fps in MA3 seem to be resolved. Good job!

vanvught, 2025-01-10:
> . In all sources, 24fps doesn’t work on LTC out—the timecode output is erratic.

I cannot replicate this issue observed.

https://github.com/user-attachments/assets/90c0ab6f-b39e-4081-b6e0-c0a081cea6a7

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-01-10:
> most of them with source=MIDI. RTPMidi output still doesn’t work.

I have planned to rewrite the DIN-MIDI reader. I will check the To-Do list here https://github.com/vanvught/rpidmx512/issues/297#issue-2769219406 when done.

kyleawayan, 2025-01-11:
> I have added some more functionality. It should not break other's. I have done a quick test here -> https://github.com/vanvught/rpidmx512/issues/297#issuecomment-2573390994 The output from [...]

Hi Arjan,

Thank you for making the change. Looking at your video, this looks awesome. Unfortunately, I cannot get it to work with Depence. The Art-Net timecode works fine with Timecode Expert. I'm sure this is a fault on Depence's part. I would probably trust ShowKontrol more to get the Art-Net timecode implementation correctly.

> In all sources, 24fps doesn’t work on LTC out—the timecode output is erratic. It seems to be a general issue with that output, regardless of the source; at other framerates, it appears to work fine. [...]

My setup:

```
TXL20 Timecode Expert on MacBook -> (24fps Art-Net Timecode) -> Orange Pi -> (SMPTE audio input with audio interface) -> Resolume Arena on PC
```

works fine at 24fps. Alongside Resolume Arena, I tested the SMPTE audio input with the [Timecode Monitor](https://txl20.com/timecode-monitor/) software on the same computer running Resolume, and it works fine.

vanvught, 2025-01-24:
`source=midi `is now working
@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan 

[orangepi_zero.uImage.gz](https://github.com/user-attachments/files/18538551/orangepi_zero.uImage.gz)

joakyhsf69, 2025-01-24:
Indeed, the `source=midi` is now working for me, but I've still noticed some issues. When `source=artnet`, the MIDI, both DIN and RTPMIDI, still output 0 in the timecode. I’m not sure if this is a new issue or if it was already present and I just hadn’t noticed before. 

Additionally, I’m not sure if it’s a problem with the device, but sometimes I use free software like TouchDesigner to receive LTC, along with Resolume. With Resolume, the LTC output works fine at any frame rate, but in TouchDesigner, it only works correctly at 30 fps; other frame rates cause problems. I’m receiving the LTC in both programs using the same sound card. With other LTC sources, such as a track generated by the page "https://elteesee.pehrhovey.net/", it works fine in both programs. I’m not sure why—something must change for one source to work and the other not.

Other than that, this is the best and most stable firmware I’ve tested. It no longer causes issues in MA3, regardless of the source or frame rate. I don’t have devices with TCNet, so I can’t test that part, either as a source or output. Great work, cheers!

joakyhsf69, 2025-01-29:
Hello again, the issue I mentioned regarding the LTC out with TouchDesigner has been resolved by changing the sound card. I don’t know why, but with a different sound card, it works fine in both Resolume Arena and TouchDesigner.  

However, what is not working for me is the MIDI output with **source=artnet**. Neither MA3 nor MIDI-OX work with **rtpMIDI** or **MIDI DIN**. In MIDI-OX, I can see the frames coming in, but the clock does not move, and in MA3, I get an error stating that the framerate is not supported on the MIDI port.  

The LTC output does work fine with **source=artnet**.

vanvught, 2025-01-31:
> However, what is not working for me is the MIDI output with source=artnet. Neither MA3 nor MIDI-OX work with rtpMIDI or MIDI DIN.

That's weird. Kyle @kyleawayan and myself have no issues with `source=artnet.`

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau

vanvught, 2025-01-31:
> However, what is not working for me is the MIDI output with source=artnet. Neither MA3 nor MIDI-OX work with rtpMIDI or MIDI DIN. I

Working fine for me. See also https://github.com/vanvught/rpidmx512/issues/297#issuecomment-2613299603

[image]

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan

vanvught, 2025-02-05:
> However, what is not working for me is the MIDI output with source=artnet. Neither MA3 nor MIDI-OX work with rtpMIDI or MIDI DIN. 

Working fine for me.  UM-ONE is Roland MIDI DIN interface.  See also https://github.com/vanvught/rpidmx512/issues/297#issuecomment-2613299603

[image]

vanvught, 2025-02-07:
Dear all,

As all the issues has been resolved, I will close this issue soon. When there are new problems, then please open a new issue against the latest build V5.1 found on https://github.com/vanvught/h3dmx512-zip

Thanks, Arjan 

@joakyhsf69 @rbarreiros @hippyau @dirkdamerau @kyleawayan