vanvught/rpidmx512 #299: How to use/instal ?

tutle, 2025-05-24:
Hello, I would like to build a 4 universes artnet node unit with wifi and RDM but the page [doesn't exist anymore](https://www.orangepi-dmx.org/raspberry-pi-art-net-dmx-out), I'm a litle lost
Is it possible to build with [orange pi 3](https://fr.aliexpress.com/item/1005005913452746.html) (with wifi) ?
and 4 modules [RS485 like those](https://www.amazon.com/dp/B07YZTGHGG) ?
Any help is welcome
Have a nice day
Yvan

gobo-ws, 2025-05-24:
You can check it out on the Wayback Machine:
[https://web.archive.org/web/20210227085420/https://www.orangepi-dmx.org/raspberry-pi-art-net-dmx-out](https://web.archive.org/web/20210227085420/https://www.orangepi-dmx.org/raspberry-pi-art-net-dmx-out)
For the other stuff, I think Arjan can help.

tutle, 2025-06-20:
I will step down to 2 universe orange pi zero to begin but I'm quiet newbee and don't know how to install and also how to connect rs485 boards... Still learning.
Also, is it possible to use ethernet connection as well as wifi for safer connections ?
I made some [crmx transmitters](https://padulux-fr.translate.goog/spip/spip.php?article3&_x_tr_sl=fr&_x_tr_tl=en&_x_tr_hl=fr&_x_tr_pto=wapp) but need to learn more about coding.
Thanks